esp-rs/espflash #597: cargo install cargo-espflash fails with error of build command for `openssl-sys v0.9.101`

freew01f, 2024-02-23:
I run  cargo install cargo-espflash got error 
my env is below

```
rustup show                 
Default host: aarch64-apple-darwin
rustup home:  /Users/freewolf/.rustup

installed toolchains
--------------------

stable-aarch64-apple-darwin (default)
nightly-aarch64-apple-darwin
esp

installed targets for active toolchain
--------------------------------------

aarch64-apple-darwin
riscv32imac-unknown-none-elf
riscv32imafc-unknown-none-elf
riscv32imc-unknown-none-elf

active toolchain
----------------

nightly-aarch64-apple-darwin (overridden by '/Users/freewolf/Documents/Study/esp32/test-51/rust-toolchain.toml')
rustc 1.78.0-nightly (397937d81 2024-02-22)

 cargo install cargo-espflash
    Updating crates.io index
  Installing cargo-espflash v2.1.0
    Updating crates.io index
   Compiling libc v0.2.153
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling cfg-if v1.0.0
   Compiling pkg-config v0.3.30
   Compiling vcpkg v0.2.15
   Compiling serde v1.0.197
   Compiling memchr v2.7.1
   Compiling autocfg v1.1.0
   Compiling thiserror v1.0.57
   Compiling aho-corasick v1.1.2
   Compiling crc32fast v1.4.0
   Compiling regex-syntax v0.8.2
   Compiling smallvec v1.13.1
   Compiling once_cell v1.19.0
   Compiling lock_api v0.4.11
   Compiling parking_lot_core v0.9.9
   Compiling scopeguard v1.2.0
   Compiling adler v1.0.2
   Compiling miniz_oxide v0.7.2
   Compiling quote v1.0.35
   Compiling syn v2.0.50
   Compiling regex-automata v0.4.5
   Compiling cc v1.0.86
   Compiling version_check v0.9.4
   Compiling bitflags v2.4.2
   Compiling crossbeam-utils v0.8.19
   Compiling same-file v1.0.6
   Compiling libz-sys v1.1.15
   Compiling parking_lot v0.12.1
   Compiling walkdir v2.4.0
   Compiling flate2 v1.0.28
   Compiling num-traits v0.2.18
   Compiling itoa v1.0.10
   Compiling bstr v1.9.0
   Compiling crossbeam-channel v0.5.11
   Compiling prodash v25.0.2
[... 178 lines cut ...]
  ***   (If you are new to OpenSSL, you might want to consult the    ***
  ***   'Troubleshooting' section in the INSTALL.md file first)      ***
  ***                                                                ***
  **********************************************************************
  running cd "/var/folders/gl/tjcxhg593tlg7qndbxgr1d8h0000gn/T/cargo-installtAXYUY/release/build/openssl-sys-3fdfd723e5951a19/out/openssl-build/build/src" && "make" "depend"
  running cd "/var/folders/gl/tjcxhg593tlg7qndbxgr1d8h0000gn/T/cargo-installtAXYUY/release/build/openssl-sys-3fdfd723e5951a19/out/openssl-build/build/src" && MAKEFLAGS="-j --jobserver-fds=12,13 --jobserver-auth=12,13" "make" "build_libs"
  perl "-I." "-Iutil/perl" "-Mconfigdata" "-MOpenSSL::paramnames" "util/dofile.pl" "-oMakefile" crypto/params_idx.c.in > crypto/params_idx.c

  --- stderr
  make: *** read jobs pipe: Resource temporarily unavailable.  Stop.
  make: *** Waiting for unfinished jobs....
  thread 'main' panicked at /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/openssl-src-300.2.3+3.2.1/src/lib.rs:611:9:

  Error building OpenSSL:
      Command: cd "/var/folders/gl/tjcxhg593tlg7qndbxgr1d8h0000gn/T/cargo-installtAXYUY/release/build/openssl-sys-3fdfd723e5951a19/out/openssl-build/build/src" && MAKEFLAGS="-j --jobserver-fds=12,13 --jobserver-auth=12,13" "make" "build_libs"
      Exit status: exit status: 2

      
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-espflash v2.1.0`, intermediate artifacts can be found at `/var/folders/gl/tjcxhg593tlg7qndbxgr1d8h0000gn/T/cargo-installtAXYUY`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

```
how ca i fix it ? thanks

SergioGasquez, 2024-02-23:
Other colleagues have encountered this issue too, its a current third party limitation, see the issues:
- https://github.com/rust-lang/cc-rs/issues/962
- https://github.com/rust-lang/git2-rs/issues/1026

I would recommend using the binaries from the release: https://github.com/esp-rs/espflash/releases:
```
curl -L "https://github.com/esp-rs/espflash/releases/latest/download/cargo-espflash-aarch64-apple-darwin.zip" -o "${HOME}/.cargo/bin/cargo-espflash.zip"
unzip "${HOME}/.cargo/bin/cargo-espflash.zip" -d "${HOME}/.cargo/bin/" 
rm "${HOME}/.cargo/bin/cargo-espflash.zip"
chmod u+x "${HOME}/.cargo/bin/cargo-espflash"

NobodyXu, 2024-02-23:
I tried on my own project that updating to make 4.4.1 fixed this error.
Could you give it a shot please?

playfulFence, 2024-02-23:
@NobodyXu yes, your solution definitely fixes the error! Thank you for your response though! ❤️ 

A quick guide (at least worked for me. Requires updating the PATH which may be not ideal, but at least fixes the issue):

```shell
brew install make
PATH="/opt/homebrew/opt/make/libexec/gnubin:$PATH"
```
Second command depends on your particular file location on the system, usually `brew` will give the path where it installed the new make.

NobodyXu, 2024-02-23:
That's good to hear!

I also submit a PR in cc to fix the issue for older make (<4.3)

NobodyXu, 2024-02-24:
cc 1.0.87 has released which should fix this

freew01f, 2024-02-26:
> @NobodyXu yes, your solution definitely fixes the error! Thank you for your response though! ❤️
>
> A quick guide (at least worked for me. Requires updating the PATH which may be not ideal, but at [...]

It works thanks very much

NobodyXu, 2024-02-26:
cc 1.0.88 has released, which should fix this even if you are using make < 4.3

freew01f, 2024-02-26:
> Seems like the issue still persist with make < 4.3. I made #602 which updates the `cc` version to 1.0.88, but the issue still persists when doing `cargo install cargo-espflash --git https://github.com/SergioGasquez/espflash --branch fix/mac-install --force`

cc1.088 have the other problem

NobodyXu, 2024-02-26:
> cc1.088 have the other problem 

Can you please elaborate?

freew01f, 2024-02-26:
> > cc1.088 have the other problem
> 
> Can you please elaborate?

https://github.com/esp-rs/esp-idf-sys/issues/287

see here

1.0.84 -87 can not build

NobodyXu, 2024-02-26:
@freew01f That's a different error.
It's due to `--target` flag we now passed to `clang`, probably same as rust-lang/cc-rs#903

PR #602 (2024-03-04): Fix macOS installation
- Update `cargo-espflash` deps and `cargo update` to get [`cc>=1.0.88`](https://github.com/esp-rs/espflash/issues/597#issuecomment-1963259738) which should fix #597 
- Bump some CI actions version to avoid warnings, see https://github.com/esp-rs/espflash/actions/runs/8046179600 for example

~Draft until we confirm that #597 is solved with this dependency update.~
closed 2024-03-04 by SergioGasquez