esp-rs/espflash #601: Read-only mode when monitoring?

hongquan, 2024-02-25:
I'm using `espflash` v2.1.0 and I found that when I do `espflash monitor`, it acquires exclusive access on the serial port. I want to use ESP32's `UART0` to read command from PC, it means that I want the serial port is writable for other program. Could you add one mode to `espflash monitor` so that it only need read access on the port?

SergioGasquez, 2024-02-27:
Hi! Have you tried something like https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/serial_interrupts.rs?  Does that fulfill your requirements?

hongquan, 2024-02-27:
Hi @SergioGasquez , I don't understand use case of that [example](https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/serial_interrupts.rs), so it may not be what I'm looking for.

What I need is that `espflash monitor` command don't take exclusive access on `/dev/ttyUSB0`, so I can use another program, like Python `pyserial` to write some command to my ESP32 board's `UART0`.

On another hand, I want to use ESP32 `UART0` both for flashing, monitor and receiving command from PC.

MabezDev, 2024-02-27:
@hongquan Thanks for the request! I don't think this is something we want to handle in espflash itself. Feel free to use espflash as a library and manage your own monitor which doesn't take full control of the port.

closed 2024-02-27 by MabezDev

hongquan, 2024-02-28:
Thank you, @MabezDev .
I respect your vision of `espflash`. If the feature cannot be added to `espflash`, I will use pyserial's `miniterm` to monitor the board.