esp-rs/espflash #604: Flashing with baud rate higher than 115,200 causes a timeout in `v3.0.0-rc.1`

AnthonyGrondin, 2024-02-27:
I'm running the same flash command, with `--baud 921600`, and it used to work previously, (IE in v2.1.0) but it is now returning a timeout error.

**With v2.1.0 (expected result)**:
```log
     Running `espflash flash --port /dev/ttyUSB1 --baud 921600 --monitor target/xtensa-esp32s3-none-elf/release/my_app`
[2024-02-27T19:59:58Z INFO ] 🚀 A new version of espflash is available: v3.0.0-rc.1
[2024-02-27T19:59:58Z INFO ] Serial port: '/dev/ttyUSB1'
[2024-02-27T19:59:58Z INFO ] Connecting...
[2024-02-27T19:59:59Z INFO ] Using flash stub
[2024-02-27T19:59:59Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       7c:df:a1:e2:5a:0c
App/part. size:    982,864/8,323,072 bytes, 11.81%
[00:00:00] [========================================]      14/14      0x0                                                                                                                        
[00:00:00] [========================================]       1/1       0x8000                                                                                                                     
[00:00:15] [========================================]     599/599     0x10000                                                                                                                    
[2024-02-27T20:00:16Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 
...
```

**With v3.0.0-rc.1 (bug: timeout)**:
```log
     Running `espflash flash --port /dev/ttyUSB1 --baud 921600 --monitor target/xtensa-esp32s3-none-elf/release/my_app`
[2024-02-27T20:01:50Z INFO ] Serial port: '/dev/ttyUSB1'
[2024-02-27T20:01:50Z INFO ] Connecting...
[2024-02-27T20:01:51Z INFO ] Using flash stub
[2024-02-27T20:01:52Z WARN ] Setting baud rate higher than 115,200 can cause issues
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ChangeBaudrate command

...
```

SergioGasquez, 2024-02-27:
Hi! Thanks for reporting this issue. I just did a quick try and I was able to reproduce it. It looks like a stub issue as using `--no-stub` solves the issue. Here are the debug logs using the stub: 
```
[2024-02-27T20:25:44Z WARN ] Setting baud rate higher than 115,200 can cause issues
[2024-02-27T20:25:44Z DEBUG] Change baud to: 921600
[2024-02-27T20:25:44Z DEBUG] Writing command: ChangeBaudrate { new_baud: 921600, prior_baud: 115384 }
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ChangeBaudrate command
```

The `prio_baud` looks a bit weird to me.

When trying the same command for an ESP32-C3, the command succeeds:
```
❯ espflash flash  --baud 921600 --monitor target/riscv32imc-unknown-none-elf/debug/esp32c3
[2024-02-27T20:30:54Z INFO ] Serial port: '/dev/ttyACM0'
[2024-02-27T20:30:54Z INFO ] Connecting...
[2024-02-27T20:30:54Z INFO ] Using flash stub
[2024-02-27T20:30:54Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       60:55:f9:c0:39:7c
App/part. size:    223,248/4,128,768 bytes, 5.41%
[2024-02-27T20:30:54Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-02-27T20:30:54Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-02-27T20:30:54Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-02-27T20:30:55Z INFO ] Flashing has completed!
```

SergioGasquez, 2024-02-27:
A bit more of information, I've regenerated the S3 stub with current `main` of `esp-flasher-stub`, enabling `dprint`s and the stubs receives the change baud command:
```
process command: ChangeBaudrate
                 Waiting for command
```
So, either we are not [processing it properly](https://github.com/esp-rs/esp-flasher-stub/blob/main/src/protocol.rs#L446-L453) or `espflash` is not reading the [OHAI response](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/serial-protocol.html#initial-synchronisation) properly

I will be on vacations for the rest of the week! Will take a look at it as soon as I come back!