esp-rs/espflash #614: esp32c2 use espflash v3 is error，v2.1 is normal

xiaguangbo, 2024-03-13:
And now only recognize 2MB space，before can recognize 4MB

```
Using 26MHz bootloader
App/part. size:    479,184/2,031,616 bytes, 23.59%
```

## espflash v3 log

```
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo run
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.23s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/esp32c2`
[2024-03-13T15:53:25Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-13T15:53:25Z INFO ] Connecting...
[2024-03-13T15:53:25Z INFO ] Using flash stub
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0f:6f:b0
Using 26MHz bootloader
App/part. size:    479,184/2,031,616 bytes, 23.59%
[00:00:01] [========================================]      12/12      0x0                                                                                                                                                                                                                                                                                                       
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                                                                                                                                    
[00:00:21] [========================================]     216/216     0x10000                                                                                                                                                                                                                                                                                                   [2024-03-13T15:53:51Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 001f0000
I (126) boot: End of partition table
I (132) esp_image: segment 0: paddr=00010020 vaddr=3fca8770 size=0108ch (  4236) load
I (147) esp_image: segment 1: paddr=000110b4 vaddr=40380000 size=06f64h ( 28516) load
I (169) esp_image: segment 2: paddr=00018020 vaddr=3c048020 size=1e298h (123544) map
I (214) esp_image: segment 3: paddr=000362c0 vaddr=40386f64 size=0180ch (  6156) load
I (216) esp_image: segment 4: paddr=00037ad4 vaddr=00000000 size=08544h ( 34116) 
I (234) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=44f8ch (282508) map
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (352) cpu_start: Unicore app
I (353) cpu_start: Pro cpu up.

assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)
Core  0 register dump:
MEPC    : 0x40380ee4  RA      : 0x403839f6  SP      : 0x3fcde5d0  GP      : 0x3fca8f70  
0x40380ee4 - panic_abort
    at /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/panic.c:472
0x403839f6 - __ubsan_include
[... 69 lines cut ...]
3fcde970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde990: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde9b0: 0x600c5100 0x600c5100 0x00000020 0x40055c20 0x52b5a563 0x37de2ba5 0xd7a54895 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380ea6
0x40380ea6 - esp_restart_noos
    at /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/port/soc/esp32c2/system_internal.c:96
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (36) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (36) boot: compile time Jan 16 2024 14:05:06
I (36) boot: chip revision: v1.0
I (41) boot.esp32c2: MMU Page Size  : 32K
...
```

SergioGasquez, 2024-03-14:
Can you post the `espflash v2 log`? When espflash detects the board, it detects `Flash size:        4MB`. You can also use `--flash-size 4mb` arg, or use `cargo espflash` so it will use the partition table from your project: https://github.com/esp-rs/espflash/tree/main/cargo-espflash#bootloader-and-partition-table

xiaguangbo, 2024-03-14:
> Can you post the `espflash v2 log`? When espflash detects the board, it detects `Flash size: 4MB`. You can also use `--flash-size 4mb` arg, or use `cargo espflash` so it will use the partition table [...]

## espflash v2

```
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo run
   Compiling esp-idf-sys v0.34.1
   Compiling esp-idf-hal v0.43.1
   Compiling esp-idf-svc v0.48.1
   Compiling esp32c2 v0.1.0 (/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2)
    Finished `dev` profile [optimized + debuginfo] target(s) in 12.56s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/esp32c2`
[2024-03-14T08:51:13Z INFO ] 🚀 A new version of espflash is available: v3.0.0
[2024-03-14T08:51:13Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T08:51:13Z INFO ] Connecting...
[2024-03-14T08:51:13Z INFO ] Using flash stub
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0f:6f:b0
App/part. size:    479,184/4,128,768 bytes, 11.61%
[00:00:01] [========================================]      12/12      0x0                                                                                                                                                                                                                                                                                                 
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                                                                                                                                                                              
[00:00:12] [======================>                 ]     119/216     0x10000 
```

My esp32c2 uses a 26M crystal oscillator, so the default baud rate is 74880, espflash v2 will be garbled when using "monitor" directly, so I have to manually set the baud rate of "monitor" to 74880, v3 fixes this problem, and will automatically set the baud rate of "monitor".

## espflash monitor --baud=74880

```
@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ espflash monitor --baud=74880
[2024-03-14T08:52:37Z INFO ] 🚀 A new version of espflash is available: v3.0.0
[2024-03-14T08:52:37Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T08:52:37Z INFO ] Connecting...
[2024-03-14T08:52:38Z INFO ] Using flash stub
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16d4
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x2940
SHA-256 comparison failed:
Calculated: ad6ff16b8d13d022675746e243b11a98af205407b4efcf05e212db2c83d4dc6e
Expected: a711db46493820858635c6aba1190c19fe3574ac73a8fc8fa7d99da179796012
Attempting to boot anyway...
entry 0x403acb70
I (39) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (39) boot: compile time Jun  8 2023 05:47:15
I (41) boot: chip revision: v1.0
I (44) boot.esp32c2: MMU Page Size  : 32K
I (49) boot.esp32c2: SPI Speed      : 30MHz
I (54) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 003f0000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3fca8770 size=0108ch (  4236) load
I (117) esp_image: segment 1: paddr=000110b4 vaddr=40380000 size=06f64h ( 28516) load
I (134) esp_image: segment 2: paddr=00018020 vaddr=3c048020 size=1e298h (123544) map
I (177) esp_image: segment 3: paddr=000362c0 vaddr=40386f64 size=0180ch (  6156) load
I (179) esp_image: segment 4: paddr=00037ad4 vaddr=00000000 size=08544h ( 34116) 
I (194) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=44f8ch (282508) map
I (294) boot: Loaded app from partition at offset 0x10000
I (294) boot: Disabling RNG early entropy source...
I (305) cpu_start: Unicore app
I (306) cpu_start: Pro cpu up.
I (314) cpu_start: Pro cpu start user code
I (314) cpu_start: cpu freq: 120000000 Hz
I (315) cpu_start: Application information:
I (317) cpu_start: Project name:     libespidf
I (323) cpu_start: App version:      1
I (327) cpu_start: Compile time:     Mar 14 2024 15:56:21
I (333) cpu_start: ELF file SHA256:  0000000000000000...
I (339) cpu_start: ESP-IDF:          v5.1.3
I (344) cpu_start: Min chip rev:     v1.0
I (349) cpu_start: Max chip rev:     v1.99 
I (353) cpu_start: Chip rev:         v1.0
I (358) heap_init: Initializing. RAM available for dynamic allocation:
I (365) heap_init: At 3FCAA730 len 00032440 (201 KiB): D/IRAM
I (372) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (380) spi_flash: detected chip: gd
I (383) spi_flash: flash io: dio
W (387) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (395) sleep: Configure to isolate all GPIO pins in sleep state
I (402) sleep: Enable automatic switching of GPIO sleep configuration
I (409) app_start: Starting scheduler on CPU0
I (414) main_task: Started on CPU0
I (414) main_task: Calling app_main()
I (424) esp32c2: Hello, world!
I (424) main_task: Returned from app_main()
```

SergioGasquez, 2024-03-14:
`assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)` . Have you updated the `sdkconfig` as we spoke in the other issue? https://github.com/esp-rs/std-training/issues/240#issuecomment-1996834497

xiaguangbo, 2024-03-14:
> `assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)` . Have you updated the `sdkconfig` as we spoke in the other issue? [esp-rs/std-training#240 (comment)](https://github.com/esp-rs/std-training/issues/240#issuecomment-1996834497)

## espflash v3

in sdkconfig.defaults add `CONFIG_XTAL_FREQ_26=y`，`cargo clean` `cargo run` is ok!
but now only recognize 2MB space...`App/part. size:    479,184/2,031,616 bytes, 23.59%`，But can identify to Flash is 4MB

```
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo run
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.22s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/esp32c2`
[2024-03-14T09:55:07Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T09:55:07Z INFO ] Connecting...
[2024-03-14T09:55:08Z INFO ] Using flash stub
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0f:6f:b0
Using 26MHz bootloader
App/part. size:    479,184/2,031,616 bytes, 23.59%
[00:00:01] [========================================]      12/12      0x0                                                                                     
[00:00:00] [========================================]       1/1       0x8000                                                                                  
[00:00:21] [========================================]     216/216     0x10000                                                                                 [2024-03-14T09:55:33Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 001f0000
I (126) boot: End of partition table
I (132) esp_image: segment 0: paddr=00010020 vaddr=3fca87b0 size=0108ch (  4236) load
I (147) esp_image: segment 1: paddr=000110b4 vaddr=40380000 size=06f64h ( 28516) load
I (169) esp_image: segment 2: paddr=00018020 vaddr=3c048020 size=1e298h (123544) map
I (214) esp_image: segment 3: paddr=000362c0 vaddr=40386f64 size=01840h (  6208) load
I (216) esp_image: segment 4: paddr=00037b08 vaddr=00000000 size=08510h ( 34064) 
I (234) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=44f88h (282504) map
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (352) cpu_start: Unicore app
I (353) cpu_start: Pro cpu up.
I (364) cpu_start: Pro cpu start user code
I (364) cpu_start: cpu freq: 120000000 Hz
I (364) cpu_start: Application information:
I (369) cpu_start: Project name:     libespidf
I (377) cpu_start: App version:      1
I (384) cpu_start: Compile time:     Mar 14 2024 17:44:26
I (393) cpu_start: ELF file SHA256:  0000000000000000...
I (402) cpu_start: ESP-IDF:          v5.1.3
I (410) cpu_start: Min chip rev:     v1.0
I (417) cpu_start: Max chip rev:     v1.99 
I (424) cpu_start: Chip rev:         v1.0
I (432) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCAA770 len 00032400 (201 KiB): D/IRAM
I (452) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (464) spi_flash: detected chip: gd
I (469) spi_flash: flash io: dio
W (476) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (489) sleep: Configure to isolate all GPIO pins in sleep state
I (499) sleep: Enable automatic switching of GPIO sleep configuration
I (510) app_start: Starting scheduler on CPU0
I (518) main_task: Started on CPU0
I (518) main_task: Calling app_main()
I (528) esp32c2: Hello, world!
I (528) main_task: Returned from app_main()
```

## Manually set file path

now recognize 1MB space

```
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo espflash flash --bootloader='target/riscv32imc-esp-espidf/debug/bootloader.bin' --partition-table='target/riscv32imc-esp-espidf/debug/partition-table.bin'
[2024-03-14T09:51:54Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T09:51:54Z INFO ] Connecting...
[2024-03-14T09:51:54Z INFO ] Using flash stub
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.16s
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0f:6f:b0
Bootloader:        target/riscv32imc-esp-espidf/debug/bootloader.bin
Partition table:   target/riscv32imc-esp-espidf/debug/partition-table.bin
Using 26MHz bootloader
App/part. size:    479,184/1,048,576 bytes, 45.70%
[00:00:01] [========================================]      12/12      0x0                                                                                     
[00:00:00] [========================================]       1/1       0x8000                                                                                  
[00:00:20] [=====================================>  ]     204/216     0x10000
```

SergioGasquez, 2024-03-14:
Does this also gets resolved with https://github.com/esp-rs/std-training/issues/240 @xiaguangbo ?

xiaguangbo, 2024-03-14:
> Does this also gets resolved with [esp-rs/std-training#240](https://github.com/esp-rs/std-training/issues/240) @xiaguangbo ?

That problem is the crystal oscillator setting problem, has been solved. And I have to run `cargo clean` once. I set it before but didn't execute `cargo clean`, which not only didn't solve the problem, but also caused problems with espflash v2.

SergioGasquez, 2024-03-14:
Can you try using `cargo espflash flash --monitor`? I've tried reproducing your issue with no luck.
```
❯ cargo generate esp-rs/esp-idf-template cargo --name issue614 -d mcu=esp32c2 -d advanced=false
❯ cd issue614/
❯ echo "CONFIG_XTAL_FREQ_26=y" >> sdkconfig.defaults
❯ cargo r
...
Finished `dev` profile [optimized + debuginfo] target(s) in 2m 07s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/issue614`
2024-03-14T14:32:50Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T14:32:50Z INFO ] Connecting...
[2024-03-14T14:32:50Z INFO ] Using flash stub
[2024-03-14T14:32:51Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       08:3a:8d:40:51:c4
Using 26MHz bootloader
App/part. size:    479,984/2,031,616 bytes, 23.63%
[2024-03-14T14:32:51Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-03-14T14:32:52Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:05] [========================================]     217/217     0x10000[2024-03-14T14:32:58Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 001f0000
I (126) boot: End of partition table
I (132) esp_image: segment 0: paddr=00010020 vaddr=3fca87b0 size=0108ch (  4236) load
I (147) esp_image: segment 1: paddr=000110b4 vaddr=40380000 size=06f64h ( 28516) load
I (169) esp_image: segment 2: paddr=00018020 vaddr=3c048020 size=1e2d0h (123600) map
I (214) esp_image: segment 3: paddr=000362f8 vaddr=40386f64 size=01840h (  6208) load
I (216) esp_image: segment 4: paddr=00037b40 vaddr=00000000 size=084d8h ( 34008)
I (234) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=452a4h (283300) map
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (353) cpu_start: Unicore app
I (353) cpu_start: Pro cpu up.
I (364) cpu_start: Pro cpu start user code
I (365) cpu_start: cpu freq: 120000000 Hz
I (365) cpu_start: Application information:
I (369) cpu_start: Project name:     libespidf
I (377) cpu_start: App version:      1
I (384) cpu_start: Compile time:     Mar 14 2024 15:32:33
I (393) cpu_start: ELF file SHA256:  0000000000000000...
I (402) cpu_start: ESP-IDF:          v5.1.3
I (410) cpu_start: Min chip rev:     v1.0
I (417) cpu_start: Max chip rev:     v1.99
I (425) cpu_start: Chip rev:         v1.0
I (432) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCAA770 len 00032400 (201 KiB): D/IRAM
I (453) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (464) spi_flash: detected chip: gd
I (470) spi_flash: flash io: dio
W (476) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (489) sleep: Configure to isolate all GPIO pins in sleep state
I (499) sleep: Enable automatic switching of GPIO sleep configuration
I (510) app_start: Starting scheduler on CPU0
I (518) main_task: Started on CPU0
I (518) main_task: Calling app_main()
I (528) issue614: Hello, world!
I (528) main_task: Returned from app_main()     
```

SergioGasquez, 2024-03-14:
Did you add `CONFIG_XTAL_FREQ_26=y` to your `sdkconfig`? I've just managed to reproduce and solve it by:
```
❯ cargo generate esp-rs/esp-idf-template cargo --name issue614_2 -d mcu=esp32c2 -d advanced=false
❯ cd issue614-2/
❯ cargo b
❯ espflash erase-flash
❯ cargo r
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.05s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/issue614-2`
[2024-03-14T14:34:36Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T14:34:36Z INFO ] Connecting...
[2024-03-14T14:34:36Z INFO ] Using flash stub
[2024-03-14T14:34:37Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       08:3a:8d:40:51:c4
Using 26MHz bootloader
App/part. size:    479,984/2,031,616 bytes, 23.63%
[00:00:00] [========================================]      12/12      0x0    [00:00:00] [========================================]       1/1       0x8000 [00:00:04] [========================================]     217/217     0x10000[2024-03-14T14:34:44Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x297c
entry 0x403acb70
I (32) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (32) boot: compile time Jan 16 2024 14:05:06
I (32) boot: chip revision: v1.0
I (37) boot.esp32c2: MMU Page Size  : 32K
I (44) boot.esp32c2: SPI Speed      : 30MHz
I (51) boot.esp32c2: SPI Mode       : DIO
I (59) boot.esp32c2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 001f0000
I (126) boot: End of partition table
I (132) esp_image: segment 0: paddr=00010020 vaddr=3fca8770 size=0108ch (  4236) load
I (147) esp_image: segment 1: paddr=000110b4 vaddr=40380000 size=06f64h ( 28516) load
I (169) esp_image: segment 2: paddr=00018020 vaddr=3c048020 size=1e2e0h (123616) map
I (214) esp_image: segment 3: paddr=00036308 vaddr=40386f64 size=0180ch (  6156) load
I (216) esp_image: segment 4: paddr=00037b1c vaddr=00000000 size=084fch ( 34044)
I (234) esp_image: segment 5: paddr=00040020 vaddr=42000020 size=452ach (283308) map
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (353) cpu_start: Unicore app
I (353) cpu_start: Pro cpu up.

assert failed: esp_clk_init clk.c:70 (rtc_clk_xtal_freq_get() == CONFIG_XTAL_FREQ)
Core  0 register dump:
MEPC    : 0x40380ee4  RA      : 0x403839f6  SP      : 0x3fcde5d0  GP      : 0x3fca8f70
0x40380ee4 - panic_abort
    at /home/sergio/Documents/Espressif/tests/issue614-2/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/panic.c:472
0x403839f6 - __ubsan_include
    at /home/sergio/Documents/Espressif/tests/issue614-2/.embuild/espressif/esp-idf/v5.1.3/components/esp_system/ubsan.c:313
0x3fca8f70 - __global_pointer$
    at ??:??
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000001  S1      : 0x3fcde742  A0      : 0x3fcde628  A1      : 0x3fca8f05
[... 182 lines cut ...]
I (364) cpu_start: Pro cpu start user code
I (365) cpu_start: cpu freq: 120000000 Hz
I (365) cpu_start: Application information:
I (369) cpu_start: Project name:     libespidf
I (377) cpu_start: App version:      1
I (384) cpu_start: Compile time:     Mar 14 2024 15:35:14
I (393) cpu_start: ELF file SHA256:  0000000000000000...
I (402) cpu_start: ESP-IDF:          v5.1.3
I (410) cpu_start: Min chip rev:     v1.0
I (417) cpu_start: Max chip rev:     v1.99
I (425) cpu_start: Chip rev:         v1.0
I (432) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FCAA770 len 00032400 (201 KiB): D/IRAM
I (453) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (464) spi_flash: detected chip: gd
I (470) spi_flash: flash io: dio
W (476) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (489) sleep: Configure to isolate all GPIO pins in sleep state
I (499) sleep: Enable automatic switching of GPIO sleep configuration
I (510) app_start: Starting scheduler on CPU0
I (518) main_task: Started on CPU0
I (518) main_task: Calling app_main()
I (528) issue614_2: Hello, world!
I (528) main_task: Returned from app_main()
```

owenthewizard, 2024-03-14:
I also have this issue on esp32 (WT-ETH01), 2 works but 3 doesn't connect. `CONFIG_XTAL_FREQ_26=y` and `cargo clean; cargo run` does not resolve.

SergioGasquez, 2024-03-14:
Mind trying to use `--no-stub`? You can add it to the custom runner command in `.cargo/config.tom`:
```toml
runner = "espflash flash --monitor --no-stub"
```
Or add it in the espflash command: `espflash flash --monitor --no-stub target/riscv32imc-esp-espidf/debug/issue614-2`

I'm still confused, as I was able to flash and monitor a 26 MHz C2 with no issues. Sadly, I don't have any ESP32, so I can't really try reproducing it.

owenthewizard, 2024-03-14:
> Mind trying to use `--no-stub`?

Same result. I should mention my ESP32 uses a 40 MHz crystal, so I don't think that's related.

```
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```

SergioGasquez, 2024-03-14:
> Same result. I should mention my ESP32 uses a 40 MHz crystal, so I don't think that's related.
>
> ```
> Chip type:         esp32 (revision v1.0)
> Crystal frequency: 40MHz
> Flash size:        4MB [...]

When using the working espflash 2, does it also reports 40MHz ?? Do you know that for a fact or just guessing it from what espflash reports?

owenthewizard, 2024-03-14:
> When using the working espflash 2, does it also reports 40MHz ??

The output us from 2. 3 doesn't event get to that point.

> just guessing it from what espflash reports?

yes, this.

This is the output from 3:
```
     Running `espflash flash --monitor --no-stub target/xtensa-esp32-espidf/debug/esp32-wifi-bridge`
[2024-03-14T21:32:46Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T21:32:46Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

SergioGasquez, 2024-03-14:
> The output us from 2. 3 doesn't event get to that point.
>
>
> yes, this.
>
> This is the output from 3:
> ```
>      Running `espflash flash --monitor --no-stub [...]

Mind enabling debug logs? `RUST_LOG=debug cargo r`

owenthewizard, 2024-03-14:
@SergioGasquez https://dpaste.com/F4ANNZZ2S

xiaguangbo, 2024-03-15:
@SergioGasquez The steps you are taking to reproduce the issue are correct. In summary, after compiling once, if you add `CONFIG_XTAL_FREQ_26=y`, you must then execute `cargo clean` followed by `cargo run`. Alternatively, you can add `CONFIG_XTAL_FREQ_26=y` before the initial compilation and directly run `cargo run`. Thank you for your assistance.

There is currently an issue with the partition table. We are now only able to use 2MB, whereas previously we could use 4MB. According to the logs, the partition table has changed from:
`boot:  2 factory          factory app      00 00 00010000 003f0000`
to:
`boot:  2 factory          factory app      00 00 00010000 001f0000`

SergioGasquez, 2024-03-15:
> @SergioGasquez https://dpaste.com/F4ANNZZ2S

This looks like a different issue, can you open a new issue with details, how to reproduce it and this pastebin?

SergioGasquez, 2024-03-15:
> There is currently an issue with the partition table. We are now only able to use 2MB, whereas previously we could use 4MB. According to the logs, the partition table has changed from:

A few things to test:
- Using `--flash-size 4mb`? You can add it into your custom runner (in `.cargo/config.toml`):
```toml
runner = "espflash flash --monitor --flash-size 4mb" 
```
- Using `cargo espflash` instead of `espflash` due to its [integration with `esp-idf-sys` project](https://github.com/esp-rs/espflash/tree/main/cargo-espflash#bootloader-and-partition-table): 
```
cargo espflash flash --monitor
```
  - This would use the generated partition table from the esp-idf project, which if you have configured in `sdkconfig.defaults` will have the right flash size
-  Use a custom partition table and pass it with `--partition-table <path>` arg

xiaguangbo, 2024-03-15:
@SergioGasquez 
`runner = "espflash flash --monitor --flash-size 4mb" ` is ok.
`cargo espflash flash --monitor` is no.

How to configure partition tables in `sdkconfig.defaults`

---

`cargo espflash flash --monitor`:
```
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo espflash erase-flash
[2024-03-15T08:52:20Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-15T08:52:20Z INFO ] Connecting...
[2024-03-15T08:52:21Z INFO ] Using flash stub
[2024-03-15T08:52:22Z INFO ] Erasing Flash...
[2024-03-15T08:52:23Z INFO ] Flash has been erased!
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo espflash flash --monitor
[2024-03-15T08:52:27Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-15T08:52:27Z INFO ] Connecting...
[2024-03-15T08:52:27Z INFO ] Using flash stub
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.23s
Chip type:         esp32c2 (revision v1.0)
Crystal frequency: 26 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       34:98:7a:0f:6f:b0
Using 26MHz bootloader
App/part. size:    484,784/2,031,616 bytes, 23.86%
[00:00:01] [========================================]      12/12      0x0                                                                                            
[00:00:00] [========================================]       1/1       0x8000                                                                                         
[00:00:12] [========================>               ]     132/219     0x10000
```

SergioGasquez, 2024-03-15:
> runner = "espflash flash --monitor --flash-size 4mb" ` is ok.

Glad that we have a workarround!

> `cargo espflash flash --monitor` is not
This is because the partition table generated in your project does not have a 4mb flash size. If you update it, it will be used and should work fine.

> How to configure partition tables in `sdkconfig.defaults`

https://docs.espressif.com/projects/esp-idf/en/stable/esp32c2/api-reference/kconfig.html#config-partition-table-type

> I tested it once before, manually specifying `target/riscv32imc-esp-espidf/release/bootloader.bin` and `target/riscv32imc-espidf/release/partition-table.bin`, and the log showed that the partition table's`factory`became 1MB

If you updated your esp-idf project to make a partition-table of 4mb, `cargo espflash flash` would use it and should work. Same goes for `espflash` specifying the `--partition-table` arg.

xiaguangbo, 2024-03-15:
@SergioGasquez 

./partitions_singleapp.csv:
```
nvs,      data, nvs,     0x9000,        0x6000,
phy_init, data, phy,     0xf000,        0x1000,
factory,  app,  factory, 0x10000,       0x3f0000,
```

runner = "espflash flash --monitor --partition-table=partitions_singleapp.csv"` is ok.

Why ESPFlash V2 works directly. I think it's troublesome to change the documents.
espflash is best able to automatically identify, default to use the maximum space

closed 2024-08-08 by xiaguangbo