esp-rs/espflash #616: v3 Error while connecting to esp32

owenthewizard, 2024-03-15:
Flashing a WT32-ETH01, v2 flashes fine and reports:
```
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
```
v3 does not connect:
```
     Running `espflash flash --monitor --no-stub target/xtensa-esp32-espidf/debug/esp32-wifi-bridge`
[2024-03-14T21:32:46Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-03-14T21:32:46Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
Debug logs: https://dpaste.com/F4ANNZZ2S

SergioGasquez, 2024-03-25:
Sorry for the inconvenience, the `dpaste` looks like it has expired, mind pasting the logs here?  Have you tried the suggestion of pressing the buttons?

birkenfeld, 2024-03-25:
I had the same problem - turns out since I don't have RTS/DTR pins connected and pulled IO0 low myself, I need `-b no-reset` in espflash v3.

SergioGasquez, 2024-03-25:
> I had the same problem - turns out since I don't have RTS/DTR pins connected and pulled IO0 low myself, I need `-b no-reset` in espflash v3.

I guess your scenario was different, since using espflash v2 didn't work either, right?

birkenfeld, 2024-03-25:
> I guess your scenario was different, since using espflash v2 didn't work either, right?

No, it worked fine. AFAICS v2 doesn't have a before-reset option, so probably it didn't try to do the reset itself?

SergioGasquez, 2024-03-25:
> No, it worked fine. AFAICS v2 doesn't have a before-reset option, so probably it didn't try to do the reset itself?

Thats correct, mind trying this @birkenfeld?

owenthewizard, 2024-03-25:
> > No, it worked fine. AFAICS v2 doesn't have a before-reset option, so probably it didn't try to do the reset itself?
> 
> Thats correct, mind trying this @owenthewizard?

Confirmed, adding `--before no-reset`resolves the issue. My TTL doesn't have RTS/CTS pins, I'm only using TX, RX, 5V, and GND.

birkenfeld, 2024-03-26:
While mine *does* have RTS and CTS, it doesn't have DTR which I understand would be necessary to use the auto-reset (@SergioGasquez ?) Can there be a mode added to pull IO0 down with RTS?

birkenfeld, 2024-03-26:
Scratch that, there also needs to be one for actually resetting the board...

timboldt, 2024-05-07:
To add to this, I have an Adafruit Feather ESP32 V2 (ESP32­ Pico ­Mini 02).

- esptool works flawlessly
- espflash 3.0.0 **fails** almost every single time (about 95% failure rate)
- espflash 2.1.0 works flawlessly

P.S. For any other Rust ESP32 newbie's having this issue, you can install 2.1.0 like this `cargo install espflash@2.1.0`.

hydra, 2024-05-09:
note that running `espflash -h` doesn't show you the arguments so you may be unaware of what arguments you can specify:

```
$ espflash -h
A command-line tool for flashing Espressif devices

Usage: espflash.exe <COMMAND>

Commands:
  board-info       Print information about a connected target device
  completions      Generate completions for the given shell
  erase-flash      Erase Flash entirely
  erase-parts      Erase specified partitions
  erase-region     Erase specified region
  flash            Flash an application in ELF format to a connected target device
  monitor          Open the serial monitor without flashing the connected target device
  partition-table  Convert partition tables between CSV and binary format
  read-flash       Read SPI flash content
  save-image       Generate a binary application image and save it to a local disk
  write-bin        Write a binary file to a specific address in a target device's flash
  checksum-md5     Calculate the MD5 checksum of the given region
  help             Print this message or the help of the given subcommand(s)

Options:
  -h, --help     Print help
  -V, --version  Print version
```

however, if you go `espflash monitor -h` you'll see more arguments applicable to the `monitor` command:

```
$ espflash monitor -h
Open the serial monitor without flashing the connected target device

Usage: espflash.exe monitor [OPTIONS]

Options:
  -a, --after <AFTER>            Reset operation to perform after connecting to the target [default:
                                 hard-reset] [possible values: hard-reset, no-reset,
                                 no-reset-no-stub]
  -B, --baud <BAUD>              Baud rate at which to communicate with target device [env:
                                 ESPFLASH_BAUD=]
  -b, --before <BEFORE>          Reset operation to perform before connecting to the target
                                 [default: default-reset] [possible values: default-reset, no-reset,
                                 no-reset-no-sync, usb-reset]
  -c, --chip <CHIP>              Target device [possible values: esp32, esp32c2, esp32c3, esp32c6,
                                 esp32h2, esp32p4, esp32s2, esp32s3]
  -C, --confirm-port             Require confirmation before auto-connecting to a recognized device
      --list-all-ports           List all available ports
      --no-stub                  Do not use the RAM stub for loading
  -p, --port <PORT>              Serial port connected to target device [env: ESPFLASH_PORT=]
  -e, --elf <FILE>               Optional file name of the ELF image to load the symbols from
      --non-interactive          Avoids asking the user for interactions like resetting the device
  -L, --log-format <LOG_FORMAT>  Logging format [default: serial] [possible values: defmt, serial]
  -h, --help                     Print help (see more with '--help')
  -V, --version                  Print version
```

same applies to the `flash` command:

```
$ espflash flash -h
Flash an application in ELF format to a connected target device

Usage: espflash.exe flash [OPTIONS] <IMAGE>

Arguments:
  <IMAGE>  ELF image to flash

Options:
[... 47 lines cut ...]
      --ram
          Load the application to RAM instead of Flash
      --no-verify
          Don't verify the flash contents after flashing
      --no-skip
          Don't skip flashing of parts with matching checksum
  -h, --help
          Print help (see more with '--help')
  -V, --version
          Print version
```

I didn't spot this in the documentation and ended up at this issue, I too have my own custom board with only IO0, TX, RX, 5V, GND and was getting:

```
Error: espflash::connection_failed  

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

Using `espflash monitor --before no-reset` allowed be to connect.

Hope this helps.

SergioGasquez, 2025-02-19:
Closing this as `--before no-reset` solves the issue when not using RTS and CTS pins.

closed 2025-02-19 by SergioGasquez