esp-rs/espflash #619: invalid stub handshake response received

sakuwa, 2024-03-25:
Hello
i got the following issue:
```
[2024-03-25T15:55:55Z INFO ] Serial port: 'COM5'
[2024-03-25T15:55:55Z INFO ] Connecting...
[2024-03-25T15:55:57Z INFO ] Using flash stub
Error:   x Error while connecting to device
  `-> Invalid stub handshake response received

```

i can connect with py -m esptool -p .... to my sparkfun esp32 thing board.

I use this instruction guide to flash the device: https://github.com/beeb/coffee-scale-app/blob/main/rs/README.md

Also i set the baudrate to a lower one (115200) in the espflash.toml without success. 

I tried to find with google the issue with the handshake and thats why i set the baudrate to a lower part. 
What did i wrong or miss?

beeb, 2024-03-25:
Can confirm, same issue here (windows, cargo-espflash 3.0.0)

SergioGasquez, 2024-03-26:
Hi! Mind trying to use the `--no-stub` flag?

> Note that the board has a 26MHz crystal and CONFIG_XTAL_FREQ_26=y is set in the sdkconfig.

Correct, more information https://docs.esp-rs.org/book/troubleshooting/std.html#wrong-xtal-frequency

sakuwa, 2024-03-26:
yeah after some more try and errors and short reading (stub is a kind of a bootloader?) right?

with this flag --no-stub is successfull but i was very confused.

@beeb yeah  i am struggeling a litte bit but i after some breaks one step after another.    But great project so far. :)

SergioGasquez, 2024-03-26:
Could you try using this branch https://github.com/SergioGasquez/espflash/tree/feat/esp32-26mhz?
```
cargo install cargo-espflash --git https://github.com/SergioGasquez/espflash --branchfeat/esp32-26mhz
 ```

sakuwa, 2024-03-26:
i got a few timeout notification but it seems to work. No handshake error or need to set the --no-stub flag

beeb, 2024-03-26:
Hey @SergioGasquez thanks for looking into it. I can confirm that with the stable release and `--no-stub` the flashing works.

However this is what I get with your branch every time I try:

```
[2024-03-26T19:18:38Z INFO ] Serial port: 'COM3'
[2024-03-26T19:18:38Z INFO ] Connecting...
[2024-03-26T19:18:38Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command
```

sakuwa, 2024-03-27:
@SergioGasquez 
I tried to make a log `cargo espflash flash --release -L serial --monitor` but i don´t find a file or something else. Sorry i´m a beginner in that case.

SergioGasquez, 2024-04-02:
> I tried to make a log cargo espflash flash --release -L serial --monitor

Logs are not enabled with the `-L` arguments. Logs are enabled by `RUST_LOG=debug cargo espflash flash --release --monitor`

> However this is what I get with your branch every time I try.

Might share debug logs? Although I think I may know what's happening: The stubs are sending the sync message but espflash is not able to read it properly (different baudrate)

SergioGasquez, 2024-04-02:
Just pushed a commit https://github.com/SergioGasquez/espflash/commit/d86a622250dee85bdf817fb7a86b300889033b78 to the branch, could you guys reinstall it and test it with logs enabled:
```
cargo install cargo-espflash --git https://github.com/SergioGasquez/espflash --branch feat/esp32-26mhz
RUST_LOG=debug cargo espflash flash --release --monitor
```

beeb, 2024-04-02:
Thanks @SergioGasquez , here are the logs from my side (I shortened the data sequences for brevity):

```
[2024-04-02T17:13:57Z DEBUG] Flash(
        FlashArgs {
            build_args: BuildArgs {
                bin: None,
                example: None,
                features: None,
                frozen: false,
                locked: false,
                package: None,
                release: true,
                target: None,
                target_dir: None,
                unstable: None,
                flash_config_args: FlashConfigArgs {
                    flash_freq: None,
                    flash_mode: None,
                    flash_size: None,
                },
            },
            connect_args: ConnectArgs {
                after: HardReset,
                baud: None,
                before: DefaultReset,
                chip: None,
                confirm_port: false,
                list_all_ports: false,
                no_stub: false,
                port: None,
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
                log_format: Serial,
                min_chip_rev: 0,
                monitor: true,
                monitor_baud: None,
                partition_table: None,
                target_app_partition: None,
                partition_table_offset: None,
                ram: false,
                no_verify: false,
                no_skip: false,
            },
        },
    )
[2024-04-02T17:13:57Z DEBUG] connecting to crates.io:443 at 13.224.103.50:443
[2024-04-02T17:13:57Z DEBUG] No cached session for DnsName("crates.io")
[2024-04-02T17:13:57Z DEBUG] Not resuming any session
[2024-04-02T17:13:57Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2024-04-02T17:13:57Z DEBUG] Not resuming
[2024-04-02T17:13:57Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2024-04-02T17:13:57Z DEBUG] ALPN protocol is None
[2024-04-02T17:13:57Z DEBUG] created stream: Stream(RustlsStream)
[2024-04-02T17:13:57Z DEBUG] sending request GET https://crates.io/api/v1/crates/cargo-espflash/versions
[2024-04-02T17:13:57Z DEBUG] writing prelude: GET /api/v1/crates/cargo-espflash/versions HTTP/1.1
    Host: crates.io
    User-Agent: ureq/2.9.6
    Accept: */*
    accept-encoding: gzip
[2024-04-02T17:13:57Z DEBUG] Chunked body in response
[2024-04-02T17:13:57Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/cargo-espflash/versions
[2024-04-02T17:13:57Z DEBUG] dropping stream: Stream(RustlsStream)
[2024-04-02T17:13:57Z DEBUG] Config: Config {
        baudrate: None,
        bootloader: None,
        connection: Connection {
            serial: None,
        },
        partition_table: None,
        usb_device: [
            UsbDevice {
                vid: 1027,
                pid: 24597,
            },
        ],
        save_path: "C:\\Users\\valentin\\AppData\\Roaming\\esp\\espflash\\config\\espflash.toml",
    }
[2024-04-02T17:13:57Z INFO ] Serial port: 'COM3'
[2024-04-02T17:13:57Z INFO ] Connecting...
[2024-04-02T17:13:57Z DEBUG] Using Classic reset strategy with delay of 50ms
[2024-04-02T17:13:57Z DEBUG] Boot Mode: 0x3
[2024-04-02T17:13:57Z DEBUG] Download Mode: false
[2024-04-02T17:13:57Z DEBUG] Writing command: Sync
[2024-04-02T17:13:57Z DEBUG] Writing command: Sync
[2024-04-02T17:13:58Z DEBUG] Writing command: Sync
[2024-04-02T17:13:58Z DEBUG] Writing command: Sync
[2024-04-02T17:13:58Z DEBUG] Writing command: Sync
[2024-04-02T17:13:58Z DEBUG] Failed to reset, error Connection(
        WrongBootMode(
            "0x3",
        ),
    ), retrying
[2024-04-02T17:13:58Z DEBUG] Using Classic reset strategy with delay of 500ms
[2024-04-02T17:13:58Z DEBUG] Boot Mode: 0x3
[2024-04-02T17:13:58Z DEBUG] Download Mode: false
[2024-04-02T17:13:58Z DEBUG] Writing command: Sync
[2024-04-02T17:13:59Z DEBUG] Writing command: Sync
[2024-04-02T17:13:59Z DEBUG] Writing command: Sync
[2024-04-02T17:13:59Z DEBUG] Writing command: ReadReg { address: 1073745920 }
[2024-04-02T17:13:59Z INFO ] Using flash stub
[2024-04-02T17:13:59Z DEBUG] Loading flash stub for chip: Esp32
[2024-04-02T17:13:59Z DEBUG] Write 12356 byte stub text
[2024-04-02T17:13:59Z DEBUG] Writing command: MemBegin { size: 12356, blocks: 3, block_size: 6144, offset: 1074266112, supports_encryption: false }
[2024-04-02T17:13:59Z DEBUG] Writing command: MemData { data: [0, ... 129], pad_to: 4, pad_byte: 0, sequence: 0 }
[2024-04-02T17:13:59Z DEBUG] Writing command: MemData { data: [162, ... 0], pad_to: 4, pad_byte: 0, sequence: 1 }
[2024-04-02T17:14:00Z DEBUG] Writing command: MemData { data: [230, ... 0], pad_to: 4, pad_byte: 0, sequence: 2 }
[2024-04-02T17:14:00Z DEBUG] Write 744 byte stub data
[2024-04-02T17:14:00Z DEBUG] Writing command: MemBegin { size: 744, blocks: 1, block_size: 6144, offset: 1073414144, supports_encryption: false }
[2024-04-02T17:14:00Z DEBUG] Writing command: MemData { data: [86, ... 0], pad_to: 4, pad_byte: 0, sequence: 0 }
[2024-04-02T17:14:00Z DEBUG] Finish stub write
[2024-04-02T17:14:00Z DEBUG] Writing command: MemEnd { no_entry: false, entry: 1074274836 }
[2024-04-02T17:14:00Z DEBUG] Stub written!
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command
```

SergioGasquez, 2024-04-03:
Just to confirm, these logs were using the latest commit of my branch? Looks like it's retrying lots of times the [sync command](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/serial-protocol.html#initial-synchronisation) (probably because it's not able to read it due to the different baud rate). 

Also, could you try using the `--before no-reset` argument?

beeb, 2024-04-07:
Yes this is on commit `d86a622250dee85bdf817fb7a86b300889033b78`. Here's the result with the `--before no-reset` argument:

```
[2024-04-07T13:55:54Z DEBUG] Flash(
        FlashArgs {
            build_args: BuildArgs {
                bin: None,
                example: None,
                features: None,
                frozen: false,
                locked: false,
                package: None,
                release: true,
                target: None,
                target_dir: None,
                unstable: None,
                flash_config_args: FlashConfigArgs {
                    flash_freq: None,
                    flash_mode: None,
                    flash_size: None,
                },
            },
            connect_args: ConnectArgs {
                after: HardReset,
                baud: None,
                before: NoReset,
                chip: None,
                confirm_port: false,
                list_all_ports: false,
                no_stub: false,
                port: None,
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
                log_format: Serial,
                min_chip_rev: 0,
                monitor: true,
                monitor_baud: None,
                partition_table: None,
                target_app_partition: None,
                partition_table_offset: None,
                ram: false,
                no_verify: false,
                no_skip: false,
            },
        },
    )
[2024-04-07T13:55:54Z DEBUG] connecting to crates.io:443 at 13.224.103.50:443
[2024-04-07T13:55:54Z DEBUG] No cached session for DnsName("crates.io")
[2024-04-07T13:55:54Z DEBUG] Not resuming any session
[2024-04-07T13:55:54Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2024-04-07T13:55:54Z DEBUG] Not resuming
[2024-04-07T13:55:54Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2024-04-07T13:55:54Z DEBUG] ALPN protocol is None
[2024-04-07T13:55:54Z DEBUG] created stream: Stream(RustlsStream)
[2024-04-07T13:55:54Z DEBUG] sending request GET https://crates.io/api/v1/crates/cargo-espflash/versions
[2024-04-07T13:55:54Z DEBUG] writing prelude: GET /api/v1/crates/cargo-espflash/versions HTTP/1.1
    Host: crates.io
    User-Agent: ureq/2.9.6
    Accept: */*
    accept-encoding: gzip
[2024-04-07T13:55:54Z DEBUG] Chunked body in response
[2024-04-07T13:55:54Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/cargo-espflash/versions
[2024-04-07T13:55:54Z DEBUG] dropping stream: Stream(RustlsStream)
[2024-04-07T13:55:54Z DEBUG] Config: Config {
        baudrate: None,
        bootloader: None,
        connection: Connection {
[... 51 lines cut ...]
[2024-04-07T13:55:57Z DEBUG] Failed to reset, error Connection(
        ConnectionFailed,
    ), retrying
[2024-04-07T13:55:57Z DEBUG] Writing command: Sync
[2024-04-07T13:55:57Z DEBUG] Writing command: Sync
[2024-04-07T13:55:57Z DEBUG] Writing command: Sync
[2024-04-07T13:55:57Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Failed to reset, error Connection(
        ConnectionFailed,
    ), retrying
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Writing command: Sync
[2024-04-07T13:55:58Z DEBUG] Failed to reset, error Connection(
        ConnectionFailed,
    ), retrying
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
  ```

SergioGasquez, 2024-05-31:
Hi! Could you try with the new `espflash` version? I guess this is a stub issue, so it probably should be transferred