esp-rs/espflash #620: CORE Board ESP32 Super Mini - Failed to connect to on-device flash

Kezii, 2024-03-25:
espflash fails with the esp32-c3 super mini board

[image]

```
[...]
[2024-03-25T22:52:07Z DEBUG] Finish stub write
[2024-03-25T22:52:07Z DEBUG] Writing command: MemEnd { no_entry: false, entry: 1077411840 }
[2024-03-25T22:52:07Z DEBUG] Stub written!
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1073745920 }
[2024-03-25T22:52:07Z DEBUG] Re-detected chip: Esp32c3
[2024-03-25T22:52:07Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2024-03-25T22:52:07Z DEBUG] Writing command: SpiAttachStub { spi_params: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 } }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620960 }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2415919104, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048351, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620968, value: 23, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610621016, value: 0, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620928, value: 262144, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610621016 }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2147483648, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048192, mask: None }
[2024-03-25T22:52:07Z DEBUG] Flash detect failed
[2024-03-25T22:52:07Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 6, q: 17, d: 8, hd: 11, cs: 16 }
[2024-03-25T22:52:07Z DEBUG] Writing command: SpiAttachStub { spi_params: SpiAttachParams { clk: 6, q: 17, d: 8, hd: 11, cs: 16 } }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620960 }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2415919104, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048351, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620968, value: 23, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610621016, value: 0, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620928, value: 262144, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-25T22:52:07Z DEBUG] Writing command: ReadReg { address: 1610621016 }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2147483648, mask: None }
[2024-03-25T22:52:07Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048192, mask: None }
[2024-03-25T22:52:07Z DEBUG] Flash detect failed
[2024-03-25T22:52:07Z DEBUG] SPI flash autodetection failed
Error: espflash::flash_connect

  × Failed to connect to on-device flash

```

SergioGasquez, 2024-03-26:
Can you try using `--no-stub`?

Kezii, 2024-03-26:
that does not seem to make a difference

```
[...]
[2024-03-26T11:07:33Z INFO ] Serial port: '/dev/ttyACM0'
[2024-03-26T11:07:33Z INFO ] Connecting...
[2024-03-26T11:07:33Z DEBUG] Using UsbJtagSerial reset strategy
[2024-03-26T11:07:33Z DEBUG] Boot Mode: 0x5
[2024-03-26T11:07:33Z DEBUG] Download Mode: false
[2024-03-26T11:07:33Z DEBUG] Writing command: Sync
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1073745920 }
[2024-03-26T11:07:33Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2024-03-26T11:07:33Z DEBUG] Writing command: SpiAttach { spi_params: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 } }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620960 }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2415919104, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048351, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620968, value: 23, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610621016, value: 0, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620928, value: 262144, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610621016 }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2147483648, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048192, mask: None }
[2024-03-26T11:07:33Z DEBUG] Flash detect failed
[2024-03-26T11:07:33Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 6, q: 17, d: 8, hd: 11, cs: 16 }
[2024-03-26T11:07:33Z DEBUG] Writing command: SpiAttach { spi_params: SpiAttachParams { clk: 6, q: 17, d: 8, hd: 11, cs: 16 } }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620960 }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2415919104, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048351, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620968, value: 23, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610621016, value: 0, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620928, value: 262144, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-03-26T11:07:33Z DEBUG] Writing command: ReadReg { address: 1610621016 }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2147483648, mask: None }
[2024-03-26T11:07:33Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048192, mask: None }
[2024-03-26T11:07:33Z DEBUG] Flash detect failed
[2024-03-26T11:07:33Z DEBUG] SPI flash autodetection failed
Error: espflash::flash_connect

  × Failed to connect to on-device flash
```

Kezii, 2024-03-29:
apparently, they assembled those board wrong, the chip should be an ESP32-C3FH4, but it's a normal esp32 without the flash

this is an hardware error on who assembled those eval boards, espflash is behaving as expected

the working part should look like this

[image]

closed 2024-03-29 by Kezii