esp-rs/espflash #621: [ESP32-S2] Not able to monitor device after flashing - device not booting properly

italicmew, 2024-03-25:
I am currently attempting to flash my ESP-32-s2 device using the `esp-template` and `espflash` tool. However, I am facing issues in getting the compiled binary to run successfully on my device. My initial problem was with resetting the device, as it was not resetting as expected compared to esptool.py. After several attempts, I discovered that setting `-b no-reset` was necessary to initiate the flashing process successfully. 

Even after flashing, my device does not boot up or display anything. I noticed that when I manually reset the device, the serial ports disappear.

```bash
$ cargo generate esp-rs/esp-template
$ CRATE_CC_NO_DEFAULTS=1 cargo build
$ cargo espflash flash -b no-reset --no-stub --monitor

[2024-03-25T22:53:49Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-03-25T22:53:49Z INFO ] Detected 2 serial ports
[2024-03-25T22:53:49Z INFO ] Ports which match a known common dev board are highlighted
[2024-03-25T22:53:49Z INFO ] Please select a port
[2024-03-25T22:53:50Z INFO ] Serial port: '/dev/cu.usbmodem01'
[2024-03-25T22:53:50Z INFO ] Connecting...
    Finished dev [optimized + debuginfo] target(s) in 0.23s
Chip type:         esp32s2 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, Embedded Flash 4MB, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
MAC address:       84:f7:03:e8:6c:5e
App/part. size:    396,576/1,048,576 bytes, 37.82%
[2024-03-25T22:53:50Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2024-03-25T22:53:50Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-03-25T22:53:51Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-03-25T22:53:51Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Error:   × Broken pipe
```

<details><summary>Click to expand Full logs</summary>

```
 RUST_LOG=debug cargo espflash flash -b no-reset --no-stub --monitor
[2024-03-25T23:07:26Z DEBUG] Flash(
        FlashArgs {
            build_args: BuildArgs {
                bin: None,
                example: None,
                features: None,
                frozen: false,
                locked: false,
                package: None,
                release: false,
                target: None,
                target_dir: None,
                unstable: None,
                flash_config_args: FlashConfigArgs {
                    flash_freq: None,
                    flash_mode: None,
                    flash_size: None,
                },
            },
            connect_args: ConnectArgs {
                after: HardReset,
                baud: None,
                before: NoReset,
                chip: None,
                confirm_port: false,
                list_all_ports: false,
                no_stub: true,
                port: None,
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
[... 187 lines cut ...]
Writing at 0x0003f9d6... (83 %)
Writing at 0x000404c5... (85 %)
Writing at 0x00040ffa... (87 %)
Writing at 0x00041b1d... (88 %)
Writing at 0x000426de... (90 %)
Writing at 0x000432b6... (91 %)
Writing at 0x0004439b... (93 %)
Writing at 0x000451bb... (95 %)
Writing at 0x00045b1c... (96 %)
Writing at 0x000466fb... (98 %)
Writing at 0x0004742a... (100 %)
Wrote 227296 bytes (125420 compressed) at 0x00010000 in 1.9 seconds (effective 944.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 388.7 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
Done
</details>

Os: MacOS
espflash Version: 3.0.1-dev

italicmew, 2024-03-25:
Running the same thing on a normal esp-32 seems to work fine with exactly same setup.

<sumary>ESP32 flash logs</sumary><details>

```log
cargo espflash flash --monitor
[2024-03-25T23:32:46Z INFO ] Detected 2 serial ports
[2024-03-25T23:32:46Z INFO ] Ports which match a known common dev board are highlighted
[2024-03-25T23:32:46Z INFO ] Please select a port
[2024-03-25T23:32:48Z INFO ] Serial port: '/dev/cu.usbserial-0001'
[2024-03-25T23:32:48Z INFO ] Connecting...
[2024-03-25T23:32:48Z INFO ] Using flash stub
    Finished dev [optimized + debuginfo] target(s) in 0.28s
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, Coding Scheme None
MAC address:       24:0a:c4:12:81:08
App/part. size:    412,432/4,128,768 bytes, 9.99%
[2024-03-25T23:32:50Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2024-03-25T23:32:50Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-03-25T23:32:50Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-03-25T23:32:51Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
0x3fff0030 - esp_wifi::HEAP_DATA
    at ??:??
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - core::ptr::read_volatile
    at /Users/fabricio/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1583
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 07:48:23
I (32) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=11d4ch ( 73036) map
I (132) esp_image: segment 1: paddr=00021d74 vaddr=3ffb0000 size=00e10h (  3600) load
I (134) esp_image: segment 2: paddr=00022b8c vaddr=3ffb8d24 size=00190h (   400) load
I (138) esp_image: segment 3: paddr=00022d24 vaddr=40080000 size=0cc7ch ( 52348) load
I (168) esp_image: segment 4: paddr=0002f9a8 vaddr=00000000 size=00670h (  1648) 
I (169) esp_image: segment 5: paddr=00030020 vaddr=400d0020 size=44ac8h (281288) map
I (281) boot: Loaded app from partition at offset 0x10000
I (281) boot: Disabling RNG early entropy source...
INFO - Logger is setup
Hello world!
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 50, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
Loop...
Loop...
Loop...
Loop...
```

</details>

SergioGasquez, 2024-03-26:
- Just to confirm, you are pressing Ctrl + R after the flashing has completed, right?
- `esptool` is only flashing, and `espflash` also flashes successfully.
  - `esptool` is using `--before=no_reset --after=no_reset`
- Have you tried to use `espflash monitor`? 
  - Might also be useful to try the  `-b no-reset-no-sync` flag

Vollbrecht, 2024-03-26:
I am not sure how its done in the no_std template, but the s2 is a bit special - e.g it can use usb-otg **or** usb-cdc-acm, but not both at the same time. So if you are flashing it and your binary doesn't setup usb-otg and your are not falling back to manually enabling usb-cdc the board just simply will not be there after it is started.

Vollbrecht, 2024-03-26:
on esp-idf-template based project you need to set usb-cdc manually to be used via a sdkconfig but not sure how that all works on the no_std site.

italicmew, 2024-03-26:
@Vollbrecht Are you referring to [this esp usb doc](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-guides/usb-otg-console.html), right?

I was able to make the `esp-idf-template` work setting `CONFIG_ESP_CONSOLE_USB_CDC=y` in the `sdkconfig.defaults` file.

```terminal
cargo espflash monitor -b no-reset --no-stub
[2024-03-26T20:46:24Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-03-26T20:46:24Z INFO ] Detected 2 serial ports
[2024-03-26T20:46:24Z INFO ] Ports which match a known common dev board are highlighted
[2024-03-26T20:46:24Z INFO ] Please select a port
[2024-03-26T20:46:26Z INFO ] Serial port: '/dev/tty.usbmodem12301'
[2024-03-26T20:46:26Z INFO ] Connecting...
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

I (38) boot: ESP-IDF v5.1.3 2nd stage bootloader
I (38) boot: compile time Mar 26 2024 17:34:50
I (38) boot: chip revision: v0.0
I (39) boot.esp32s2: SPI Speed      : 40MHz
I (39) boot.esp32s2: SPI Mode       : DIO
I (39) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (39) boot: Partition Table:
I (39) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (42) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (43) boot:  2 factory          factory app      00 00 00010000 00100000
I (44) boot: End of partition table
I (44) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1e060h (122976) map
I (77) esp_image: segment 1: paddr=0002e088 vaddr=3ffbe3d0 size=01894h (  6292) load
I (80) esp_image: segment 2: paddr=0002f924 vaddr=40024000 size=006f4h (  1780) load
I (81) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=46250h (287312) map
I (158) esp_image: segment 4: paddr=00076278 vaddr=400246f4 size=09cdch ( 40156) load
I (176) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (227) main_task: Started on CPU0
I (227) main_task: Calling app_main()
I (2227) blink_idf: Hello, world!
I (4227) blink_idf: Hello, world!
```

I am still unable to make this work with the `no_std` template. Do you have any code that I could review to understand how we can resolve this problem? I noticed your comments on similar issues in other repositories and I'm wondering if I can help with a solution.

SergioGasquez, 2024-03-26:
We have an usb cdc example in esp-hal, see https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/usb_serial.rs

italicmew, 2024-03-29:
After spending some time on this, I believe I've grasped the core of the issue. The USB-CDC initialization occurs in the bootloader if `CONFIG_ESP_CONSOLE_USB_CDC` is set, as seen [here](https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/bootloader_support/src/bootloader_console.c#L113). It's possible that the espflash bootloader was compiled without this configuration enabled.

Moreover, the flash stub provided doesn't seem to function as expected either, likely due to the absence of USB OTG support.

The example provided by @SergioGasquez seems to partially work for what is intended. However, it does not initialize the console itself, and I don't see any logs after reset, I still not able to see bootloader nor application logs.
Also, when flashed this example onto my esp32s2, the USB serial file descriptor appears on my macOS, but unfortunately, `espflash monitor` fails to communicate with it, leading to crashes. 

```terminal
$ cargo espflash monitor -b no-reset
[2024-03-29T18:52:45Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-03-29T18:52:45Z INFO ] Detected 2 serial ports
[2024-03-29T18:52:45Z INFO ] Ports which match a known common dev board are highlighted
[2024-03-29T18:52:45Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2024-03-29T18:52:48Z INFO ] Serial port: '/dev/cu.usbmodem12301'
[2024-03-29T18:52:48Z INFO ] Connecting...
[2024-03-29T18:52:48Z DEBUG] Writing command: Sync
Error:   × Main thread panicked.
  ├─▶ at /Users/xxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/espflash-3.0.0/src/connection/mod.rs:363:30
  ╰─▶ called `Result::unwrap()` on an `Err` value: ParseIntError { kind: InvalidDigit }
  help: set the `RUST_BACKTRACE=1` environment variable to display a backtrace.
```

It appears that `idf.py monitor` is functioning correctly as it echoes back what I send in uppercase. However, it seems to be a separate issue unrelated to the current discussion.

italicmew, 2024-03-31:
Alright, I've decided to take a more proactive approach and have connected a USB-Serial adapter to the UART0 pins. As expected, all the console logs that were meant to be redirected to USB are present.

Moreover, when I flash a C esp-idf example with USB CDC enabled, all the console logs are successfully redirected to USB.

Now, here's my idea: What if I flash the bootloader generated in the C esp-idf project into my Rust esp-idf project using the esp flash --bootloader argument?

I attempted this approach using the following command:

`cargo espflash flash -b no-reset --bootloader my_custom_bootloader/bootloader.bin --no-stub -m dio -s 4mb -f 80mhz`

However, I couldn't get it to work as expected. The first-stage bootloader fails to the new bootloader; it simply does not print any logs in UART0 related to second stage bootloader, not creates any new USB file descriptor. Here's the output I'm getting from UART0:

```
<RESET>

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe61d0,len:0x1808
load:0x4004b000,len:0x4
load:0x4004b004,len:0xb10
load:0x4004f000,len:0x3200
entry 0x4004b1dc

<JUNK BEING PRINT...>

```

Here is the output of the esp-idf C project flashed using esptool:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe61d0,len:0x1808
load:0x4004b000,len:0x4
load:0x4004b004,len:0xb10
load:0x4004f000,len:0x3200
entry 0x4004b1dc

<no junk>
<new usb file descriptor available>
```

Here is the output of rust project flashed with the espflash bootloader:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6108,len:0x17dc
load:0x4004b000,len:0x4
load:0x4004b004,len:0xab4
load:0x4004f000,len:0x3160
entry 0x4004b1c4
I (26) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (26) boot: compile time Jun  7 2023 08:05:03
I (28) boot: chip revision: v0.0
I (31) boot.esp32s2: SPI Speed      : 40MHz
I (36) boot.esp32s2: SPI Mode       : DIO
I (41) boot.esp32s2: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=03804h ( 14340) map
I (101) esp_image: segment 1: paddr=0001382c vaddr=3ffb3444 size=00008h (     8) load
I (106) esp_image: segment 2: paddr=0001383c vaddr=40022000 size=01444h (  5188) load
I (116) esp_image: segment 3: paddr=00014c88 vaddr=00000000 size=0b390h ( 45968) 
I (134) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=05260h ( 21088) map
I (141) boot: Loaded app from partition at offset 0x10000
I (141) boot: Disabling RNG early entropy source...
INFO - Logger is setup
Hello world!
```

italicmew, 2024-03-31:
@SergioGasquez Is there any special steps I need to take to compile the bootloader from esp-idf? https://github.com/esp-rs/espflash/tree/main/espflash/resources  - you mentioned that it's compiled by hand

Vollbrecht, 2024-03-31:
If you are using the esp-idf-template and build a hello-world, you can find that a corresponding bootloader is build inside the output directory. So your output contains 3 files. The bootloader.bin the partition-table.bin and the hello-world elf file

Vollbrecht, 2024-03-31:
`cargo espflash` has special code inside it so it can automatically pick up that file, while direct usage of `espflash` cannot directly locate that file and you manually need to tell espflash where it can be found.

italicmew, 2024-03-31:
@Vollbrecht Yup, I'm using the `esp-idf-template` - I compiled and generated the bootloader.bin and instructed `espflash` to use it with `--bootloader path/to/esp-idf-template/../bootloader.bin`. But that failed, see my [previous](https://github.com/esp-rs/espflash/issues/621#issuecomment-2028771631) comment.  Should I also use the partition table created by the `esp-idf-template`? I didn't use that.

italicmew, 2024-03-31:
```
cargo espflash flash -b no-reset --bootloader /Users/tiburcio/Documents/workspace/esp-idf-template/build/bootloader/bootloader.bin --no-stub -m dio -s 4mb -f 80mhz --partition-table /Users/tiburcio/Documents/workspace/esp-idf-template/build/partition_table/partition-table.bin
```

SergioGasquez, 2024-04-02:
I think you should use `--no-stub` as the current stubs do not support USB-OTG (https://github.com/esp-rs/esp-flasher-stub/issues/32) as you mentioned

italicmew, 2024-04-02:
I've been consistently using `--no-stub` in all my espflash flash commands, I'm aware of that issue with the stub. However, the challenge is the inability to access console output via USB on my developer board, which directly connects its USB to the ESP32-S2 chip.

To address this, I believe flashing a bootloader compiled with `CONFIG_ESP_CONSOLE_USB_CDC=y`, passing it using `--bootloader` to `espflash`, could resolve the issue. Unfortunately, I was not able to do it.

I'm not sure whether this is an issue or not. Is it expected behavior for console output to always redirect to the serial used for flashing the device? This seems to be the default behavior of esp-idf, if I'm not mistaken. Perhaps we could enhance the granularity of espflash arguments to enable selection of an ESP32-S2 target that utilizes USB-OTG. For that we would need a new esp32bootloader-withusb.bin and more targets to be configured, which does not sounds as good solution.

italicmew, 2024-04-02:
Side note: Quick search on google shows a mixture of devBoards. Most of my results are boards without usb-serial converts. 

[image]

SergioGasquez, 2024-04-04:
When using `cargo-espflash` you dont need to point at the bootloader generated in the output folder as this is the default behavior. The `--bootloader` arg is useful for `espflash` or when not using the generated one in `cargo-espflash`.

> This seems to be the default behavior of esp-idf(flashtool), if I'm not mistaken. 

Could you verify this, please?

> Perhaps we could enhance the granularity of espflash arguments to enable selection of an ESP32-S2 target that utilizes USB-OTG. For that we would need a new esp32bootloader-withusb.bin and more targets to be configured, which does not sounds as good solution.

I am not sure if I like the idea of adding more bootloaders, which result in a more difficult maintenance and are not really useful for most users.

jath03, 2024-04-28:
I have been experiencing this issue while trying to flash the [Flipper Zero Wi-Fi Module](https://docs.flipper.net/development/hardware/wifi-developer-board/schematics).  I have been able to flash using `espflash flash -b no-reset --no-stub`, but logging hasn't been working when I try a no_std example like [this one](https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/docs/examples.md).  The missing piece, I think is in [esp-println](https://github.com/esp-rs/esp-println/blob/main/src/lib.rs#L216), where, for `esp32s2`, it just writes data over UART.  

What I would need to do is add a feature in the esp-println crate to use USB CDC and then somehow implement that.  [The espressif docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/usb-otg-console.html) mention a USB CDC implementation in the ROM code.  Does anyone know where I could find more information about the ROM code and the functions that might be there for USB CDC?

daagaak, 2024-05-18:
> I am not sure if I like the idea of adding more bootloaders, which result in a more difficult maintenance and are not really useful for most users.

I'm a little puzzled by this stance. I have three different "hobby" ESP32-S2 based dev boards (Adafruit ESP32-S2, Feather-S2 and an Adafruit MagTag) that all rely on USB CDC for their serial connection. So those boards are ergonomically hard to use since I moved up to espflash 3.

whirm, 2025-02-07:
I'm hitting the same issue on a wemos.cc ESP32-S2 mini.

SergioGasquez, 2025-02-10:
> I'm hitting the same issue on a wemos.cc ESP32-S2 mini.

What devkit are you using? Does it work if you use `--no-stub`? Does it work if you use `-b no-reset --no-stub`?

whirm, 2025-02-10:
> What devkit are you using? Does it work if you use `--no-stub`? Does it work if you use `-b no-reset --no-stub`?

Yeah, I can flash with `-b no-reset --no-stub` &co.

The issue I'm hitting is that, with espflash V3, the bootloader doesn't enable USB CDC so monitoring is gone in boards designed around having it available. 

To make things worse, some boards like the the  [Wemos S2 Mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) that I'm using don't even expose all the pins required for JTAG.

It makes things so difficult (more so for a learning hobbyist like me) that I just switched to flashing with V2.

Tristaan, 2025-02-17:
Same issue on a WaveShare ESP32-S2 Pico and LCD.
The following logs come for me:
```
% cargo espflash monitor -b no-reset --no-stub -c esp32s2                                                                                      12:51
[2025-02-17T23:51:23Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2025-02-17T23:51:24Z INFO ] Serial port: '/dev/ttyACM0'
[2025-02-17T23:51:24Z INFO ] Connecting...
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

Error:   × Protocol error
```

After pressing `CTRL+R`.

Tristaan, 2025-02-20:
> Since we updated the stubs and improved the monitoring, can you try to install `espflash`/`cargo-espflash` from main and seeing if the issue is resolved?

Okay, now stub works, it flashes without any issues using the no-reset option, but `CTRL+R` does nothing inside monitor.

SergioGasquez, 2025-03-25:
https://github.com/esp-rs/espflash/pull/813 might also improve/fix this issue, can any of you install latest `main` and see if the issue is resolved?

closed 2025-04-28 by SergioGasquez