esp-rs/espflash #622: `write-bin` does not flash binary

TheOnAndOnlyZenomat, 2024-04-07:
Hey,
I have a binary
```
00000000: 01a0                                     ..
```
which is just a jump to `0x0`.

When I try to write this binary with
```bash
espflash write-bin 0 img.img
```
nothing gets actually written. The output looks normal
```
[2024-04-07T17:54:34Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-04-07T17:54:34Z INFO ] Connecting...
[2024-04-07T17:54:34Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       60:55:f9:c8:a7:6c
[00:00:00] [========================================]       1/1       0x10000                                                                        
[2024-04-07T17:54:36Z INFO ] Binary successfully written to flash!
```
but when reading the memory back using
```bash
espflash read-flash 0 64 out.bin
```
it is just all `0xff`.

When I try to flash the same binary with `esptool.py`
```bash
esptool.py write_flash 0 img.img
```
it works.
I can read back the memory with `espflash` and `esptool.py` and both show `01a0` at `0x0` in the flash.

Am I missing the point of the subcommand, or is this actually a bug. If the latter is the case, I'd be happy to do some debugging and testing for you.

MabezDev, 2024-04-07:
Hmm, looks like a bug. It's ignoring your input address of `0x0` and proceeding with the default application address of `0x10000`.

jordonsc, 2024-07-19:
I don't believe it's writing to the wrong location, if I have good data in the provided address, after using `espflash write-bin` that data is replaced by 0xFF bytes as if it was erased. 

The success of `write-bin` is dependent on my payload. If I change the payload slightly, it might succeed. Change again, it might fail - but it is deterministic.

closed 2025-02-26 by SergioGasquez