esp-rs/espflash #626: Default bootloader sdkconfig ?

Arlaz, 2024-04-22:
Hello 👋
I’ve been struggling to make my project work with WiFi and a custom bootloader I need.

After many tests, it appears that my code runs correctly when using your « build by hand » espflash bootloader.

Would you mind sharing what modifications you made to it ? And / or the sdkconfig.default files you used with esp-idf to build your bootloader ?

Thank you for this project !

SergioGasquez, 2024-04-23:
> I’ve been struggling to make my project work with WiFi and a custom bootloader I need.

Is it an `std` or `no_std` project? If it's an `std` I would suggest using `cargo-espflash` so it uses the bootloader/partition-table generated in your esp-idf project, see https://github.com/esp-rs/espflash/tree/main/cargo-espflash#bootloader-and-partition-table

> Would you mind sharing what modifications you made to it ? And / or the sdkconfig.default files you used with esp-idf to build your bootloader?

The bootloaders shipped with espflash (https://github.com/esp-rs/espflash/tree/main/espflash/resources) are built using the default `sdkconfig`, no changes and using an example from the ESP-IDF version that we select.

Arlaz, 2024-04-23:
It is a `no_std` project, that's why I find this behaviour weird.
I will try to make it work with a bootloader based on the same commit as yours.

Thank you for your answer !

closed 2024-04-23 by Arlaz