esp-rs/espflash #628: Adding a retry feature

dscso, 2024-04-24:
I often get the error:
```log
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command
```
After trying multiple times (3 times or so) it works. I think it would simplify my workflow a lot if I could just do something like this:
```bash
 espflash flash --retry 3 target/xtensa-esp32-none-elf/release/esp32-rust-mqtt
```
Maybe its also me doing something wrong, but I think this could be a useful addition :)

SergioGasquez, 2024-04-24:
This shouldn't happen, mind giving more details about your setup (OS, hw trying to flash, espflash version...)?

dscso, 2024-04-24:
MacOS 14.2.1 (M1)
ESP32-WROOM-32 (xtensa architecture) 4MB flash connected via USB
espflash 3.0.0

dscso, 2024-04-25:
I tried another cable but the issue stays. I don't have another chip to test yet. But if this is a problem that only I am facing, I think the feature is not necessary :).

closed 2024-04-25 by SergioGasquez