esp-rs/espflash #630: write-bin: Skip writing if binary has not changed.

jounathaen, 2024-04-26:
First: I love `espflash`! It is a great tool! Thanks for all the work!

I'm trying to optimize my flash times by putting static data into a separate partition, which works quite decent via `write-bin`. But unlike `flash`, `write-bin` does not check the MD5 sum of the section and thus does not skip if the same data is already present at that address. This prevents me from adding the `write-bin` step into the flashing script.

Desired feature:
- Add a `--skip-if-present` option to `write-bin`, that checks the md5 hash before writing and skips if it matches the data to be written.

Alternative Approaches:
- allow binary files to be flashed via `flash` as well
- Default to skipping blocks in general: https://github.com/esp-rs/espflash/issues/588

SergioGasquez, 2024-04-30:
> - Default to skipping blocks in general: https://github.com/esp-rs/espflash/issues/588

Some initial investigation showed that it was not worth to implement that feature. I would go for adding `--skip-if-present` flag or the other way, skip by default and add a `--no-skip` flag like we have in the `flash` subcommand.

harshilv17, 2026-11-09:
Hi @SergioGasquez, I'd like to work on this. Plan: add `--skip-if-present` to `write-bin` that reuses the existing flash MD5 check and skips the write if the region already matches. I'll test it on an ESP32.

SergioGasquez, 2026-11-09:
Hi @harshilv17! Feel free to proceed and work on this feature, but note that we are still not sure if we will do a major/minor/patch bump for the next release. I guess patch/minor are completely fine to accept, but if your PR contains major breaking changes, it might be on hold.

closed 2026-11-11 by SergioGasquez