esp-rs/espflash #631: Allow a partition as a target for `write-bin`

jounathaen, 2024-04-26:
Currently, `write-bin` requires the address for the data to be written. It would be nice if one could specify a partition instead. This would also allow a length check, so that partition boundaries are not violated.

jessebraham, 2025-04-01:
@JurajSadel do you still intend to work on this feature, or should I take over here?

JurajSadel, 2025-04-01:
> [@JurajSadel](https://github.com/JurajSadel) do you still intend to work on this feature, or should I take over here?

I've had an idea about how it could work but it didn't feel like a good solution (back then we wanted to refactor the codebase) so I skipped this issue. Feel free to take over if you are interested.

PR #828 (2025-04-04): Allow flashing binary to a named partition
Closes #631

This PR allows specifying a partition label for `write-bin`, in addition to a numeric address. If a partition label is passed, espflash tries to parse the provided/configured partition table, and resolve the label. This information is then used to a) check the size of the flashed binary and b) specify the address where the binary should be written.

As an alternative approach, we might read back the partition table from the device itself. Since users have the option to configure a partition table in a config file, this option would need a [... cut]
closed 2025-04-04 by SergioGasquez