esp-rs/espflash #636: espflash only works with "--list-all-ports"

trinhanhtuan, 2024-05-15:
Hi all,
I have a very strange issue while working on podman container. The `espflash` only can access `/dev/ttyACM0` if I use `--list-all-ports`. Below is illustration when try the `monitor` but the same issue happens with `flash`, `board-info`, etc. as well. Any clues? Thanks.

```
root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# id
uid=0(root) gid=0(root) groups=0(root)
root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# espflash --version
espflash 3.0.0-rc.2
root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# ls -l /dev/ttyACM0 
crw-rw----+ 1 nobody nogroup 166, 0 May 15 09:08 /dev/ttyACM0
root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# espflash monitor --port /dev/ttyACM0 --baud 9600 --before usb-reset
[2024-05-15T10:19:10Z INFO ] 🚀 A new version of espflash is available: v3.0.0
Error: espflash::serial_not_found

  x The serial port '/dev/ttyACM0' could not be found
  help: Make sure the correct device is connected to the host system

root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# espflash monitor --list-all-ports --baud 9600 --before usb-reset
[2024-05-15T10:19:23Z INFO ] 🚀 A new version of espflash is available: v3.0.0
✔ Use serial port '/dev/ttyACM0'? · yes
[2024-05-15T10:19:24Z INFO ] Serial port: '/dev/ttyACM0'
[2024-05-15T10:19:24Z INFO ] Connecting...
[2024-05-15T10:19:24Z INFO ] Using flash stub
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378a0a
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (28) boot: compile time Jun  7 2023 08:07:32
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v0.2
I (36) boot.esp32s3: Boot SPI Speed : 40MHz
I (41) boot.esp32s3: SPI Mode       : DIO
I (46) boot.esp32s3: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 007f0000
I (89) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c010020 size=01b18h (  6936) map
I (104) esp_image: segment 1: paddr=00011b40 vaddr=3fc89150 size=00004h (     4) load
I (111) esp_image: segment 2: paddr=00011b4c vaddr=40378000 size=01150h (  4432) load
I (120) esp_image: segment 3: paddr=00012ca4 vaddr=00000000 size=0d374h ( 54132) 
I (141) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=03e54h ( 15956) map
I (146) boot: Loaded app from partition at offset 0x10000
I (146) boot: Disabling RNG early entropy source...
Blinking! 0                       <---- it works
root@rustvm01:/home/ESP32/blinking-xiaos3-nostd# 
```

Best regards,
Trinh Tuan

contra-bit, 2024-05-16:
Can confirm. I have experienced the same issue.  Thanks for reporting it. 

In my case I do not need --list-all-ports but the flag   --before usb-reset for interacting with the esp32c3

```sh
espflash monitor  --before usb-reset
```

SpencerTorres, 2024-05-16:
Tested with old ESP32, couldn't get anything to work. Turns out I had to hold the RESET button until I saw the `Connecting...` message, then it was able to connect after releasing button. PuTTY is able to connect without this step.

None of these other `--before` options worked, wondering if this is a hardware problem or something with espflash?

trinhanhtuan, 2024-05-17:
> Can confirm. I have experienced the same issue. Thanks for reporting it.
> 
> In my case I do not need --list-all-ports but the flag --before usb-reset for interacting with the esp32c3
> 
> ```shell
> espflash monitor  --before usb-reset
> ```

In my case, this issue happens with both /dev/ttyACM0 (while working with esp32s3) and /dev/ttyUSB0 (while working with esp32). They only work if I use "--list-all-ports".

contra-bit, 2024-05-17:
> In my case, this issue happens with both /dev/ttyACM0 (while working with esp32s3) and /dev/ttyUSB0 (while working with esp32). They only work if I use "--list-all-ports".

I only have tried `/dev/ttyACM0`

nullsauce, 2024-06-07:
Same issue here when running `espflash` from within a docker container on Linux. It seems like my `/dev/ttyUSB0` is recognized as a `SerialPortType::Unknown` in this setup. Passing `--list-all-ports` fixes the issue.

More details:

* The serial port is exposed in the container using `--device /dev/ttyUSB0`
* Serial port file (from within the container):
```
$ stat /dev/ttyUSB0 
  File: /dev/ttyUSB0
  Size: 0         	Blocks: 0          IO Block: 4096   character special file
Device: 0,99	Inode: 11          Links: 1     Device type: 188,0
Access: (0666/crw-rw-rw-)  Uid: ( 1000/     esp)   Gid: (   20/ dialout)
Access: 2024-06-07 08:38:25.765085351 +0000
Modify: 2024-06-07 08:38:25.765085351 +0000
Change: 2024-06-07 08:38:25.765085351 +0000
 Birth: -
```

JZYFC, 2024-08-27:
> Same issue here when running `espflash` from within a docker container on Linux. It seems like my `/dev/ttyUSB0` is recognized as a `SerialPortType::Unknown` in this setup. Passing [...]

For anyone having this issue with docker container, please add this to your `docker run` options:

```shell
-v /run/udev:/run/udev:ro
```

Reference: https://github.com/serialport/serialport-rs/issues/153

@nullsauce Hope this will help you.

jessebraham, 2024-10-25:
I believe this should be resolved with #693, I hope to publish a new release next week. I am closing this issue, but if the aforementioned PR has not fixed this then please feel free to re-open!

closed 2024-10-25 by jessebraham