esp-rs/espflash #637: MemData Command error on ESP32C6 without --no-stub flag

jlc-christie, 2024-05-15:
When running any `espflash` command (e.g. `board-info`) against an ESP32-C6-WROOM-1 I get the following error:
```bash
> espflash board-info
[2024-05-15T10:30:07Z INFO ] Detected 4 serial ports
[2024-05-15T10:30:07Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-15T10:30:07Z INFO ] Please select a port
[2024-05-15T10:30:09Z INFO ] Serial port: '/dev/cu.usbmodem574C0343841'
[2024-05-15T10:30:09Z INFO ] Connecting...
[2024-05-15T10:30:09Z INFO ] Using flash stub
Error:   × The bootloader returned an error
  ├─▶ Error while running MemData command
  ╰─▶ Other
```

If I supply the `--no-stub` flag it works as expected:
```bash
> espflash board-info --no-stub
[2024-05-15T10:30:46Z INFO ] Detected 4 serial ports
[2024-05-15T10:30:46Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-15T10:30:46Z INFO ] Please select a port
[2024-05-15T10:30:48Z INFO ] Serial port: '/dev/cu.usbmodem574C0343841'
[2024-05-15T10:30:48Z INFO ] Connecting...
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:4a:c6:d0
```

This is reproducible on the latest commit in main (`890f17ac46ea8d579118ae5a726c3a940ede9065`). 

The same error occurs with other commands too, e.g. `monitor`, `flash`, etc.

Vollbrecht, 2024-05-15:
Thanks for reporting, can you run `espflash` with `RUST_LOG=debug espflash board-info` in the no-stub case. Thanks!

jlc-christie, 2024-05-15:
> Thanks for reporting, can you run `espflash` with `RUST_LOG=debug espflash board-info` in the no-stub case. Thanks!

Thanks for the fast response 😄 

```
> RUST_LOG=debug espflash board-info
[2024-05-15T10:40:04Z DEBUG] BoardInfo(
        ConnectArgs {
            after: HardReset,
            baud: None,
            before: DefaultReset,
            chip: None,
            confirm_port: false,
            list_all_ports: false,
            no_stub: false,
            port: None,
        },
    )
[2024-05-15T10:40:04Z DEBUG] connecting to crates.io:443 at [2600:9000:203b:2600:c:7ed3:240:93a1]:443
[2024-05-15T10:40:04Z DEBUG] No cached session for DnsName("crates.io")
[2024-05-15T10:40:04Z DEBUG] Not resuming any session
[2024-05-15T10:40:04Z DEBUG] Using ciphersuite TLS13_AES_128_GCM_SHA256
[2024-05-15T10:40:04Z DEBUG] Not resuming
[2024-05-15T10:40:04Z DEBUG] TLS1.3 encrypted extensions: [ServerNameAck]
[2024-05-15T10:40:04Z DEBUG] ALPN protocol is None
[2024-05-15T10:40:04Z DEBUG] created stream: Stream(RustlsStream)
[2024-05-15T10:40:04Z DEBUG] sending request GET https://crates.io/api/v1/crates/espflash/versions
[2024-05-15T10:40:04Z DEBUG] writing prelude: GET /api/v1/crates/espflash/versions HTTP/1.1
    Host: crates.io
    User-Agent: ureq/2.9.7
    Accept: */*
    accept-encoding: gzip
[2024-05-15T10:40:04Z DEBUG] Chunked body in response
[2024-05-15T10:40:04Z DEBUG] response 200 to GET https://crates.io/api/v1/crates/espflash/versions
[2024-05-15T10:40:04Z DEBUG] dropping stream: Stream(RustlsStream)
[2024-05-15T10:40:04Z DEBUG] Config: Config {
        baudrate: None,
        bootloader: None,
        connection: Connection {
            serial: None,
        },
        partition_table: None,
        usb_device: [
            UsbDevice {
                vid: 6790,
                pid: 21971,
            },
        ],
        save_path: "/Users/jlcc/Library/Application Support/rs.esp.espflash/espflash.toml",
    }
[2024-05-15T10:40:04Z INFO ] Detected 4 serial ports
[2024-05-15T10:40:04Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-15T10:40:04Z INFO ] Please select a port
[2024-05-15T10:40:06Z INFO ] Serial port: '/dev/cu.usbmodem574C0343841'
[2024-05-15T10:40:06Z INFO ] Connecting...
[2024-05-15T10:40:06Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2024-05-15T10:40:06Z DEBUG] Boot Mode: 0x4
[2024-05-15T10:40:06Z DEBUG] Download Mode: false
[2024-05-15T10:40:06Z DEBUG] Writing command: Sync
[2024-05-15T10:40:06Z DEBUG] Writing command: Sync
[2024-05-15T10:40:06Z DEBUG] Writing command: ReadReg { address: 1073745920 }
[2024-05-15T10:40:06Z INFO ] Using flash stub
[2024-05-15T10:40:06Z DEBUG] Loading flash stub for chip: Esp32c6
[2024-05-15T10:40:06Z DEBUG] Write 15956 byte stub text
[2024-05-15T10:40:06Z DEBUG] Writing command: MemBegin { size: 15956, blocks: 3, block_size: 6144, offset: 1082130432, supports_encryption: false }
[2024-05-15T10:40:06Z DEBUG] Writing command: MemData { data: [183, 0, 128, 64, 103, 128, 128, 0, 23, 69, 0, 0, 19, 5, 133, 4, 151, 5, 1, 0, 147, 133, 69, 71, 99, 87, 181, 0, 129, 70, 20, 193, 17, 5, 227, 78, 181, 254, 23, 5, 128, 15, 19, 5, 165, 253, 151, 5, 128, 15, 147, 133, 37, 253, 99, 87, 181, 0, 129, 70, 20, 193, 17, 5, 227, 78, 181, 254, 129, 64, 1, 65, 129, 65, 1, 66, 129, 66, 1, 67, 129, 67, 1, 68, 129, 68, 1, 69, 129, 69, 1, 70, 129, 70, 1, 71, 129, 71, 1, 72, 129, 72, 1, 73, 129, 73, 1, 74, 129, 74, 1, 75, 129, 75, 1, 76, 129, 76, 1, 77, 129, 77, 1, 78, 129, 78, 1, 79, 129, 79, 151, 65, 0, 0, 147, 129, 225, 124, 243, 35, 64, 241, 183, 2, 0, 0, 147, 130, 2, 0, 99, 244, 114, 0, 111, 0, 128, 4, 23, 225, 6, 0, 19, 1, 33, 87, 145, 66, 51, 1, 81, 64, 19, 113, 1, 255, 10, 132, 111, 0, 64, 0, 65, 17, 6, 198, 151, 32, 0, 0, 231, 128, 160, 169, 151, 16, 0, 0, 231, 128, 224, 102, 55, 21, 129, 64, 183, 197, 173, 222, 147, 133, 229, 171, 35, 34, 181, 72, 151, 0, 0, 0, 231, 128, 96, 92, 111, 0, 0, 0, 65, 17, 6, 198, 151, 0, 128, 255, 231, 128, 96, 6, 19, 53, 21, 0, 125, 21, 19, 117, 53, 6, 178, 64, 65, 1, 130, 128, 65, 17, 6, 198, 131, 69, 21, 0, 3, 70, 5, 0, 131, 70, 37, 0, 3, 71, 53, 0, 162, 5, 209, 141, 194, 6, 98, 7, 217, 142, 51, 232, 182, 0, 131, 69, 85, 0, 3, 70, 69, 0, 131, 70, 101, 0, 3, 71, 117, 0, 162, 5, 209, 141, 194, 6, 98, 7, 217, 142, 213, 141, 3, 70, 149, 0, 131, 70, 133, 0, 3, 71, 165, 0, 131, 71, 181, 0, 34, 6, 85, 142, 66, 7, 226, 7, 93, 143, 89, 142, 131, 72, 213, 0, 3, 71, 197, 0, 131, 71, 229, 0, 131, 70, 245, 0, 162, 8, 51, 231, 232, 0, 194, 7, 226, 6, 221, 142, 217, 142, 131, 72, 21, 1, 131, 66, 5, 1, 3, 67, 37, 1, 131, 71, 53, 1, 162, 8, 51, 231, 88, 0, 66, 3, 226, 7, 179, 231, 103, 0, 93, 143, 131, 66, 85, 1, 131, 72, 69, 1, 3, 67, 101, 1, 3, 69, 117, 1, 162, 2, 179, 231, 18, 1, 66, 3, 98, 5, 51, 101, 101, 0, 201, 143, 66, 133, 151, 0, 128, 255, 231, 128, 64, 250, 19, 53, 21, 0, 125, 21, 19, 117, 69, 252, 178, 64, 65, 1, 130, 128, 178, 134, 46, 134, 129, 69, 23, 3, 0, 0, 103, 0, 35, 8, 65, 17, 6, 198, 34, 196, 38, 194, 46, 132, 170, 132, 151, 0, 0, 0, 231, 128, 224, 36, 41, 70, 38, 133, 162, 133, 151, 0, 0, 0, 231, 128, 32, 37, 3, 69, 180, 0, 131, 69, 164, 0, 3, 70, 196, 0, 131, 70, 212, 0, 34, 5, 77, 141, 66, 6, 226, 6, 179, 229, 198, 0, 201, 141, 3, 69, 244, 0, 3, 70, 228, 0, 131, 70, 4, 1, 3, 71, 20, 1, 34, 5, 81, 141, 194, 6, 98, 7, 51, 102, 215, 0, 73, 142, 38, 133, 151, 0, 0, 0, 231, 128, 192, 32, 38, 133, 178, 64, 34, 68, 146, 68, 65, 1, 23, 3, 0, 0, 103, 0, 131, 30, 57, 113, 6, 222, 34, 2 [... cut ...] 8, 64, 241, 5, 230, 138, 4, 182, 148, 152, 200, 144, 66, 93, 142, 144, 194, 241, 21, 17, 5, 5, 7, 233, 249, 125, 85, 115, 16, 69, 48, 178, 64, 34, 68, 146, 68, 2, 73, 65, 1, 130, 128, 0, 0, 46, 134, 213, 69, 23, 3, 0, 0, 103, 0, 131, 204, 249, 69, 1, 70, 23, 3, 0, 0, 103, 0, 195, 203, 253, 69, 1, 70, 23, 3, 0, 0, 103, 0, 3, 203, 217, 69, 1, 70, 23, 3, 0, 0], pad_to: 4, pad_byte: 0, sequence: 0 }
Error:   × The bootloader returned an error
  ├─▶ Error while running MemData command
  ╰─▶ Other
```

jlc-christie, 2024-05-15:
Following a [conversation in the esp-rs room](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$tl54Y2B_c5n0BoT6A3g8WQZhPDeYqEVqPwleGtozVNo?via=matrix.org&via=tchncs.de&via=mozilla.org), this was resolved by selecting the JTAG port _after_ holding reset + boot to put it in download mode. This issue is only a problem using the serial option.

[This is the board](https://www.amazon.co.uk/dp/B0CKR2LF83?psc=1&ref=ppx_yo2ov_dt_b_product_details) I'm using and it has a single USB Type C hub between the UART and JTAG connection.

Here are the options that appear for the board, when the board is _not_ in download mode, selecting the JTAG option will just hang for a while and then error, when in download mode it works fine without the need for the `--no-stub` flag.
[image]

robert-chiniquy, 2024-09-04:
As a note, this is happening to me with an ESP32-S3-WROOM-1, the board does have a USB-C UART bridge. This board does not have a JTAG interface, even when it is in download mode. My debug output looks pretty much like the above.

jessebraham, 2025-03-25:
I realize this issue is a bit dated, but is anybody able to verify if this is still the case when using the current `main` branch? We have swapped out the flasher stub since this issue was opened, so I'm curious if the problem has been resolved as a result or not.

aqrln, 2025-04-12:
@jessebraham I'm facing the issue using the `main` branch:

```
$ cargo install --git https://github.com/esp-rs/espflash.git espflash
...

$ espflash --version
espflash 4.0.0-dev

$ espflash board-info
[2025-04-12T19:27:42Z INFO ] Detected 2 serial ports
[2025-04-12T19:27:42Z INFO ] Ports which match a known common dev board are highlighted
[2025-04-12T19:27:42Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2025-04-12T19:27:44Z INFO ] Serial port: '/dev/cu.usbmodem58370100181'
[2025-04-12T19:27:44Z INFO ] Connecting...
[2025-04-12T19:27:44Z INFO ] Using flash stub
Error:   × The bootloader returned an error
  ├─▶ Error while running MemData command
  ╰─▶ Received message has invalid CRC

$ espflash board-info --no-stub
[2025-04-12T19:27:52Z INFO ] Detected 2 serial ports
[2025-04-12T19:27:52Z INFO ] Ports which match a known common dev board are highlighted
[2025-04-12T19:27:52Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2025-04-12T19:27:53Z INFO ] Serial port: '/dev/cu.usbmodem58370100181'
[2025-04-12T19:27:53Z INFO ] Connecting...
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:57:2a:48

Security Information:
=====================
Flags: 0x00000000 (0)
Key Purposes: [0, 0, 0, 0, 0, 0, 12]
Chip ID: 13
API Version: 0
Secure Boot: Disabled
Flash Encryption: Disabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0
```

aqrln, 2025-04-14:
@jessebraham for additional context, I'm on macOS 15.3.2, and using [this board](https://github.com/wuxx/nanoESP32-C6/blob/master/README_en.md).

closed 2025-05-27 by JurajSadel