esp-rs/espflash #640: Arduino UNO R4 WiFi - ESP32-S3 Not connecting

MarcusWentz, 2024-05-25:
When I run:
```shell
espflash board-info
```
I get the following error:
```shell
[2024-05-25T11:39:02Z INFO ] Serial port: '/dev/ttyACM1'
[2024-05-25T11:39:02Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

I tried switching connected ports but I get the same error log.

I am following this Rust ESP32 tutorial:

https://www.youtube.com/watch?v=vUSHaogHs1s&list=PLkch9g9DEE0Lkm1LqcD7pZNDmXEczOo-a

Seems related to:

https://github.com/esp-rs/espflash/issues/534

@SergioGasquez what might be causing this? It sounded like a pin needed to be shorted for the Arduino Nano issue linked above.

MarcusWentz, 2024-05-25:
Yes I have heard controversy regarding having both devices on the same board.
I am not sure how Arduino handles this either. I tried flashing while clicking and holding the RESET button, but no luck. 

I have been running robotics and internet Rust projects on a Raspberry Pi 4, since it acts like a computer with GPIO pins. 
I have been curious to try Rust robotics and internet development on Arduino hardware. 

Maybe I should consider using a simpler board like a Arduino Nano ESP32-S3 since this same issue looked resolved?

Vollbrecht, 2024-05-25:
one thing you can do if you didn't try is using `espflash board-info --no-stub` also `espflash` has aditional arguments around resets, regarding pre and post resets after a command is run, You can check all commands for each subcommand by `espflash <command> help` e,g `espflash board-info help`

MarcusWentz, 2024-05-25:
I tried 
```shell
espflash board-info --no-stub
```
but the error still exists. 

Should we close this issue since 
this project is focused on ESP32 development and the Renesas R7FA4M1AB3CFM (RA4M1) seems out of scope?
I might just get a Arduino Nano ESP32-S3 instead to keep things simple.

SergioGasquez, 2024-05-27:
Hi! As mentioned before, its probably due to the fact that there are 2 devices on the same board, I'm curious if `esptool` works, and how they solve this issue. AFAIK, Arduino IDE uses `esptool` under the hood, could you try flashing with `esptool` (https://github.com/espressif/esptool/)?

MarcusWentz, 2024-05-27:
I downloaded esptool with:
```shell
pip install esptool
```
when I run:
```shell
esptool.py version
```
I get
```shell
esptool.py v4.7.0
4.7.0
```

Is there a command for getting all ESP32 connected devices like
```
espflash board-info
```
?

esptool documentation:

https://docs.espressif.com/projects/esptool/en/latest/esp32/#quick-start

MarcusWentz, 2024-05-27:
When I run 
```shell
esptool.py chip_id
```
I get the following errors:
```shell
esptool.py v4.7.0
Found 2 serial ports
Serial port /dev/ttyACM1
Connecting......................................
/dev/ttyACM1 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting......................................
/dev/ttyACM0 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
```
I think I will just try out a regular ESP32 board to keep things simple. Should we close this issue for now?

closed 2024-05-28 by SergioGasquez