esp-rs/espflash #641: espflash on WSL 2 (with usbipd) doesn't work

dom96, 2024-05-26:
I've successfully flashed my ESP32 via other tools in WSL 1, but espflash does not work for me (even if I try it with usbipd set up on WSL2).

Here is what I'm seeing:

```
$ sudo usbip list -r 172.19.96.1
Exportable USB devices
======================
 - 172.19.96.1
        1-7: Silicon Labs : CP210x UART Bridge (10c4:ea60)
           : USB\VID_10C4&PID_EA60\0001
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/00/00)

$ sudo usbip attach -r 172.19.96.1 --busid=1-7
$ sudo usbip port
Imported USB devices
====================
Port 00: <Port in Use> at Full Speed(12Mbps)
       Silicon Labs : CP210x UART Bridge (10c4:ea60)
       1-1 -> usbip://172.19.96.1:3240/1-7
           -> remote bus/dev 001/007
$ espflash board-info
Error: espflash::no_serial

  × No serial ports could be detected
  help: Make sure you have connected a device to the host system. If the device is connected but not listed, try
        using the `--list-all-ports` flag.
```

Am I missing something or should this work?

SergioGasquez, 2024-05-27:
Hi! Have you tried the `--list-all-ports` flag as suggested by the error message? I'm not a Windows user, but I know some colleagues are using WSL2 with no issues.

dom96, 2024-05-27:
Yep, I have. It doesn't change the output.

akauppi, 2024-06-12:
>I've successfully flashed my ESP32 via other tools in WSL 1, but espflash does not work for me (even if I try it with usbipd set up on WSL2).

My setup is:
- Windows 10 Home
- `usbipd` v.4.2.0 installed (Windows side)
- WSL2 (Ubuntu 22.04.4 LTS by `lsb_release -a`)

Have you tried attaching the device completely from Windows side, i.e.

```
> usbipd attach --wsl -b 1-7
```

I don't ever (need to) do `usbip` on the WSL side. In particular, `espflash board-info` works for me with the above setup, and ESP32-C3-DevKit-RUST-1 (chip rev 0.4) board.

akauppi, 2024-06-12:
..if that doesn't help, I would try installing the `CP210x Universal Windows Driver` (v.11.3.0) on Windows side from [here](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads).

I can see you are using a standard UART (not JTAG) connector. Could it be that even though WSL properly lists it as "Silicon Labs : CP210x UART Bridge", it'd still need a driver on the Windows side?

Eplankton, 2024-09-07:
> ..if that doesn't help, I would try installing the `CP210x Universal Windows Driver` (v.11.3.0) on Windows side from [...]

Exactly same problem for me.

parsiya, 2024-10-13:
I solved this by enabling `systemd` based on [usbipd-win/issues/948#](https://github.com/dorssel/usbipd-win/issues/948#issuecomment-2138336528).

In WSL2, create or edit `/etc/wsl.conf` and add the following to it. Source: https://learn.microsoft.com/en-us/windows/wsl/wsl-config#systemd-support

```
[boot]
systemd=true
```

`wsl.exe --shutdown` from the Windows side and now the device is recognized in WSL2 Debian.

closed 2025-01-20 by jessebraham

korir248, 2025-03-09:
> I solved this by enabling `systemd` based on [usbipd-win/issues/948#](https://github.com/dorssel/usbipd-win/issues/948#issuecomment-2138336528).
> 
> In WSL2, create or edit `/etc/wsl.conf` and add the following to it. Source: https://learn.microsoft.com/en-us/windows/wsl/wsl-config#systemd-support
> 
> ```
> [boot]
> systemd=true
> ```
> 
> `wsl.exe --shutdown` from the Windows side and now the device is recognized in WSL2 Debian after running `usbipd attach --wsl --busid <busid>` as `ttyUSB0`.

@parsiya Thank you for this. This fixed the error I was getting in my wsl