esp-rs/espflash #645: Add support for --no-default-features

etiennetremel, 2024-06-01:
It would be a nice addition to allow disabling default features via a flag, similar to what `cargo build --no-default-features --features one,two` does.

Currently, espflash support adding features but doesn't allow disabling the one default one.

Related to https://github.com/esp-rs/espflash/issues/305

SergioGasquez, 2024-06-03:
Could you elaborate on your use case? Currently, the only default feature of espflash is `cli` which can be disabled when used as a library and its fundamental when using it as a binary.

etiennetremel, 2024-06-03:
Sorry, I wasn't really clear.

I'm using espflash from the CLI, so the feature that are being compiled and flashed into the esp32 are from my project, not from the espflash Cargo.toml:

```bash
cargo espflash flash --release --features one,two
```

My project has default features which I do not want, using the cargo build there is a flag that can be used to drop all default features but that's not a flag available on the espflash flash sub command.

I'm expecting to be able to do the same as cargo build, which would look like this:

```bash
cargo espflash flash --release --features one,two --no-default-features
```

closed 2024-07-01 by SergioGasquez