esp-rs/espflash #646: How to use with ftdi FT232-ttl adapter (esp32s3 board)? Able to monitor but not able to flash...

cmorganBE, 2024-06-07:
esp323s3 here, I'm able to boot into bootrom mode via BOOT0 and reset and can confirm if I open the device up with picocom (at the bottom).

But when the chip is in this mode I'm not able to program, I get timeouts:

```
$ espflash flash build/firmware.elf -p /dev/ttyUSB0 -c esp32s3 -b no-reset-no-sync
[2024-06-07T19:18:10Z WARN ] Pre-connection option 'NoResetNoSync' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-06-07T19:18:10Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-06-07T19:18:10Z INFO ] Connecting...
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

Am I missing something? 

I'm able to monitor as well btw:

```
$ espflash monitor -p /dev/ttyUSB0 --chip esp32s3 -b no-reset-no-sync
[2024-06-07T19:21:23Z WARN ] Pre-connection option 'NoResetNoSync' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-06-07T19:21:23Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-06-07T19:21:23Z INFO ] Connecting...
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x33 (DOWNLOAD(USB/UART0))
waiting for download
```

```
$ picocom -b 115200 /dev/ttyUSB0
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x33 (DOWNLOAD(USB/UART0))
waiting for download
```

SergioGasquez, 2024-06-11:
A few things to try/questions:
- Have you tried without the `-b no-reset-no-sync` argument? 
- Could you also try to use `--no-stub`?
- Which version of `espflash` are you using?
- Can you post the debug logs `RUST_LOG=debug espflash ...`

cmorganBE, 2024-06-13:
@SergioGasquez any other info from here that could help?

cmorganBE, 2024-06-13:
@SergioGasquez can it? Is the protocol used for flashing the esp32 a standard? I'm not looking for serial interactivity, or jtag/swd debugging, just flashing (which is what drew me to espflash :-) )

SergioGasquez, 2024-06-13:
> @SergioGasquez can it? Is the protocol used for flashing the esp32 a standard? I'm not looking for serial interactivity, or jtag/swd debugging, just flashing (which is what drew me to espflash :-) )

probe-rs can also be used to flash via the USB/Serial-JTAG, see https://docs.esp-rs.org/book/tooling/debugging/probe-rs.html#probe-rs and probe-rs docs

cmorganBE, 2024-06-13:
@SergioGasquez sure, but why wouldn't espflash be the place for this? It's already handling chip specific things and has the proper options to exclude reset/boot etc.

Wouldn't I have to do a bunch of the same stuff to use probe-rs for this?

If espflash isn't the right place for this kind of thing I'd be interested in why.

cmorganBE, 2024-06-21:
Any updates on this issue? I'd like to use espflash for code load here without needing a build and esp-idf sdk on the system to load code.

Vollbrecht, 2024-06-21:
you mention you are only connectiong tx/rx ? If you are not programming through usb-jtag but through an usb/uart converter on the uart site you need to have the correct connections. In the general case its not enough to have tx/rx connected but also dtr/rts signals. They usually go to the esp32s3 reset gpio and to the gpio0 of the chip respectivly. If that is not present than you cant expect to have a working autodownload! In that case you need to manually reset it. For example look at one [official](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-DevKitC-1_V1.1_20220413.pdf) dev board schematic how its connected correctly.