esp-rs/espflash #649: espflash fails to automatically set the port

HaoboGu, 2024-06-24:
Hello! I'm using macOS with a custom esp32s3 board. When I connected the usb cable to the serial JTAG, it shows correctly: 

[image]

However, I cannot use `espflash` to access the board without specifying `--port`:

```bash
$ espflash board-info                 
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter

$ ls -l /dev/cu*
crw-rw-rw-  1 root  wheel  0x9000005  6 24 22:47 /dev/cu.usbmodem1101

$ espflash board-info --port /dev/cu.usbmodem1101
[2024-06-24T14:50:56Z INFO ] Serial port: '/dev/cu.usbmodem1101'
[2024-06-24T14:50:56Z INFO ] Connecting...
[2024-06-24T14:50:56Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       24:58:7c:e2:cc:90
```

The issue occurs on both latest release and git version.

```
$ espflash -V                                    
espflash 3.1.0
```

SergioGasquez, 2024-06-25:
Is your port listed if you use `--list-all-ports` flag?

SergioGasquez, 2024-06-25:
By the way, I dont have any issues recognizing a `ESP32-S3-DevKitC-1 v1.0` connected via USB Serial JTAG port.
```
$ espflash -V
espflash 3.1.0
$ espflash board-info
[2024-06-25T09:58:21Z INFO ] Detected 2 serial ports
[2024-06-25T09:58:21Z INFO ] Ports which match a known common dev board are highlighted
[2024-06-25T09:58:21Z INFO ] Please select a port
[2024-06-25T09:58:21Z INFO ] Serial port: '/dev/cu.usbmodem11101'
[2024-06-25T09:58:21Z INFO ] Connecting...
[2024-06-25T09:58:22Z INFO ] Using flash stub
[2024-06-25T09:58:22Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       60:55:f9:f5:20:d4
```

HaoboGu, 2024-06-25:
> Is your port listed if you use `--list-all-ports` flag?

No, still

```
$ espflash board-info --list-all-ports

Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter
```

SergioGasquez, 2024-07-18:
Have you tried with other flashing tools like `esptool`?

HaoboGu, 2024-07-19:
> Have you tried with other flashing tools like `esptool`?

I just tried `esptool`:

```shell
pip install esptool
esptool.py flash_id
```

It recognizes the chip
```
$ esptool.py flash_id
esptool.py v4.7.0
Found 4 serial ports
Serial port /dev/cu.usbmodem211401
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: ec:da:3b:d1:87:c4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 2016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

dumski, 2024-07-31:
Same here with Seeed Xiao ESP32-XX boards...

MabezDev, 2025-02-26:
Could you try again with git main? We've recently reworked port detection thanks to some advancements in the `serialport` library.

HaoboGu, 2025-02-27:
I tried espflash v3.3.0, now it works on my mac.

closed 2025-02-27 by HaoboGu