esp-rs/espflash #650: Add a flag to wait for serial port (Feature request)

mikhail-m1, 2024-07-02:
It would be nice to have a flag (something like `--wait`) to wait for serial port to appear in the system. I can make a PR if you are ready to merge it.

SergioGasquez, 2024-07-03:
Hi! Do you mind elaborating on your use case for this flag? I guess you want to run the command and then plug the board? Will the command ever timeout?

SergioGasquez, 2024-07-18:
Hi! Sorry for the late reply, if this is something that would help your scenario we would be happy to accept your contribution

harshilv17, 2026-11-09:
Hi @SergioGasquez, I'd like to take this. Plan: add a `--wait` flag to the connection args; when the given or auto-detected port isn't present, poll every 500ms until it appears (Ctrl+C to quit, no timeout by default). I'll test it on an ESP32 by starting the command before plugging the board in.

SergioGasquez, 2026-11-09:
Hi @harshilv17! Feel free to proceed and work on this feature, but note that we are still not sure if we will do a major/minor/patch bump for the next release. I guess patch/minor are completely fine to accept, but if your PR contains major breaking changes, it might be on hold.

closed 2026-11-10 by SergioGasquez