esp-rs/espflash #657: QIO does not work (at least with esp-s3-wroom-1-N16R8)

liebman, 2024-07-28:
If you specify the flash mode or `qio` to espflash then it fails to boot:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3818,len:0x16f8
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3818,len:0x16f8
ets_loader.c 78 
```

However If I build and copy a `bootloader.bin` from an `esp-idf` project that specifies the QIO flash mode and just let `espflash` flash as DIO it works. And the esp-idf bootloader switches to QIO.

[This section](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/bootloader.html#spi-flash-configuration) in the api guide includes:

> The First Stage Bootloader in ROM reads the Second Stage Bootloader header information from flash and uses this information to load the rest of the Second Stage Bootloader from flash. However, at this time the system clock speed is lower than configured and not all flash modes are supported.

Also esp-idf seems to mark the bootloader as DIO when you have specified QIO.

liebman, 2024-07-28:
More info found [here](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/spi-flash-modes.html#how-is-flash-mode-communicated-to-the-espressif-chip) that confirms that the bootloader should be marked as DIO even for QIO.

USTHzhanglu, 2024-12-18:
i think is not espflash question
with idf, through config FLASHMODE_QIO, sdkconfig config CONFIG_ESPTOOLPY_FLASHMODE="dio" and couldn't change.
by cmd
```sh
esptool.py --chip esp32s3 merge_bin -o hello_world.bin --flash_mode qio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 hello.bin 0x8000 partition_table/partition-table.bin
```
and flash,it also error
```log
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
--- 0x400454d5: ets_loader_map_range in ROM

SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce2820,len:0x1918
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
```

damischa1, 2026-10-02:
Some data on this from an ESP32-S3 (LILYGO T-2CAN: 16 MB quad flash, 8 MB octal PSRAM), espflash with an esp-hal app.

**Same failure, same cause.** `--flash-mode qio` writes QIO into the 2nd-stage bootloader's header. The ROM then loads the bootloader in QIO before anything has set the flash's QE bit, and it ends at `ets_loader.c 78` with a TG0WDT reset loop.

The workaround from the first post works. I built an ESP-IDF bootloader with `CONFIG_ESPTOOLPY_FLASHMODE_QIO=y` and `CONFIG_ESPTOOLPY_FLASHFREQ_80M=y`, then ran `espflash flash --bootloader <it> --flash-freq 80mhz` with the default (DIO) mode. The log shows `SPI Mode: QIO`. The app then reads flash with `fread_qio` set in SPI0 `CTRL`, and esp-hal's PSRAM init leaves that alone.

**On @bugadani's question (one bootloader or one per mode).** The QIO switch in the IDF bootloader is compile-time. In the run above, the QIO-built bootloader's header said DIO, and it still enabled QIO. So a QIO-built bootloader cannot serve as the default. Boards whose flash has no QE support, or that use the WP/HD pins otherwise, would stop booting. That makes it a separate binary per chip, selected when the user asks for `qio`/`qout`.

What the selection would do is what ESP-IDF and esptool's docs describe:

- use the QIO-built bootloader;
- keep the bootloader's own header at DIO, since the ROM must load it in DIO;
- put QIO only in the app header.

**Why it is worth it.** An esp-hal app that executes from flash under load is limited by cache-line fills. Measured on that board with the Xtensa PMU, with BLE streaming on core 0 and a blocking SPI transfer on core 1:

- A 32-byte I-cache miss went from ~2.4 µs (DIO 80 MHz) to ~1.4 µs (QIO 80 MHz).
- The worst-case transfer went from ~250 µs to ~150 µs.

Details: https://github.com/esp-rs/esp-hal/issues/6422#issuecomment-5950448234

I would be happy to open a PR along those lines:

- `qio` entries in `resources/bootloaders/manifest.yaml`, built with the existing `build-bootloaders` xtask;
- picking that bootloader when the flash mode is `qio`/`qout`, with the bootloader header left at DIO;
- a clear error for chips that have no QIO bootloader yet.

Starting with the ESP32-S3, or every chip in the manifest? Let me know which you would prefer.

closed 2026-10-26 by SergioGasquez