esp-rs/espflash #658: Build fails in `time` dependency on rustc 1.80.0 and nightly

mbuesch, 2024-07-31:
Tested commit 45946f84e0b54043f4ac63bd520adccd36bf63a0 (currently latest)

```
error[E0282]: type annotations needed for `Box<_>`
  --> /.../cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.3.34/src/format_description/parse/mod.rs:83:9
   |
83 |     let items = format_items
   |         ^^^^^
...
86 |     Ok(items.into())
   |              ---- type must be known at this point
   |
help: consider giving `items` an explicit type, where the placeholders `_` are specified
   |
83 |     let items: Box<_> = format_items
   |              ++++++++

   Compiling gix-packetline v0.17.5
For more information about this error, try `rustc --explain E0282`.
```

LarryMerino, 2024-08-01:
I think I have a similar problem when I try to install cargo-espflash. I have no problem installing espflash but I get this error when I try with cargo-espflash

```
error[E0283]: type annotations needed
  --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-credentials-0.23.1/src/program/mod.rs:83:63
   |
83 |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
   |                                                               ^^^^^^
   |
   = note: multiple `impl`s satisfying `BString: AsRef<_>` found in the `bstr` crate:
           - impl AsRef<BStr> for BString;
           - impl AsRef<[u8]> for BString;
help: try using a fully qualified path to specify the expected types
   |
83 |                 gix_command::prepare(gix_path::from_bstr(<BString as AsRef<T>>::as_ref(&args)).into_owned())
   |                                                          +++++++++++++++++++++++++++++++    ~

error[E0283]: type annotations needed
   --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-credentials-0.23.1/src/program/mod.rs:83:38
    |
83  |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ cannot infer type for reference `&_`
    |
    = note: multiple `impl`s satisfying `Cow<'_, BStr>: From<&_>` found in the `bstr` crate:
            - impl<'a> From<&'a BStr> for Cow<'a, BStr>;
            - impl<'a> From<&'a BString> for Cow<'a, BStr>;
    = note: required for `&_` to implement `Into<Cow<'_, BStr>>`
note: required by a bound in `from_bstr`
   --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-path-0.10.9/src/convert.rs:135:34
    |
135 | pub fn from_bstr<'a>(input: impl Into<Cow<'a, BStr>>) -> Cow<'a, Path> {
    |                                  ^^^^^^^^^^^^^^^^^^^ required by this bound in `from_bstr`

   Compiling strum_macros v0.26.4
```

Rustc version 
```
❯ rustc --version
rustc 1.82.0-nightly (28a58f2fa 2024-07-31)
```

mbuesch, 2024-08-01:
My problem happens with the `Cargo.lock` or with `--locked` from crates.io
and I think yours happens without `--locked` from crates.io or if you `cargo update` the repo first.

So, I think these are in fact two distinct problems.

But the combination of both means that currently cargo-espflash neither builds with locked nor without it.

SergioGasquez, 2024-08-02:
I think https://github.com/esp-rs/espflash/pull/659 should fix the building issues, mind trying it?

closed 2024-08-02 by jessebraham

mbuesch, 2024-08-02:
Thanks. This fixed the problem.

MathyV, 2024-08-14:
@SergioGasquez this issue just bit me, building from git works for me. Could we get a new release (possibly 3.1.1 with just this patch) as following the instructions to get going with rustup and esp-rs now just don't work (and on a personal note I'm giving a workshop on Sunday and it would be nice if it was fixed :-) ).

jessebraham, 2024-08-14:
@MathyV unfortunately Sergio is on vacation right now, and I will be going on vacation tomorrow as well 😅 @MabezDev can maybe help out here but can't make any promises.

 I'll be back on Monday, which doesn't help with your workshop unfortunately, but can get something sorted next week if it has not already been taken care of.

MabezDev, 2024-08-15:
v3.1.1 has been released.