esp-rs/espflash #665: espflash in vscode devcontainer dont work

lgLindstrom, 2024-08-05:
Hi
I am running windows 11 and docker desktop. My ESP development board is DevkitC from Espressif.

Following instructions in "rust on esp"  to created a rust project  by using the command:

`$ cargo generate esp-rs/esp-template`

I used the option to create a dev container. The newly created project directory I opened VsCode and started the dev container.

After a while everyting was set up and the devcontainer was running.

In a vscode terminalar I could issue the command :

`$ cargo build`

First build took a while but it worked.

Then I tried:

`$ cargo run`

It returned the error:

Error: espflash::no_serial

```
  × No serial ports could be detected
  help: Make sure you have connected a device to the host system. If the device is connected but not listed, try using the `--list-all-ports` flag.
```

Googling on the problem found a solution with usbipd from a windows terminal.
` $ usbipd attach --wsl --busid=1-2`

Also I added this to my devcontainer.json file:

```
  "runArgs": [
    "--device=/dev/ttyUSB0",
    "--privileged"
  ],
```

I restarted the Dev-Container and tried to verify the serial connection by issue this command.
```
$ dmesg | egrep --color 'tty'

[42978.070537] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[43690.903522] usb 1-1: cp210x converter now attached to ttyUSB0
```

It shows that my development board is connected to ttyUSB0.

But ,, running:
```
$ espflash board-info -p /dev/ttyUSB0
Error: espflash::serial_not_found

  × The serial port '/dev/ttyUSB0' could not be found
  help: Make sure the correct device is connected to the host system
```

still shows that I cant connect to my development board.

I can see that I am not the only one that have problems with espflash but is there a work around?
I am kind of stuck at the moment.

SergioGasquez, 2024-08-05:
Flashing from a container is not something espflash is intended to do, it can be achieved in some OSs, but sometimes it results in permission issues. From the `esp-template`:
> Dev Containers also allow flashing from the container using [web flash](https://github.com/bjoernQ/esp-web-flash-server) and have the [VS Code Wokwi extension](https://marketplace.visualstudio.com/items?itemName=wokwi.wokwi-vscode) already installed.

lgLindstrom, 2024-08-05:
Thanks,, I am searching instructions how to use "web flash". Can you help?

SergioGasquez, 2024-08-07:
> Thanks,, I am searching instructions how to use "web flash". Can you help?

https://github.com/esp-rs/esp-web-flash-server

closed 2024-08-07 by SergioGasquez