esp-rs/espflash #669: On macOS cargo install cargo-espflash fails on compiling gix-credentials

illabo, 2024-08-15:
- Rust 1.80.1.
- Intel Mac.
- Versions failing to compile 3.1.1, 3.1.0, 3.0.0.
```
error[E0283]: type annotations needed
  --> /Users/illabo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-credentials-0.23.1/src/program/mod.rs:83:63
   |
83 |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
   |                                                               ^^^^^^
   |
   = note: multiple `impl`s satisfying `BString: AsRef<_>` found in the `bstr` crate:
           - impl AsRef<BStr> for BString;
           - impl AsRef<[u8]> for BString;
help: try using a fully qualified path to specify the expected types
   |
83 |                 gix_command::prepare(gix_path::from_bstr(<BString as AsRef<T>>::as_ref(&args)).into_owned())
   |                                                          +++++++++++++++++++++++++++++++    ~

error[E0283]: type annotations needed
   --> /Users/illabo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-credentials-0.23.1/src/program/mod.rs:83:38
    |
83  |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ cannot infer type for reference `&_`
    |
    = note: multiple `impl`s satisfying `Cow<'_, BStr>: From<&_>` found in the `bstr` crate:
            - impl<'a> From<&'a BStr> for Cow<'a, BStr>;
            - impl<'a> From<&'a BString> for Cow<'a, BStr>;
    = note: required for `&_` to implement `Into<Cow<'_, BStr>>`
note: required by a bound in `from_bstr`
   --> /Users/illabo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-path-0.10.9/src/convert.rs:135:34
    |
135 | pub fn from_bstr<'a>(input: impl Into<Cow<'a, BStr>>) -> Cow<'a, Path> {
    |                                  ^^^^^^^^^^^^^^^^^^^ required by this bound in `from_bstr`

For more information about this error, try `rustc --explain E0283`.
error: could not compile `gix-credentials` (lib) due to 2 previous errors
```

sourcebox, 2024-08-15:
Seems not to be macOS-dependent as it also happens on Linux.

Hahzeeboo, 2024-08-16:
I'm getting the same issue on windows 11

kaikongbj, 2024-08-17:
I have the same issue on windows 11

Sid9017, 2024-08-26:
I have the same issue on M1 mac

tirithen, 2024-08-29:
Hi @jessebraham or @icewind1991 ! I wonder if you have an idea of how to fix this compilation error?

I have the problem as well when running Arch Linux.

SergioGasquez, 2024-08-30:
Use `cargo install cargo-espflash --locked` to avoid the error

SergioGasquez, 2024-08-30:
Closing as duplicated of #671

closed 2024-08-30 by SergioGasquez