esp-rs/espflash #670: "Failed to connect to device" on Adafruit QT PY ESP32-S2

dylif, 2024-08-16:
Info:
- OS: Arch Linux 6.10.5-arch1-1
- Board: Adafruit QT PY ESP32-S2
- Toolchains: `esp-13.2.0_20230928`, `xtensa-esp32-elf-clang`
- ESP-IDF Version: 5.2.2

I've put the board into boot loader mode, and I can communicate with it using `esptool`:
```
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 70:04:1d:f2:f6:d6
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
Staying in bootloader.
```
When I run `espflash flash target/xtensa-esp32s2-espidf/debug/humid --monitor `, I get the following error:
```
[2024-08-16T02:59:47Z INFO ] Serial port: '/dev/ttyACM0'
[2024-08-16T02:59:47Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
Please let me know if you need any further information.

GamePad64, 2024-08-16:
I have the same issue on 3.1.0 and 3.1.1 on my custom-designed ESP32-S2 board. Works perfectly with espflash 2.x and esptool.py

MabezDev, 2024-08-16:
Does it work if you pass `--no-stub`?

dylif, 2024-08-16:
> Does it work if you pass `--no-stub`?

Unfortunately, this does not work.

> I have the same issue on 3.1.0 and 3.1.1 on my custom-designed ESP32-S2 board. Works perfectly with espflash 2.x and esptool.py

I just tried using espflash 2.1.0 and it works! Not sure what changed between the versions. Leaving the ticket open as this seems like a bug.

superkooks, 2024-08-23:
I'm having the same issue with an ESP32 on a custom board. I have bisected the issue and found `2f3d110` is the first bad commit.
Interestingly, strace says that espflash doesn't even read or write to the serial port. I can confirm that it doesn't write anything as the LEDs on my UART adapter don't flash.

FreeMasen, 2024-08-26:
I am having the same problem. Here are some additional details to hopefully point us all in the right direction

<detail>

<summary>Successful Arduino Build->Upload</summary>

```shell
--chip esp32s2 \
    --port "/dev/cu.usbmodem01" \
    --baud 921600 \
    --before default_reset \
    --after no-reset \
    write_flash  \
    -z \
    --flash_mode keep \
    --flash_freq keep \
    --flash_size keep \
        0x1000 "<snip>/esp-test2.ino.bootloader.bin" \
        0x8000 "<snip>/esp-test2.ino.partitions.bin" \
        0xe000 "<snip>Arduino15/packages/esp32/hardware/esp32/3.0.4/tools/partitions/boot_app0.bin" \
        0x10000 "<snip>/esp-test2.ino.bin" \
        0x2d0000 "<snip>/Library/Arduino15/packages/esp32/hardware/esp32/3.0.4/variants/adafruit_qtpy_esp32s2/tinyuf2.bin"
esptool.py v4.6
```

</detail>

<detail></detail>

<summary>successful trace logs from espflash 2.1.0 </summary>

```
RUST_LOG=trace cargo run --release 
    Finished `release` profile [optimized] target(s) in 0.08s
     Running `espflash flash --monitor target/xtensa-esp32s2-espidf/release/esp-test`
[2024-08-26T22:58:39Z DEBUG] Flash(
        FlashArgs {
            connect_args: ConnectArgs {
                baud: None,
                port: None,
                no_stub: false,
            },
            flash_config_args: FlashConfigArgs {
                flash_freq: None,
                flash_mode: None,
                flash_size: None,
            },
            flash_args: FlashArgs {
                bootloader: None,
                erase_parts: None,
                erase_data_parts: None,
                format: None,
                monitor: true,
                monitor_baud: None,
                partition_table: None,
                ram: false,
            },
            image: "target/xtensa-esp32s2-espidf/release/esp-test",
        },
    )
[2024-08-26T22:58:39Z DEBUG] connecting to crates.io:443 at 54.230.202.35:443
[2024-08-26T22:58:39Z DEBUG] No cached session for DnsName("crates.io")
[2024-08-26T22:58:39Z DEBUG] Not resuming any session
[2024-08-26T22:58:39Z TRACE] Sending ClientHello Message {
        version: TLSv1_0,
        payload: Handshake {
            parsed: HandshakeMessagePayload {
                typ: ClientHello,
                payload: ClientHello(
                    ClientHelloPayload {
[... 143 lines cut ...]
[2024-08-26T22:59:13Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2024-08-26T22:59:13Z DEBUG] Flash detect OK!
Chip type:         esp32s2 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
MAC address:       84:f7:03:f6:9a:c8
App/part. size:    385,984/4,128,768 bytes, 9.35%
[00:00:00] [========================================]       7/7       0x1000    [00:00:00] [========================================]       1/1       0x8000    [00:00:03] [========================================]     110/110     0x10000   [2024-08-26T22:59:17Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

[2024-08-26T22:59:23Z TRACE] registering event source with poller: token=Token(0), interests=READABLE2024-08-26T22:59:23Z TRACE] registering event source with poller: token=Token(1), interests=READABLE
```

</detail>

<detail>

<summary>TODO: 3.x.x trace logs</summary>

</detail>

I have to step away from my computer right now but I will try and get the 3.x.x logs added here shortly

FreeMasen, 2024-08-27:
Here are the logs every permutation I though would be worth trying

- before
  - default-reset
  - no-reset
  - no-reset-no-sync
  - usb-reset
- after
  - hard-reset
  - no-reset
  - no-reset-no-stub
- flash-mode
  - qio
  - qout
  - dio
  - dout

[out.log](https://github.com/user-attachments/files/16762368/out.log)

sekwah41, 2024-08-30:
Having the same issue so if there is anything I can provide to help or test I will also be able to :)

dylif, 2024-09-03:
> I believe my case is actually a separate issue, but I think setting the flags `--before no-reset-no-sync -c esp32s2` might help in your case.

I apologize for the late response. Unfortunately this does not work for me. I get a different error now. I have confirmed the device is in bootloader mode and the correct serial port is selected.

```
[2024-09-03T00:57:49Z WARN ] Pre-connection option 'NoResetNoSync' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-09-03T00:57:49Z INFO ] Serial port: '/dev/ttyACM0'
[2024-09-03T00:57:49Z INFO ] Connecting...
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

dylif, 2024-09-28:
I actually ended up getting this to work with the following command:
`espflash flash --port /dev/ttyACM0 --after hard-reset --before no-reset --no-stub --flash-mode qio --flash-size 4mb -c esp32s2`

closed 2024-09-28 by dylif