esp-rs/espflash #671: Installing cargo-espflash with cargo fails

FabienTregan, 2024-08-17:
Hi. I try installing `cargo-espflash` but it fails :
```
~/code/no_std-training$ cargo install cargo-espflash espflash
…
   Compiling gix-bitmap v0.2.11
error[E0283]: type annotations needed
  --> /home/fabien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/gix-credentials-0.23.1/src/program/mod.rs:83:63
   |
83 |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
…
    Replaced package `espflash v3.1.0` with `espflash v3.1.1` (executable `espflash`)
error: failed to compile `cargo-espflash v3.1.1`, intermediate artifacts can be found at `/tmp/cargo-installmpKzeQ`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
     Summary Successfully installed espflash! Failed to install cargo-espflash (see error(s) above).
error: some crates failed to install
```
I think I've tried nightly and stable toolchains. I tried changing the defualt tool chains to those two:
```
nightly-x86_64-unknown-linux-gnu (default)
stable-x86_64-unknown-linux-gnu
```
```
~$ uname -v
#45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 15 16:40:02 UTC 2
~$ rustup -V
rustup 1.27.1 (54dd3d00f 2024-04-24)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.79.0 (129f3b996 2024-06-10)`
```
After updating the tool chain, (stable to `rustc 1.80.1 (3f5fd8dd4 2024-08-06, nightly to `rustc 1.82.0-nightly (506052d49 2024-08-16)`) the error is still there.
```

FabienTregan, 2024-08-17:
I quickly checked something: on Jun 20 16:41, I installed v3.1.0 on this same computer with no issue, I'm pretty sure the rust install was the same that failed this morning trying to install v3.1.1

jnnks, 2024-08-20:
All previous espflash versions (2.x.x) did not work for me either. Always the same error in gix-credentials, across multiple different gix-credentials versions. I assume it has to do with the rust version gix is using, gix-credentials has declared 1.65 in their [Cargo.toml](https://github.com/Byron/gitoxide/blob/main/gix-credentials/Cargo.toml).

I ended up installing v3.1.1 using `cargo binstall espflash`.

licsber, 2024-08-21:
same above

```
error[E0283]: type annotations needed
  --> C:\Users\licsber\.cargo\registry\src\mirrors.ustc.edu.cn-4affec411d11e50f\gix-credentials-0.23.1\src\program\mod.rs:83:63
   |
83 |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
   |                                                               ^^^^^^
   |
   = note: multiple `impl`s satisfying `BString: AsRef<_>` found in the `bstr` crate:
           - impl AsRef<BStr> for BString;
           - impl AsRef<[u8]> for BString;
help: try using a fully qualified path to specify the expected types
   |
83 |                 gix_command::prepare(gix_path::from_bstr(<BString as AsRef<T>>::as_ref(&args)).into_owned())
   |                                                          +++++++++++++++++++++++++++++++    ~

error[E0283]: type annotations needed
   --> C:\Users\licsber\.cargo\registry\src\mirrors.ustc.edu.cn-4affec411d11e50f\gix-credentials-0.23.1\src\program\mod.rs:83:38
    |
83  |                 gix_command::prepare(gix_path::from_bstr(args.as_ref()).into_owned())
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ cannot infer type for reference `&_`
    |
    = note: multiple `impl`s satisfying `Cow<'_, BStr>: From<&_>` found in the `bstr` crate:
            - impl<'a> From<&'a BStr> for Cow<'a, BStr>;
            - impl<'a> From<&'a BString> for Cow<'a, BStr>;
    = note: required for `&_` to implement `Into<Cow<'_, BStr>>`
note: required by a bound in `from_bstr`
   --> C:\Users\licsber\.cargo\registry\src\mirrors.ustc.edu.cn-4affec411d11e50f\gix-path-0.10.9\src\convert.rs:135:34
    |
135 | pub fn from_bstr<'a>(input: impl Into<Cow<'a, BStr>>) -> Cow<'a, Path> {
    |                                  ^^^^^^^^^^^^^^^^^^^ required by this bound in `from_bstr`

   Compiling gix-packetline v0.17.5
For more information about this error, try `rustc --explain E0283`.
error: could not compile `gix-credentials` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-espflash v3.1.1`, intermediate artifacts can be found at `C:\Users\licsber\AppData\Local\Temp\cargo-installU1hXOn`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
     Summary Successfully installed espflash, ldproxy! Failed to install cargo-espflash (see error(s) above).
error: some crates failed to install
```

ThomasBoom89, 2024-08-21:
Run into the same issue and updating cargo as dependency will fix it.
You can look at this commit https://github.com/esp-rs/espflash/pull/666/commits/f00ba45829e25ccc185c3235cba22faaa54791e8 in the open pull request #666 and install from local within the cargo-espflash directory like so -> ``` cargo install --path ./ ```

jjcfrancisco, 2024-08-22:
> Run into the same issue and updating cargo as dependency will fix it.
> You can look at this commit [f00ba45](https://github.com/esp-rs/espflash/commit/f00ba45829e25ccc185c3235cba22faaa54791e8) in the open pull request #666 and install from local within the cargo-espflash directory like so -> `cargo install --path ./`

I tried this but got the same error.

SergioGasquez, 2024-08-30:
Use cargo install cargo-espflash --locked to avoid the error

felix-kolbe, 2024-09-03:
> Use `cargo install cargo-espflash --locked` to avoid the error

This also solved it for me, trying to install it in WSL (running Ubuntu 24.04.1 LTS). 🎉

VorpalBlade, 2024-09-20:
Seems gix broke semver then? Has this been reported to that project so they can yank the bad versions?

eklipse2k8, 2024-09-22:
Seeing this when running 

```sh
cargo install cargo-espflash --locked
```

```
warning: package `bytemuck v1.16.0` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
```

isaac-mcfadyen, 2024-10-14:
Solution above with `--locked` is working but bytemuck is indeed yanked—if PR #666 fixes it then it might just not be on `crates.io` yet?

jessebraham, 2024-10-14:
We will try to get a new release out soon, our focus has been elsewhere lately. Sorry for any inconvenience, I will update/close this issue when we have published a new version.

PR #687 (2024-10-18): New releases
These patch releases just update the dependencies and resolve the outstanding issues due to yanked dependencies. Some minor changes were merged previously, see `CHANGELOG.md` for more information.

Closes #671
closed 2024-10-18 by SergioGasquez