esp-rs/espflash #677: Errors flashing Waveshare ESP32-S3-Touch-LCD-7 (ver1.1)

robert-chiniquy, 2024-09-04:
Ran into this `MemData` error: https://github.com/esp-rs/espflash/issues/637 and, after adding `--no-stub`, am now getting this `FlashDeflData` error:

```
[2024-09-04T21:20:42Z INFO ] Connecting...
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f0:f5:bd:4b:9f:24
App/part. size:    524,400/1,048,576 bytes, 50.01%
Error:   × The bootloader returned an error
  ├─▶ Error while running FlashDeflData command
  ╰─▶ Other
```

I've experimented with flash mode, no-reset, and baud rates, running out of ideas.

Thanks!

robert-chiniquy, 2024-09-04:
In case it helps, here's the `RUST_LOG=debug` output:

```
[2024-09-04T21:23:25Z INFO ] Connecting...
[2024-09-04T21:23:25Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2024-09-04T21:23:25Z DEBUG] Boot Mode: 0x0
[2024-09-04T21:23:25Z DEBUG] Download Mode: false
[2024-09-04T21:23:25Z DEBUG] Writing command: Sync
[2024-09-04T21:23:25Z DEBUG] Writing command: Sync
[2024-09-04T21:23:25Z DEBUG] Writing command: ReadReg { address: 1073745920 }
[2024-09-04T21:23:25Z DEBUG] Attempting flash enable with: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 }
[2024-09-04T21:23:25Z DEBUG] Writing command: SpiAttach { spi_params: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 } }
[2024-09-04T21:23:25Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-09-04T21:23:25Z DEBUG] Writing command: ReadReg { address: 1610620960 }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2415919104, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048351, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620968, value: 23, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610621016, value: 0, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620928, value: 262144, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: ReadReg { address: 1610620952 }
[2024-09-04T21:23:25Z DEBUG] Writing command: ReadReg { address: 1610621016 }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620952, value: 2147483648, mask: None }
[2024-09-04T21:23:25Z DEBUG] Writing command: WriteReg { address: 1610620960, value: 1879048192, mask: None }
[2024-09-04T21:23:26Z DEBUG] Flash detect OK!
[2024-09-04T21:23:26Z DEBUG] Writing command: SpiSetParams { spi_params: SpiSetParams { fl_id: 0, total_size: 8388608, block_size: 65536, sector_size: 4096, page_size: 256, status_mask: 65535 } }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641476 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641480 }
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f0:f5:bd:4b:9f:24
[2024-09-04T21:23:26Z DEBUG] Writing command: SpiAttach { spi_params: SpiAttachParams { clk: 0, q: 0, d: 0, hd: 0, cs: 0 } }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641496 }
[2024-09-04T21:23:26Z DEBUG] Writing command: ReadReg { address: 1610641488 }
App/part. size:    524,400/1,048,576 bytes, 50.01%
[2024-09-04T21:23:26Z DEBUG] Writing command: FlashMd5 { offset: 0, size: 21008 }
[2024-09-04T21:23:26Z DEBUG] Writing command: FlashDeflBegin { size: 21008, blocks: 14, block_size: 1024, offset: 0, supports_encryption: true }
[2024-09-04T21:23:26Z DEBUG] Writing command: FlashDeflData { data: [120, 218, 173, 123, 11, 124, 19, 85, 214, 248, 157, 201, 163, 105, 58, 64, 250, 32, 22, 168, 56, 73, 73, 105, 145, 98, 218, 2, 45, 8, 75, 82, 106, 161, 8, 90, 42, 20, 68, 252, 72, 218, 78, 104, 177, 47, 218, 148, 215, 118, 215, 76, 171, 37, 69, 248, 182, 45, 238, 210, 186, 184, 59, 19, 141, 20, 86, 62, 74, 125, 33, 218, 181, 97, 181, 139, 10, 44, 160, 174, 168, 176, 86, 52, 174, 242, 146, 85, 250, 72, 31, 153, 255, 185, 51, 105, 41, 8, 174, 251, 125, 127, 126, 156, 204, 220, 115, 239, 61, 247, 220, 123, 206, 61, 247, 156, 115, 167, 223, 202, 73, 90, 213, 52, 199, 116, 5, 33, 20, 12, 128, 114, 241, 15, 34, 34, 83, 78, 204, 235, 213, 34, 36, 192, 191, 9, 128, 192, 16, 21, 128, 216, 76, 57, 234, 157, 44, 67, 4, 110, 121, 149, 176, 224, 135, 28, 64, 1, 160, 4, 8, 2, 48, 151, 151, 51, 101, 118, 218, 102, 45, 40, 100, 242, 232, 130, 98, 218, 80, 62, 21, 96, 182, 33, 143, 142, 53, 148, 199, 141, 82, 35, 107, 78, 73, 153, 61, 54, 142, 222, 104, 45, 167, 115, 173, 133, 184, 153, 213, 78, 103, 206, 167, 141, 155, 12, 198, 148, 77, 208, 34, 167, 164, 196, 142, 38, 60, 98, 188, 55, 41, 161, 232, 62, 232, 86, 88, 17, 135, 73, 208, 133, 37, 214, 60, 186, 212, 90, 102, 47, 176, 23, 148, 20, 211, 118, 107, 78, 33, 67, 51, 101, 101, 37, 101, 58, 104, 93, 164, 70, 116, 65, 57, 93, 92, 98, 167, 49, 1, 92, 121, 43, 34, 233, 214, 92, 123, 73, 217, 102, 218, 90, 90, 122, 157, 150, 161, 92, 34, 240, 19, 237, 237, 76, 185, 253, 103, 119, 122, 112, 153, 249, 198, 182, 116, 121, 33, 240, 101, 200, 187, 161, 79, 98, 81, 198, 136, 62, 105, 5, 229, 192 [... cut ...] 8, 62, 86, 14, 170, 9, 10, 11, 203, 83, 178, 241, 39, 44, 66, 70, 113, 110, 73, 89, 25, 147, 107, 191, 161, 231, 109, 155, 223, 114, 50, 160, 159, 5, 185, 116, 113, 69, 81, 14, 83, 246, 211, 22, 254, 230, 35, 58, 223, 138, 143, 101, 16, 90, 25, 236, 38, 171, 184, 67, 240, 193, 6, 26, 133, 15, 107, 145, 186, 68, 139, 41, 47, 93, 35, 238, 131, 91, 81], pad_to: 0, pad_byte: 255, sequence: 0 }
Error:   × The bootloader returned an error
  ├─▶ Error while running FlashDeflData command
  ╰─▶ Other
```

SergioGasquez, 2024-09-05:
Hi! Which `espflash` version are you using? Have you tried using `esptool` or `espflash@2.1`?

robert-chiniquy, 2024-09-05:
I've been using `espflash@3.1.1`, I'll see if I can try 2.1.

robert-chiniquy, 2024-09-05:
2.1 looks the same except `FlashDeflData` is `FlashDeflateData`.

robert-chiniquy, 2024-09-05:
With `esptool.py` 4.7.0, I can't find a way to disable use of a stub, so this may not be helpful, but:
```
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: f0:f5:bd:4b:9f:24
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

SergioGasquez, 2024-09-06:
> With `esptool.py` 4.7.0, I can't find a way to disable use of a stub, so this may not be helpful, but:

There is also a `--no-stub` flag: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flasher-stub.html#disabling-the-stub-loader

robert-chiniquy, 2024-09-06:
Ah thank you, I don't know how I missed that.

I'm still checking my work, maybe my offsets are wrong? With lower baud rates I just get a timeout. The cable is known good, I can flash all my other boards with it. But these are the errors I'm currently encountering with esptool:

```sh
# espflash board-info  --no-stub
[2024-09-06T16:05:14Z INFO ] 🚀 A new version of espflash is available: v3.1.1
[2024-09-06T16:05:14Z INFO ] Detected 4 serial ports
[2024-09-06T16:05:14Z INFO ] Ports which match a known common dev board are highlighted
[2024-09-06T16:05:14Z INFO ] Please select a port
[2024-09-06T16:05:15Z INFO ] Serial port: '/dev/cu.usbmodem58750032371'
[2024-09-06T16:05:15Z INFO ] Connecting...
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       f0:f5:bd:4b:9f:24

# esptool.py --no-stub --chip esp32s3 --port /dev/cu.usbmodem58750032371 --baud 57600 write_flash \
 0x1000 target/xtensa-esp32s3-espidf/release/bootloader.bin \
 0x8000 target/xtensa-esp32s3-espidf/release/partition-table.bin \
 0x10000 target/xtensa-esp32s3-espidf/release/try-esplash.bin
esptool.py v4.7.0
Serial port /dev/cu.usbmodem58750032371
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: f0:f5:bd:4b:9f:24
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00072fff...
Erasing flash...
Took 0.09s to erase flash block
Writing at 0x00001000... (4 %)
A fatal error occurred: Failed to write to target Flash after seq 0 (result was 01050000: Requested resource not found)

# esptool.py --chip esp32s3 --port /dev/cu.usbmodem58750032371 --baud 57600 write_flash \
 0x1000 target/xtensa-esp32s3-espidf/release/bootloader.bin \
 0x8000 target/xtensa-esp32s3-espidf/release/partition-table.bin \
 0x10000 target/xtensa-esp32s3-espidf/release/try-esplash.bin
esptool.py v4.7.0
Serial port /dev/cu.usbmodem58750032371
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: f0:f5:bd:4b:9f:24
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

SergioGasquez, 2024-09-09:
> I'm still checking my work, maybe my offsets are wrong?

Try using the following offsets https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/flashing-firmware.html?highlight=0x8000#id1 (bootloader at 0x0, partition table a 0x8000 and app at 0x10000)

robert-chiniquy, 2024-09-10:
```sh
esptool.py --no-stub --chip esp32s3 --port /dev/cu.usbmodem58750032371 --baud 57600 write_flash \
 0x0000 target/xtensa-esp32s3-espidf/release/bootloader.bin \
 0x8000 target/xtensa-esp32s3-espidf/release/partition-table.bin \
 0x10000 target/xtensa-esp32s3-espidf/release/try-esplash.bin
esptool.py v4.7.0
Serial port /dev/cu.usbmodem58750032371
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: f0:f5:bd:4b:9f:24
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00072fff...
Erasing flash...
Took 0.09s to erase flash block
Writing at 0x00000000... (4 %)
A fatal error occurred: Failed to write to target Flash after seq 0 (result was 01050000: Requested resource not found)
```

SergioGasquez, 2024-09-10:
Do you have any other devkit to try? The fact that both `esptool` and `espflash` fail makes me think it might be a hw thing.

JihoonJustin, 2024-09-12:
Hi I have a same problem

robert-chiniquy, 2024-09-12:
I think this is a pretty cool board, so I could get another one to try to verify if it is a hw issue with the original. LMK if that would help.

JihoonJustin, 2024-09-13:
@robert-chiniquy But I used to use this board with other system (Ubuntu), Now I am trying to develop on MacOS.

JihoonJustin, 2024-09-13:
I resolved this problem on mac. I have installed usb driver.
https://github.com/WCHSoftGroup/ch34xser_macos

robert-chiniquy, 2024-09-17:
Thank you. I was able to install those drivers and it did fix my issue. A new serial device was visible. 

Definitely wish the installers weren't at an `http://` link. 

But thank you  everybody for the help! I'm sorry I didn't figure this one out on my own!

closed 2024-09-17 by robert-chiniquy