esp-rs/espflash #681: espflash monitor takes over the terminal

cacharle, 2024-09-16:
It happens often that I cannot kill `espflash monitor` with `Ctrl-C` or `Ctrl-\` and the logs stop coming aswell. I'm not sure if my code crashed or if it's the connection to the device but it would be nice to have some error message or at least be able to kill the process instead of having to relaunch my terminal.

I use alacritty as a terminal btw:

```
❯ alacritty --version
alacritty 0.13.2 (bb8ea18e)
```

```
❯ espflash --version
espflash 3.1.0
```

SergioGasquez, 2024-09-17:
Seems similar to https://github.com/esp-rs/espflash/issues/464, I do use the same `allacritty` version on Linux and I don't recall encountering this issue.

cacharle, 2024-09-17:
My colleagues and I have observed that behavior across different terminals (Windows terminal, ubuntu's terminal, alacritty) and different machines with different OSes (Linux, Windows)

I really seems independent of the platform you're running on

cacharle, 2024-10-04:
Similar behavior when it cannot find a usb:

```
$ espflash monitor --log-format defmt --elf <elf>
[2024-10-04T15:20:49Z INFO ] 🚀 A new version of espflash is available: v3.1.1
[2024-10-04T15:20:49Z INFO ] Detected 3 serial ports
[2024-10-04T15:20:49Z INFO ] Ports which match a known common dev board are highlighted
[2024-10-04T15:20:49Z INFO ] Please select a port
[2024-10-04T15:20:49Z INFO ] Serial port: '/dev/ttyACM0'
[2024-10-04T15:20:49Z INFO ] Connecting...
r^C^C^C^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^\^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^Z^Z^Z^Z
```

cacharle, 2024-10-16:
This bug is the bane of my existence, any clue where I could start looking for it in the espflash codebase?

SergioGasquez, 2024-10-16:
I don't have many pointers, but here is where we wait for the Ctrl+C press: https://github.com/esp-rs/espflash/blob/main/espflash/src/cli/monitor/mod.rs#L118-L131. Might be related to the lib we use to process them (`crossterm`), we have an “outdated” version since new version were causing some issues on Windows, https://github.com/esp-rs/espflash/blob/main/espflash/Cargo.toml#L36

dbrgn, 2024-12-15:
I'm also experiencing this bug. Sometimes I after running `espflash flash --baud 2000000 --monitor` I cannot abort the process with Ctrl+C and no log output updates are shown (but the device works as intended).

Something I noticed is that if I reset the microcontroller using the restet button, all the "missing output" suddenly shows up, followed by `× Broken pipe`. Maybe some stdout flushing isn't being done?

dbrgn, 2024-12-15:
Oh, I can trigger the issue if I simply type a few characters with focussed terminal while logging is happening (e.g. "asdf" in quick succession). If I do that, the terminal output freezes and ctrl+c doesn't work anymore either.

SergioGasquez, 2025-09-01:
A follow-up and some more insight of the investigation that mainly @jessebraham carried. 
- The code hangs in [this `flush`](https://github.com/esp-rs/espflash/blob/v4.0.1/espflash/src/cli/monitor/mod.rs#L177)
- It only happens when using USB-Serial-JTAG, UART works fine
- From our test we encountered that it happens: 
  - After 2 key presses in Linux
  - After 10 key presses in MacOS

@sirhcel do you have any idea of what could be happening and making the `flush` call hang the terminal after some keys are pressed?

closed 2025-09-16 by MabezDev