esp-rs/espflash #688: Add an option to prevent checking for new version

Globibluk, 2024-10-16:
Hi, I'm working behind a proxy and it seems the programs tries to connect to crates.io in order to find updates.
As the requests times out, it takes a lot of time to even scan a board.
Is it possible to add an option to prevent such connections?

Thanks in advance!

SergioGasquez, 2024-10-16:
Hi! Currently, we don't have any way of preventing this check, but we would happily accept a contribution adding an argument that skips calling https://github.com/esp-rs/espflash/blob/main/espflash/src/bin/espflash.rs#L167 (this should be also duplicated in cargo-espflash)

Thrathra, 2024-10-17:
Thanks for your suggestion, I didn't see the right line at first. I opened a pull request.

PR #689 (2024-10-18): Add skip update check option
Add a new global option to prevent espflash from checking updates.
It will make it work faster behind a proxy.

Closes #688
closed 2024-10-18 by jessebraham