esp-rs/espflash #690: Error when installing 2.1.0 for ESP8266 support

andreas-mausch, 2024-10-19:
I try to use `cargo espflash` to flash my ESP8266.
I understand the recent version does not support the ESP8266, but the 2.x versions should (see here: https://github.com/esp-rs/espflash/issues/519)

My setup:
cargo 1.81.0
rustc 1.81.0

Error:

```shell-session
$ cargo install cargo-espflash@2.1.0 --locked
[...]
   Compiling minimal-lexical v0.2.1
error[E0282]: type annotations needed for `Box<_>`
  --> /home/neonew/.cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.3.29/src/format_description/parse/mod.rs:83:9
   |
83 |     let items = format_items
   |         ^^^^^
...
86 |     Ok(items.into())
   |              ---- type must be known at this point
   |
   = note: this is an inference error on crate `time` caused by an API change in Rust 1.80.0; update `time` to version `>=0.3.35` by calling `cargo update`
```

Do you have an idea how to fix that? Downgrading my rustc?
Last commit to minimal-lexical was 2 years ago.

I can use the binary version (`cargo binstall cargo-espflash@2.1.0`), but still I'd prefer to build it myself.

jessebraham, 2024-10-20:
You should be able to download a pre-compiled binary from the release page, assuming you are running a supported platform:  
https://github.com/esp-rs/espflash/releases/tag/v2.1.0

If this is not possible, or does not work for some reason, I would suggest installing an older version of the toolchain and trying to install `cargo-espflash` using it:

```shell
$ rustup toolchain install 1.79.0
$ cargo +1.79.0 install cargo-espflash --locked
```

Hopefully this will work, I'm just kind of assuming it will; I have not run these commands locally. Hopefully you are successful using either of these methods.

andreas-mausch, 2024-10-21:
I can confirm with Rust 1.79.0 it worked to build from source.
It is a bit sad I can't use the recent version. I have understanding for it, though.

Thank you

jessebraham, 2024-10-21:
> It is a bit sad I can't use the recent version. I have understanding for it, though.

Yes I understand. Unfortunately, we have limited resources and given the nearing EOL of the ESP8266 and lack of any similarities to the other chips, we have opted not to support it.

Glad to hear you ultimately got things working.

closed 2024-10-21 by jessebraham