esp-rs/espflash #695: Implement application description for ESP-IDF image format

jessebraham, 2024-10-25:
Details can be found in the ESP-IDF documentation:  
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#application-description

The omission of this struct can cause issues when using newer versions of the bootloader.

SergioGasquez, 2024-11-27:
Here is my progress with this issue: https://github.com/SergioGasquez/espflash/tree/feat/app-desc. Not sure what's wrong, but the app bootloops because the checksum verification fails or because there is an invalid segment length.

Note that I've only updated the C6 bootloader

ivmarkov, 2025-01-13:
@SergioGasquez I wonder why are you guys implementing this structure as part of `espflash` itself?
My understanding is, it needs to be a part of the factory/OTA image itself?

In `esp-idf-sys`, we have a simple `app_desc!` macro that folks can use to embed their own `esp_app_desc_t` structure in the ELF executable, which grabs current date and time, project = crate name (from `Cargo.toml`), and project version = crate version from `Cargo.toml`.

Shouldn't we instead a similar approach for baremetal?
I mean, if you upload via OTA a new app image, you want this new app image to have a new version of this structure anyway, so my understanding is this is not related to `espflash`, which dopes not play a role during OTA anyway.

SergioGasquez, 2025-02-03:
We created an issue on esp-hal to investigate implementing it there

ivmarkov, 2025-02-03:
Related: #741

jessebraham, 2025-02-11:
I guess the plan is to handle this in the HAL now, @SergioGasquez should we just close this?