esp-rs/espflash #696: Implement NVS Partition Facility

DaneSlattery, 2024-10-25:
It would be useful to not have multiple separate tools that need to interact to create images. This tool already supports converting elf to bin, csv to bin, now we could support generating data partitions:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/mass_mfg.html
[or ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_partition_gen.html)

DaneSlattery, 2024-10-25:
It would also be incredibly useful to be able to write postcard structs into nvs directly.

juliankrieger, 2024-12-06:
Are there any pointers on where to begin for this issue? I would be happy to contribute.

renkenono, 2026-01-09:
Integration of [esp-nvs](https://github.com/lhemala/esp-nvs) *should* facilitate the implementation of this feature.