esp-rs/espflash #697: Cannot read flash partition

DaneSlattery, 2024-10-25:
I cannot read flash partitions with espflash. 

```
espflash read-flash --baud 9600 --block-size 500 0x9000 0x6000 ./nvs_dump_devkit
[2024-10-25T09:56:39Z INFO ] 🚀 A new version of espflash is available: v3.2.0
[2024-10-25T09:56:39Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-10-25T09:56:39Z INFO ] Connecting...
[2024-10-25T09:56:40Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       30:30:f9:55:19:54
Error: espflash::read_flash::corrupt_data

  × Corrupt data, expected 1f4 bytes but receved 1b7 bytes
```

I am able to do the same with esptool.py:
```
esptool.py read_flash 0x9000 0x6000 ./nvs_dump.bin
esptool.py v4.8.1
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 30:30:f9:55:19:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
24576 (100 %)
24576 (100 %)
Read 24576 bytes at 0x00009000 in 2.2 seconds (90.0 kbit/s)...
Hard resetting via RTS pin...

```

I don't know what is missing?

MabezDev, 2024-10-25:
Can you try again with `--no-stub`?

zakimoo, 2024-11-21:
same hier

```
espflash read-flash 0x0 0x0400000 test_3.bin

[2024-11-21T14:48:38Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-11-21T14:48:38Z INFO ] Connecting...
[2024-11-21T14:48:38Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       a0:76:4e:5a:41:70
Error: espflash::read_flash::corrupt_data

  × Corrupt data, expected 1000 bytes but receved fcf bytes
```
but with esptool.py it works

```
esptool.py -b 460800 read_flash 0 0x0400000 flash_contents.bin
esptool.py v4.8.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: a0:76:4e:5a:41:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
4194304 (100 %)
4194304 (100 %)
Read 4194304 bytes at 0x00000000 in 106.9 seconds (314.0 kbit/s)...
Hard resetting via RTS pin...
```

liclac, 2025-02-02:
Same here, also trying to dump the default NVS partition of my ESP32-S3 devboard (a [Heltec LoRa 32 V3](https://de.aliexpress.com/item/1005007752194012.html)).

```
$ espflash read-flash 0x9000 24576 config2.img
[2025-02-02T17:15:59Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-02-02T17:15:59Z INFO ] Connecting...
[2025-02-02T17:15:59Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       34:cd:b0:3d:40:24
Error: espflash::read_flash::corrupt_data

  × Corrupt data, expected 1000 bytes but receved ffe bytes
```

Here's the full log, when running with `RUST_LOG=trace`: https://gist.github.com/liclac/01ee6b5f5bea1095cf37c86c43fb9f60

Doing the same operation with `esptool.py` works:

```
$ esptool.py read_flash 0x9000 24576 config.img
esptool.py v4.8.1
Found 3 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: 34:cd:b0:3d:40:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
24576 (100 %)
24576 (100 %)
Read 24576 bytes at 0x00009000 in 2.2 seconds (90.3 kbit/s)...
Hard resetting via RTS pin...
```

I am able to use `espflash write-bin` to write this dumped image back to it:

```
$ espflash write-bin 0x9000 config.img
[2025-02-02T17:24:06Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-02-02T17:24:06Z INFO ] Connecting...
[2025-02-02T17:24:07Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       34:cd:b0:3d:40:24
[00:00:00] [========================================]       1/1       0x9000
[2025-02-02T17:24:09Z INFO ] Binary successfully written to flash!
```

I'm also able to change the data, write it, and read it back just fine:

```
$ tr \xFF \xAA < config.img > config2.img

$ espflash write-bin 0x9000 config2.img
[2025-02-02T17:26:27Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-02-02T17:26:27Z INFO ] Connecting...
[2025-02-02T17:26:27Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       34:cd:b0:3d:40:24
[00:00:00] [========================================]       1/1       0x9000
[2025-02-02T17:26:29Z INFO ] Binary successfully written to flash!

$ esptool.py read_flash 0x9000 24576 config2_.img
esptool.py v4.8.1
Found 3 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: 34:cd:b0:3d:40:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
24576 (100 %)
24576 (100 %)
Read 24576 bytes at 0x00009000 in 2.2 seconds (90.3 kbit/s)...
Hard resetting via RTS pin...

$ diff <(xxd config2.img) <(xxd config2_.img)

$ $ espflash read-flash 0x9000 24576 config.img
[2025-02-02T17:31:33Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-02-02T17:31:33Z INFO ] Connecting...
[2025-02-02T17:31:33Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       34:cd:b0:3d:40:24
Error: espflash::read_flash::corrupt_data

  × Corrupt data, expected 1000 bytes but receved ffd bytes
```

A minor observation is that for some reason, every time I re-write the partition and try reading it again, it changes how many bytes the error message says it received, just by a few bytes back or forth seemingly at random (I've seen `ffc`, `ffd` and `ffe`, in no particular order).

Happy to provide any additional information, help debug things, etc., but I'm a bit lost as to where to even start with this.

liclac, 2025-02-03:
I discovered something very interesting: it seems to work fine as long as I ask for `0x1000` bytes or less. The moment I go above that, things start failing - `0x1001` through `0x1009` gets a timeout, `0x1010` and I'm back to the corruption error.

SergioGasquez, 2025-02-06:
>  but I'm a bit lost as to where to even start with this.

I guess we should compare both implementations: 
 - `esptool`: https://github.com/espressif/esptool/blob/master/esptool/loader.py#L1273-L1308
 - `espflash`: https://github.com/esp-rs/espflash/blob/main/espflash/src/flasher/mod.rs#L1106-L1185

SergioGasquez, 2025-02-10:
I think we encountered this issue on our HIL setup:
- When using [`espflash read-flash 0x1000 0x200 flash_content.bin`](https://github.com/esp-rs/espflash/pull/754/commits/e8e9c668fb70dfe552cc2bab5739b1907027ffc2#diff-368fa112ae7912cd621beba4005a42674b3c748d7598462a3f39f9a95270d3e7R8), the CI passes: https://github.com/esp-rs/espflash/actions/runs/13242998308
- When using [`espflash read-flash 0 0x2000 flash_content.bin`](https://github.com/esp-rs/espflash/pull/754/commits/afc364122c0fabf523f17d13bd112dbf1abed97d#diff-368fa112ae7912cd621beba4005a42674b3c748d7598462a3f39f9a95270d3e7R8), the CI fails: https://github.com/esp-rs/espflash/actions/runs/13243719658/job/36965109560
```
Error: espflash::read_flash::corrupt_data

  × Corrupt data, expected 1000 bytes but receved fc3 bytes
```

Locally, I can reproduce the issue on some devkits, but in other devkits it works fine:
- ESP8684-DevKitC-02 26MHz (clock): works fine
- ESP32-C6-DevKitC-1: Fails
- ESP32-S3-DevKitC-1: Fails

jessebraham, 2025-02-11:
Small update, after some investigation this appears to be an issue with the flasher stub. I was able to successfully read sizes larger than 0x1000 by using the old C stub. Will try to hunt down what's going wrong there.

closed 2025-02-12 by SergioGasquez