esp-rs/espflash #704: Add functionality similar to `get_security_info` command in `esptool`

jessebraham, 2024-11-12:


ivmarkov, 2025-01-13:
Nice to have, but I think we need the full-blown capability of `espefuse.py` instead. Otherwise, there is actually no way to burn the SecureBoot V2 public key signature into efuse in the first place, and this is the one single thing the bootloader cannot do automatically "on first boot". But also in general, relying on the bootloader-compiled-with-encryption-enabled to auto-generate encryption XTS key, burn it, encrypt all plaintext bootloader/part table/factory image on first boot is super dangerous (IMO).

Espressif recommends [this workflow instead](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32s3/security/security-features-enablement-workflows.html), which relies on explicit efuse burns to enable both secure boot and encryption. Much more predictable. And allows for sub-sequent re-flashing of the encrypted/secure-booted device, as long as you keep the encryption key, the secure boot V2 private key around, and the ROM downloader enabled (in secure mode or not).

jessebraham, 2025-01-13:
This was opened because there have been change with how chip detection is performed with newer devices (e.g. [for the P4](https://github.com/espressif/esptool/blob/master/esptool/targets/esp32p4.py#L42)) and detecting via magic value does not work reliably with all devices. I did not consider any other application of this functionality when opening the issue.

ivmarkov, 2025-01-13:
> This was opened because there have been change with how chip detection is performed with newer devices (e.g. [for the [...]

Fair enough!

jessebraham, 2025-01-13:
I probably should have mentioned that in the original comment 😅