esp-rs/espflash #706: Return of direct-boot

JohnnyZig, 2024-12-05:
I like your tool, it is fast, simple, does the job and has UART monitor with reset.
But I am using it to flash bare metal programs not written in C nor Rust which don't use bootloader.
I am using direct-boot for ESP32-C3 but I keep my flasher at 074f8bc (2024.01.15) (last commit with direct-boot).
As I am preparing to jump to ESP32-P4 I am wondering if that old version will be enough for it?

Is there a possibility for return of direct-boot to the espflasher?
Thanks in advance!

igrr, 2024-12-06:
@JohnnyZig just wondering, what is the functionality you need from the flasher tool to support direct-boot binaries?
Generating direct boot binaries from ELF files can be done with objcopy (which you get along with the linker anyway), while the flashing tool doesn't have to know or care how the binary was generated.

JohnnyZig, 2024-12-06:
I am flashing without ESP bootloader (just app.elf).
Before 8814e83 I was using:
```
espflash flash --format direct-boot -p <serial port> --monitor app.elf
```
But after that commit there is no direct-boot support, espflash will flash bootloader:
```
espflash-v3.2.0 flash -p <serial port> app.elf
```
I am asking for revert of 8814e83 or a way to flash without bootloader with new espflash.

jessebraham, 2024-12-06:
We have no plans to re-add this functionality at this time. Probably adding new image formats is not currently possible without a new major release, we will have to do this for #701 at some point. Once this refactoring has been completed then we would likely accept a PR for this, however.

As @igrr mentioned, you should be able to produce a binary using `objcopy`, which should be writable using the `espflash write-bin` subcommand.

SergioGasquez, 2026-11-12:
Closed by https://github.com/esp-rs/espflash/pull/1114

closed 2026-11-12 by SergioGasquez