esp-rs/espflash #708: Flash verification false-negative (on worn out flash?)

i404788, 2024-12-18:
Hey there,

I've been working with one board, and I suspect I've worn out the flash on that board. However flashing still seems to succeed while I would assume the flash verification should fail if it doesn't receive the same value it wrote.

The only other possibility I can think of is that something is actually wrong with the bootloader/ROM/efuses (although I didn't touch those), so that seems unlikely to me. 

The change did coincide with changing espflash settings to use 80Mhz qio, however resetting to defaults did not change anything. Flashing a different board with the same firmware gives the usual defmt output. 

Let me know if there is something I'm missing or can test.

espflash.toml (in case useful):
```toml
[connection]

[[usb_device]]
vid = "0403"
pid = "6010"

[flash]
baud-rate = 115200
size = "8MB"
```

Bootloop logs (in case useful):
<details>

```
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:31:45:0c
App/part. size:    339,376/8,323,072 bytes, 4.08%
[00:00:01] [========================================]      14/14      0x0                                                                                                                      [00:00:00] [========================================]       1/1       0x8000                                                                                                                   [00:00:19] [========================================]     192/192     0x10000                                                                                                                  [2024-12-18T16:32:03Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf43f
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf43f
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf43f
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf43f
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf43f
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
```

</details>

SergioGasquez, 2025-02-06:
Can you try flashing with [`esptool`](https://github.com/espressif/esptool) to see if the result is the same? If the other board works as expected, I guess this must be a hw thing.

i404788, 2025-02-27:
I revisited this board today, it seems like flashing does work without issue now, however the boot loop still occurs sometimes (see original logs), doing a `ctrl+r` fixed it, so I'm guessing it's not flash but rather another hardware defect (maybe a near-shorted bootstrap pin or something).

Regardless doesn't seem to have to do with espflash, so closing now.

closed 2025-02-27 by i404788