esp-rs/espflash #711: Investigate if we need to depend on libudev

SergioGasquez, 2024-12-19:
`serialport` can now work without `libudev`, see https://github.com/serialport/serialport-rs/pull/220, this can lead to weird ports being shown but it might be worth to explore how this performs, some initial tests show good signs, see https://github.com/esp-rs/espflash/pull/709

jessebraham, 2024-12-19:
Do we know how/if this affects Linux systems based on MUSL? IIRC we have some custom logic for these systems in our port detection code, perhaps it is no longer necessary if `serialport` is now reading port info directly from `/sys/`.

SergioGasquez, 2024-12-19:
> Do we know how/if this affects Linux systems based on MUSL? 

No idea, @sirhcel sorry for the ping, do you have any idea on this?

sirhcel, 2024-12-20:
> Do we know how/if this affects Linux systems based on MUSL? IIRC we have some custom logic for these systems in our port detection code, perhaps it is no longer necessary if `serialport` is now reading port info directly from `/sys/`.

Especially since https://github.com/serialport/serialport-rs/pull/220 things look almost shiny even without libudev. On my test system, two serial ports from a PCIe card are missing, but `ttyS0` from the main board and the USB ports are reported identically now (as of [944e9e9](https://github.com/serialport/serialport-rs/commits/944e9e9a50164c784c8ed1e3293dd955c5ef2294/), serial numbers redacted):

```
$ cargo run --target x86_64-unknown-linux-gnu --features usbportinfo-interface --example list_ports
[...]
Found 7 ports:
    /dev/ttyS0
        Type: Unknown
    /dev/ttyS4
        Type: PCI
    /dev/ttyS5
        Type: PCI
    /dev/ttyUSB2
        Type: USB
        VID: 0403
        PID: 6001
        Interface: 00
        Serial Number: FTFOO
        Manufacturer: FTDI
        Product: UT232R
    /dev/ttyUSB3
        Type: USB
        VID: 0403
        PID: 6010
        Interface: 00
        Serial Number: FTBAR
        Manufacturer: FTDI
        Product: FT2232H MiniModule
    /dev/ttyUSB4
        Type: USB
        VID: 0403
        PID: 6010
        Interface: 01
        Serial Number: FTBAR
        Manufacturer: FTDI
        Product: FT2232H MiniModule
    /dev/ttyUSB5
        Type: USB
        VID: 10c4
        PID: ea60
        Interface: 00
        Serial Number: YOURCP2102NSERIALHERE
        Manufacturer: Silicon Labs
        Product: CP2102N USB to UART Bridge Controller
```
```
$ cargo run --target x86_64-unknown-linux-musl --no-default-features --features usbportinfo-interface --example list_ports
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.02s
     Running `target/x86_64-unknown-linux-musl/debug/examples/list_ports`
Found 5 ports:
    /dev/ttyS0
        Type: Unknown
    /dev/ttyUSB2
        Type: USB
        VID: 0403
        PID: 6001
        Interface: 00
        Serial Number: FTFOO
        Manufacturer: FTDI
        Product: UT232R
    /dev/ttyUSB3
        Type: USB
        VID: 0403
        PID: 6010
        Interface: 00
        Serial Number: FTBAR
        Manufacturer: FTDI
        Product: FT2232H MiniModule
    /dev/ttyUSB4
        Type: USB
        VID: 0403
        PID: 6010
        Interface: 01
        Serial Number: FTBAR
        Manufacturer: FTDI
        Product: FT2232H MiniModule
    /dev/ttyUSB5
        Type: USB
        VID: 10c4
        PID: ea60
        Interface: 00
        Serial Number: YOURCP2102NSERIALHERE
        Manufacturer: Silicon Labs
        Product: CP2102N USB to UART Bridge Controller
``` 

Do you have some secret sauce for PCI port detection in place at espflash you can give we a reference to? I would love to get the port enumeration with and without libudev an a par.

SergioGasquez, 2024-12-24:
> Do you have some secret sauce for PCI port detection in place at espflash you can give we a reference to? I would love to get the port enumeration with and without libudev at a par.

[Here is where all the finding/selecting of serial ports](https://github.com/esp-rs/espflash/blob/main/espflash/src/cli/serial.rs)  happens, the only "magic" that we do is with linux-musl that we parse sysfs, see the two implementations of `detect_usb_serial_ports`

sirhcel, 2025-01-04:
Thank you for the reference @SergioGasquez! At a first glance, it looks like the USB ports enumerated by espflash should be enumerated by serialport-rs as well. But the enumeration will additionally return some more ports from the motherboard of my Linux machine from https://github.com/esp-rs/espflash/issues/711#issuecomment-2557658853. So the result from using the non-libudev enumeration from serialport-rs will be "between" espflash's enumeration without and with libudev.

The discrepancy between the devices enumerated by libudev and our manual for MUSL/non-libudev is bugging me since a while and I found some time to look into it and https://github.com/serialport/serialport-rs/pull/238 sets out to close this gap and should bring the enumeration en par with libudev.

Do you have some interesting corner cases from the enumeration done by espflash? May be there is something relevant for serialport-rs too.

SergioGasquez, 2025-01-09:
> Do you have some interesting corner cases from the enumeration done by espflash?

I dont recall any

https://github.com/serialport/serialport-rs/pull/238 looks promising, maybe we can do some tests after it lands. We are working on planning espflash v4, so this should definitely be investigated.