esp-rs/espflash #726: `espflash write-bin` not compatible with Secure Download mode

ivmarkov, 2025-01-19:
Once [Secure Download mode](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32s3/security/flash-encryption.html#best-practices) is enabled on the chip, no matter what I do I can't force the chip into writing a binary image with `espflash`.

Some commands I tried:
```sh
espflash write-bin -b no-reset-no-sync --chip esp32s3 0x10000 factory.bin
espflash write-bin -b no-reset-no-sync --chip esp32s3 0x10000 factory.bin
espflash write-bin -b no-sync --chip esp32s3 0x10000 factory.bin
espflash write-bin -b no-reset --chip esp32s3 0x10000 factory.bin
espflash write-bin -b usb-reset --chip esp32s3 0x10000 factory.bin
```

These fail either with:
```sh
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

or with:
```sh
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running ReadReg command
```

This ^^^ failure mode is interesting, because it seems `espflash` is capable of setting the chip in download mode after all (with some reset sequences), but then (I'm speculating) it tries to read (some?) register, which might not be supported in secure DL mode.

=========

### `esptool`

In contrast, `esptool` (a) is capable to detect that the chip is in Security Download mode (try `esptools tool get_security_info`) and (b) capable of writing to the flash (albeit with the default baud rate):
```sh
esptools tool write_flash 0x10000 factory.bin --force
```

ivmarkov, 2025-01-19:
For completeness, just to mention that I managed to convince the monitor to show logs when the chip is in secure DL mode by using:
```sh
espflash monitor -b no-reset-no-sync --chip esp32s3
```

chris-subtlebytes, 2025-01-19:
I hit this a while ago too and I think I just switched to esptool for the one device I accidentally enabled secure download mode on.

For my future development, I added this to my `sdkconfig.defaults` to prevent secure DL mode from enabling:

```
CONFIG_SECURE_BOOT_INSECURE=y # required for CONFIG_SECURE_INSECURE_ALLOW_DL_MODE to actually enable
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
CONFIG_SECURE_BOOT_ALLOW_JTAG=y # probably unnecessary, doubly so because gdb keeps crashing on me.
```

ivmarkov, 2025-01-19:
> I hit this a while ago too and I think I just switched to esptool for the one device I accidentally enabled secure download mode on.
> 
> For my future development, I added this to my `sdkconfig.defaults` to prevent secure DL mode from enabling:
> 
> ```
> # required for CONFIG_SECURE_INSECURE_ALLOW_DL_MODE to actually enable
> CONFIG_SECURE_BOOT_INSECURE=y
> 
> # prevent secure DL mode
> CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
> 
> # probably unnecessary, doubly so because gdb keeps crashing on me.
> CONFIG_SECURE_BOOT_ALLOW_JTAG=y
> ```
> 
> Just make sure the local changes never get committed to release. Although I think it's not _so_ bad security-wise if they do.

Sure, I have the exact same settings in my debug builds for development, but for release I actually do want secure DL mode, because it is, well, secure.

And yes, `esfactory` uses `esptool` now as a (temporary) backup.