esp-rs/espflash #738: Unable to run released binary on a Raspberry Pi 3B (Raspbian 11 = "bullseye")

lure23, 2025-02-02:
I'm mostly using `probe-rs` but would now need to set up `espflash` as well, on a Raspberry Pi.

I noticed there's a [prebuilt release of 3.3.0](https://github.com/esp-rs/espflash/releases/tag/v3.3.0) for `armv7-unknown-linux-gnueabihf` - GREAT!

Downloaded, `scp`:ed over to the Pi, launching gives:

```
$ ./espflash 
./espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by ./espflash)
./espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.32' not found (required by ./espflash)
./espflash: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.34' not found (required by ./espflash)
```

This looks similar to #454 - which was [resolved](https://github.com/esp-rs/espflash/issues/454#issuecomment-1723550074).

My options are:

a) upgrading to `Bookworm` - would like to avoid that, since it seems like a full reinstall. My Pi is stable as it is, and I have no other urgent need to upgrade.
b) cross-compile; what I do for `probe-rs`

Perhaps my question is: is the 3.3.0 binary release expected to work on Raspbian OS 11 ("bullseye")?

**system info**

```
$ uname -a
Linux rpi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
```

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

lure23, 2025-02-02:
As an aside-note, I also tried cross-compiling:

```
$ git clone https://github.com/esp-rs/espflash.git
$ cd espflash
$ cross build --release --target=armv7-unknown-linux-gnueabihf
[...]
   Compiling nix v0.29.0
   Compiling proc-macro-crate v1.3.1
   Compiling clap_builder v4.5.26
   Compiling tracing v0.1.41
   Compiling wyz v0.5.1
error: failed to run custom build command for `libudev-sys v0.1.4`
```

>Note: I have `pkgconfig` and `libudev-dev` installed. What's missing is likely some headers for `armv7-unknown-linux-gnueabihf` target.

Vollbrecht, 2025-02-03:
You don't need a full reinstall to updated a debian installation ;D I have a server running that started as debian8 and is running 12 now only updating via apt :p 

But problem aside in the past i cross compiled probe-rs for raspberry pi3b+ via [cross](https://github.com/cross-rs/cross). Building `espflash` should be even simpler. It just was something like `cross build --target armv7-unknown-linux-gnueabihf --release` but you need to obvious make sure that the container build is using a older glibc that you are running on your pi with debian11.

SergioGasquez, 2025-02-03:
> would the released armv7-unknown-linux-gnueabihf binary be expected to work on "bullseye" (11) or does it require a later OS. 

It requires a later OS, as it was built with gh-hosted-runners that don't support that GLIBC version anymore.

closed 2025-02-03 by lure23