esp-rs/espflash #770: Support writing binary images directly from flash command.

AVee, 2025-02-14:
Currently the flash command expects an ELF image to be flashed to the device. This makes sense in the context of a cargo project since the output there will always be an ELF image. But on the command line it would be useful to be able to directly flash an already converted image. In our case we already build those for OTA updates and it would be convenient to be able to use the same files when flashing directly.

It looks to me like it's should be doable to just detect the type of file passed to the flash command based on the magic bytes in the file. An ELF file could be converted and flashed, but when a ESP binary is found it's just flashed directly. All of that would then be transparent for the user and 'just work' regardless of the type of image used. Would this be a sensible approach, or would a more explicit (separate command or extra argument) be preferred?

This might be a solution to #684 as well.

AVee, 2025-02-17:
I made a working POC by using code from the `image_format` module to implement the `FirmwareImage` trait. That works, but does a lot of needless work splitting the file into sectors and then merging them back again the exact same way. 

@jessebraham A clean solution for this would involve some refactoring of that code, probably in a way where a lot of handling moves into the FirmwareImage trait and it implementers. I'd be willing to give that a go, but I don't know if you have anything planned for #772 in that area.

jessebraham, 2025-02-24:
Sorry for the late response.

With regards to your first comment, I think is is reasonable enough functionality to add. I guess it overlaps a bit with the current `write-bin` subcommand, which may be a bit confusing, but I think that this subcommand is maybe not the most intuitive to begin with, so maybe no an issue.

I do plan on doing more refactoring here, but I need to spend some time thinking about how things should look. There are still some details I'm unsure of. If you have any ideas then I think for now, feel free to go ahead and implement them; we can always make additional changes after the fact if needed. If you don't have the time/energy to implement it yourself, feel free to leave some notes here and I can look into it at some point.

AVee, 2025-11-27:
Given the refactor, would the most sensible thing to do now be to get support for esp binary images in the object library? Or would you go another direction?