esp-rs/espflash #781: add defmt "--log-format" option

CvRXX, 2025-02-24:
With probe-rs you can pass the argument `--log-format` as described in https://defmt.ferrous-systems.com/custom-log-output#passing-log-format-to-printers. Would it be possible to add this to espflash as well so we can edit the log output?

jessebraham, 2025-02-26:
At this time we have a fair bit of work ahead of us here, so I'm not sure we will prioritize this right now. However, I think eventually this would be nice functionality to add. We'd be happy to accept a PR from the community, otherwise one of us can revisit this once our current work has been completed.

SergioGasquez, 2025-02-26:
#520 is basically the same issue

jessebraham, 2025-02-26:
> [#520](https://github.com/esp-rs/espflash/issues/520) is basically the same issue

They're related but different, if I'm understanding correctly. In the linked issue they are asking for a different logging format, this issue is asking for support for providing an arbitrary format string.

CvRXX, 2025-02-28:
I would be happy to write a PR. Any things I should consider? I'm a beginning rust programmer (month of experience or so) but am experienced in modern c++

SergioGasquez, 2025-02-28:
> Any things I should consider?

I guess this would only apply when using `--log-format defmt`, maybe we should use `--defmt-format` name (?). I would suggest you look at `probe-rs` source for inspiration of this feature, here is my old branch trying to close #520: https://github.com/esp-rs/espflash/compare/main...SergioGasquez:espflash:feat/defmt-verbose, dont think it will help you much but just in case.

closed 2025-03-25 by jessebraham

CvRXX, 2025-03-28:
Sorry for the late reply, work got to me. Amazing that you guys fixed the issue and sorry for not picking it up. Keep up the good work!

PR #818 (2025-03-25): Option to supply custom defmt formats
The formatting options are identical to probe-rs's, except in espflash we can't control timestamp/location with switches. The user can supply a custom format string if such control is necessary.

Closes #520 
Closes #781