esp-rs/espflash #784: Review and update default partition tables for each target device

jessebraham, 2025-02-26:
As a result of #617 I discovered that for the ESP32-S3 the default factory partition is set to 1MB, which is unnecessarily small. This should be increased, but we should also review the other target devices and ensure that reasonable default settings are being used.

MabezDev, 2025-02-26:
Given that we can detect flash size, maybe we can just set the default partition table to the size of flash?

jessebraham, 2025-02-26:
The `save-image` subcommand does not require a device to be connected, so we still need some default value with how things are currently structured. Perhaps we can change this, but I have not investigated this yet so am not sure whether or not that's feasible.

MabezDev, 2025-02-26:
Ah I see, that does indeed complicate things :D. IIRC, the smallest flash size we offer with our chips is 2MB, so at the very least we should be able to bump the default for 2MB for `save-image` or for situations where flash size can't be detected.

closed 2025-03-04 by jessebraham