esp-rs/espflash #806: Question about app image format and bootloader

coolwanglu, 2025-03-07:
I'm using ESP32S3. I enabled Flash Encryption using a hello world ESP-IDF project. Then I flashed my esp-rs project (without bootloader), but it stopped working. The bootloader complained someting about "efuse blk revision", but I think it's just bad image data.

I compared the image of the ESP-IDF project and the esp-rs project. Both have the `esp_image_header_t` header, but the `esp-rs` binary does not have the [app description header](https://github.com/espressif/esp-idf/blob/v5.4/components/esp_app_format/include/esp_app_desc.h#L21). 

[image]

It works if I use the bootloader from `esp-rs`, (generate a merged image, encrypt and flash).

Is this expected? I thought the bootloader is generic, but maybe something changed between v5.1 and v5.4?

Not sure if this is related to #745

jessebraham, 2025-03-07:
I'm not sure what you're referring to by `esp-rs` exactly (this is just the name of the organization), but I'm assuming you mean a project using `esp-hal`?

We just recently merged a PR adding support for the ESP-IDF application descriptor (https://github.com/esp-rs/esp-hal/pull/3124), so using the `main` branch should resolve the problem when using the latest ESP-IDF bootloader.

Alternatively, using an older version of the bootloader, such as the one currently bundled with `espflash`, should work as well. IIRC this descriptor was added in ESP-IDF 5.3 (plus or minus a release).

coolwanglu, 2025-03-07:
Yeah, I mean an `esp-hal` project.
Thank a lot for your reply!

closed 2025-03-07 by coolwanglu