esp-rs/espflash #810: Espflash support for large flash sizes?

AdamHeavens, 2025-03-20:
Does espflash support larger then 16Mb flash sizes? I see parameters to specify 128Mb but I get the following and nothing over 16Mb will flash?

espflash flash --partition-table=partitions.csv  --baud 921600 --chip esp32s3 --port /dev/ttyUSB1 --flash-size 128mb target/xtensa-esp32s3-espidf/release/app-name

```
[2025-03-20T20:28:57Z INFO ] Serial port: '/dev/ttyUSB1'
[2025-03-20T20:28:57Z INFO ] Connecting...
[2025-03-20T20:28:57Z INFO ] Using flash stub
[2025-03-20T20:28:59Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        128MB
Features:          WiFi, BLE
MAC address:       70:04:1d:ae:cf:e4
Partition table:   partitions.csv
Error:   × Failed to parse partition table
  ╰─▶ The partition table is invalid
```

```
python3 $IDF_PATH/components/partition_table/gen_esp32part.py partitions.csv partitions.bin
Parsing CSV input...
Verifying table...
```

```
Partitions.csv
# Name,   Type, SubType,  Offset,   Size
nvs,data,nvs,0x9000,0x24000
otadata,data,ota,0x2d000,0x2000
factory,app,factory,0x30000,0x600000
app0,app,ota_0,0x630000,0x600000
app1,app,ota_1,0xC30000,0x600000
storage,data,spiffs,0x1230000,0x4000000
nvs_key,data,nvs_keys,0x5230000,0x400000
```

Thanks

Adam

SergioGasquez, 2025-03-24:
> Does espflash support larger then 16Mb flash sizes? I see parameters to specify 128Mb but I get the following and nothing over 16Mb will flash?

It should work, Id suggest updating to `espflash@main` : `cargo install espflash --git https://github.com/esp-rs/espflash`

> ```
> python3 $IDF_PATH/components/partition_table/gen_esp32part.py partitions.csv partitions.bin
> Parsing CSV input...
> Verifying table...
> ```

Does this command finishes? Or it gets stuck?

bjoernQ, 2025-04-09:
I can at least reproduce that the shown partition table is rejected with the current main (37f28f6ba0b8bb2b8578d737be7bc2a2c6d79a04)

SergioGasquez, 2025-04-28:
Hi! I’m closing this issue due to inactivity and because we haven’t been able to reproduce it. If the problem persists, feel free to reopen the issue. Thanks!

closed 2025-04-28 by SergioGasquez