esp-rs/espflash #815: Timeout on cargo run

Nickbot606, 2025-03-25:
Hello,

I am currently in the process of writing an embedded system for the xiao-esp32-c6. I currently have a macbook (silicon). I have followed your instructions on your readme to a T multiple times and I seem to be getting issues once I try to cargo run: 

[2025-03-25T04:01:45Z INFO ] Detected 2 serial ports
[2025-03-25T04:01:45Z INFO ] Ports which match a known common dev board are highlighted
[2025-03-25T04:01:45Z INFO ] Please select a port
[2025-03-25T04:01:46Z INFO ] Serial port: '/dev/tty.usbmodem101'
[2025-03-25T04:01:46Z INFO ] Connecting...
[2025-03-25T04:01:46Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command

When using the Arduino app, it seems to work fine. I don't know necessarily what I could be doing wrong at this point.

Thanks in advance.

bjoernQ, 2025-03-25:
Transfered this issue since it's related to espflash, not esp-generate

jessebraham, 2025-03-25:
> That doesn't look right

This is normal on macOS when using the USB Serial JTAG interface.

Nickbot606, 2025-03-26:
Hello and thank you for the quick response! 

I am currently running version 3.3.0 of espflash.

When I try to flash the `/dev/tty.usbmodem101` I get tty as well `/dev/cu.usbmodem101` however I get the same error message. The project does build itself but the issue is definitely when flashing. 

The weirdest part though is the fact that if I use the Arduino app and write a quick hello world in C, it works fine.

(From what I could see I believe this is my target architecture but I'm still new to embedded rust so please correct me if I'm wrong for the xiao-esp32-c6 chip https://www.seeedstudio.com/Seeed-Studio-XIAO-ESP32C6-p-5884.html)

As for the cargo.toml:

[target.riscv32imac-unknown-none-elf]
runner = "espflash flash --monitor --chip esp32c6"

[env]

[build]
rustflags = [
  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",
]

target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["alloc", "core"]

iaburton, 2025-03-27:
Hello!

Just wanted to comment that I seem to be having this issue as well; I'm unable to run any command without a timeout, unless I use the `--no-stub` option. This means some commands like `erase-flash` are unable to be used since they cannot be used with `--no-stub`. This is Arch Linux with espflash v3.3.0 & an esp32

```
❯ espflash erase-flash -S --chip esp32
[2025-03-27T06:18:30Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-03-27T06:18:30Z INFO ] Connecting...
[2025-03-27T06:18:31Z INFO ] Using flash stub
Error: espflash::timeout

  × Error while connecting to device
  ╰─▶ Timeout while running command

```

I've tried various combinations of options for baud/before/after/resetting without success. Any ideas? Thanks.

SergioGasquez, 2025-03-27:
Can you both try if the issue still persists with `espflash@main`: cargo install espflash --git https://github.com/esp-rs/espflash`?

iaburton, 2025-03-28:
> Can you both try if the issue still persists with `espflash@main`: cargo install espflash --git [https://github.com/esp-rs/espflash`](https://github.com/esp-rs/espflash%60)?

Well that seems to do the trick!

```
target/xtensa-esp32-none-elf/debug on  master [?]
❯ espflash -V
espflash 4.0.0-dev

target/xtensa-esp32-none-elf/debug on  master [?]
❯ espflash hold-in-reset
[2025-03-28T00:56:10Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-03-28T00:56:10Z INFO ] Connecting...
[2025-03-28T00:56:14Z INFO ] Using flash stub
[2025-03-28T00:56:14Z INFO ] Holding target device in reset

target/xtensa-esp32-none-elf/debug on  master [?] took 4s
❯ espflash erase-flash -S --chip esp32
[2025-03-28T00:56:23Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-03-28T00:56:23Z INFO ] Connecting...
[2025-03-28T00:56:27Z INFO ] Using flash stub
[2025-03-28T00:56:27Z INFO ] Erasing Flash...
[2025-03-28T00:56:48Z INFO ] Flash has been erased!
```

Nickbot606, 2025-03-28:
Likewise! It seemed to work. Thanks for you help! 
I used a basic `cargo generate esp-rs/esp-template` with the c6 selected for the speed studio xiao esp32 c6 and it seems to work fine.

Just in case anyone was curious, I found this to be the best config.toml file:
```
[target.riscv32imac-unknown-none-elf]
runner = "espflash flash --monitor"

[env]
ESP_LOG="info"

[build]
rustflags = [
  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",
]

target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["core"]
```

Here's my generated cargo.toml file. I found I had to roll back due to my chip to older versions of some repos.
```
[package]
name = "project_2"
version = "0.1.0"
authors = []
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-backtrace = { version = "0.14.2", features = [
    "esp32c6",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.21.0", features = [ "esp32c6" ] }
esp-println = { version = "0.12.0", features = ["esp32c6", "log"] }
log = { version = "0.4.22" }
[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

closed 2025-03-28 by Nickbot606