esp-rs/espflash #823: espflash using pseudo-terminal slave device

dan-corneanu, 2025-03-31:
I’m trying to flash an ESP32 DevKitC board using espflash from a Docker development container on OSX. The board is connected to my host via USB, and I’m using `ser2net` on the host to forward the serial connection to a TCP port. Inside the container, I use `socat` to connect the TCP port to a pseudo serial device.

Here are the commands I’m using:

1. On the host:
```bash
ser2net -n -c ./ser2net-esp32.yml
```
2. In the Docker container:
```bash
sudo socat pty,link=/tmp/dev/tty-esp32,waitslave tcp:host.docker.internal:1234
```

The `ser2net-esp32.yml` file looks like this:
```yml
connection: &toS0telnet
  accepter: telnet(rfc2217),tcp,1234
  connector: serialdev,/dev/tty.usbserial-130,local
```

Then, I try to flash the ESP32 with:
```bash
espflash flash -p /tmp/dev/tty-esp32 --monitor --chip esp32 target/xtensa-esp32-none-elf/release/esp32-blink
```

However, I get the following error:

```
× The serial port '/dev/pts/2' could not be found
  help: Make sure the correct device is connected to the host system
```

Both `/tmp/dev/tty-esp32` and `/dev/pts/2` exist in the container:
```bash
esp@e86dc5ff0d16:/app$ ls -all /tmp/dev
total 8
drwxr-xr-x 2 esp  esp  4096 Mar 31 06:42 .
drwxrwxrwt 1 root root 4096 Mar 31 06:42 ..
lrwxrwxrwx 1 root root   10 Mar 31 06:42 tty-esp32 -> /dev/pts/2

esp@e86dc5ff0d16:/app$ ls -all /dev/pts/2
crw--w---- 1 root tty 136, 2 Mar 31 07:10 /dev/pts/2
```

I expect `espflash` to detect and use the pseudo terminal (`/tmp/dev/tty-esp32`), but it cannot find it. 
Any advice on resolving this?

MabezDev, 2025-03-31:
I believe all our port detection comes from https://github.com/serialport/serialport-rs, I suggest filing an issue there too. I'm not sure we can do much on our end here in espflash.

kruton, 2025-07-09:
I think the problem is that given the command line argument `-p /dev/ttyS9000`, you're using [`serialport::available_ports()`](https://github.com/esp-rs/espflash/blob/2e3f3c362cd6f052a27d50cbea3494bb2839cf87/espflash/src/cli/serial.rs#L112) and matching entries listed there instead of just trying to call [`serialport::new("/dev/ttyS9000", 115200).open()`](https://docs.rs/serialport/latest/serialport/fn.new.html). This probably works in most of the cases, but the `socat` created pseudo-serial devices don't show up in `udev` lists like the rest of the real serial devices.

However, you're going to immediately run into issues when `espflash` tries to reset the device because none of the current strategies will work.