esp-rs/espflash #854: Review public API prior to next release

jessebraham, 2025-04-23:
The goal is to minimize API surface while still being as useful as possible when using `espflash` as a library for various use cases.

Related to #794

MabezDev, 2025-05-22:
I spent a bit of time looking into this, here are some notes from me:

 * We likely want to cfg some error cases away that are dependent on the `serialport` feature
 * We have a `Target` trait, with no public implementors, seems to me we should ditch the trait and just implement these directly on the `Chip` enum (and use the chip enum in more places, instead of the `struct Esp32` stuff), especially given this sort of impl in the trait:
   ```rust
   fn chip(&self) -> Chip {
        Chip::Esp32
    }
   ``` 
* Same as above but for the Efuse trait
* Image format module is dependent on changes in https://github.com/esp-rs/espflash/issues/745 so ignoring for now
* A tonne of stuff not respecting the Rust API guidelines, but covered by https://github.com/esp-rs/espflash/issues/864
* Seems like we have some valuable input here: https://github.com/esp-rs/espflash/issues/794#issuecomment-2885299089. Let's take this into account, particularly about making sure `Flasher` can be constructed, and `reset()` via library calls is possible.

Overall, it's not looking to bad, I think most of this could be refactored in a few days. I think we should resolve #745 before tackling these.

I'll also note now, that I don't think we need this to be perfect, we can release another major version of this tool without too much churn later, so to me this is about resolving #745, cleaning up the API as much as we can within a reasonable time constraint then cutting the release.

MabezDev, 2025-05-23:
Other notes from my comment in #793:

* Regardless of above, I think `Connection` should be constructed outside of `Flasher` to avoid passing in `Connection`'s constructor params into `Flasher::connect`
* I don't know where `SpiAttachParams` should live, it acts like an input but we only ever use `SpiAttachParams::default()` or `SpiAttachParams::esp32_pico_d4()`. If we don't plan on allowing the users to configure this, we should try and associate it with `Chip`, it should simplify a lot of things.
* For cases where there are a large number of input params, that we can't untangle, we should consider a config struct to encapsulate it together.