esp-rs/espflash #860: [ESP32-S2] Not able to connect prior using esptool

aemarkov, 2025-05-10:
Hello!
I am trying to flash Wemos S2 mini (ESP32-S2) via USB. But `espflash` unable to connect to chip until `esptool.py` connects first.

**Versions**
OS: Linux, distro Manjaro, kernel 6.13.12
espflash version: 4.0.0-dev (d76820f30aac3108e113878eb9879ca700812e21)

**Steps to reproduce**
1. Install espflash from current main, because #813 is not included into current stable release (v3.3.0)
```
cargo install espflash --git https://github.com/esp-rs/espflash
```

2. Connect Wemos S2 to via USB and turn to bootloader/download/whaterver mode by holding GPIO0 and resetting. New tty device appears. 
3. Try to connect.
Expected behavior: espflash connects to board and displays info
Actual behavior: error, see below. Also, during connection attempt, chip is reseted (see via  dmesg).
```
RUST_LOG="debug"  espflash board-info
[2025-05-10T19:39:42Z INFO ] Serial port: '/dev/ttyACM0'
[2025-05-10T19:39:42Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```
Detailed log:
```
[2025-05-10T19:40:51Z INFO ] Serial port: '/dev/ttyACM0'
[2025-05-10T19:40:51Z INFO ] Connecting...
[2025-05-10T19:40:51Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2025-05-10T19:40:51Z DEBUG] Writing command: Sync
[2025-05-10T19:40:52Z DEBUG] Writing command: Sync
[2025-05-10T19:40:52Z DEBUG] Writing command: Sync
[2025-05-10T19:40:52Z DEBUG] Writing command: Sync
[2025-05-10T19:40:52Z DEBUG] Writing command: Sync
[2025-05-10T19:40:52Z DEBUG] Failed to reset, error Connection(
        ConnectionFailed,
    ), retrying
[2025-05-10T19:40:52Z DEBUG] Using UnixTight reset strategy with delay of 500ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Io(
                    Uncategorized,
                ),
                description: "Input/output error (os error 5)",
            },
        ),
    ), retrying
[2025-05-10T19:40:53Z DEBUG] Using Classic reset strategy with delay of 50ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Unknown,
                description: "I/O error",
            },
        ),
    ), retrying
[2025-05-10T19:40:53Z DEBUG] Using Classic reset strategy with delay of 500ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Unknown,
                description: "I/O error",
            },
        ),
    ), retrying
[2025-05-10T19:40:53Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Io(
                    Uncategorized,
                ),
                description: "Input/output error (os error 5)",
            },
        ),
    ), retrying
[2025-05-10T19:40:53Z DEBUG] Using UnixTight reset strategy with delay of 500ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Io(
                    Uncategorized,
                ),
                description: "Input/output error (os error 5)",
            },
        ),
    ), retrying
[2025-05-10T19:40:53Z DEBUG] Using Classic reset strategy with delay of 50ms
[2025-05-10T19:40:53Z DEBUG] Failed to reset, error Connection(
        Serial(
            Error {
                kind: Unknown,
                description: "I/O error",
            },
        ),
    ), retrying
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

4. Reset Wemos S2 to the bootloader again
5. Connect using `esptool.py`
`esptool` successfully connects to the target. Also, when it starts connecting, chip is reseted (see via dmesg) and then tool connects to the ESP.
```
esptool.py -p /dev/ttyACM0 chip_id
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 48:27:e2:57:81:12
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S2 has no Chip ID. Reading MAC instead.
MAC: 48:27:e2:57:81:12
Error: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
```
6. Now, try again connecting via `espflash`  WITHOUT reseting/reconnecting the ESP. And now it works correctly. Device is not reseted during connection.
Detailed log
```
[2025-05-10T19:47:51Z INFO ] Serial port: '/dev/ttyACM0'
[2025-05-10T19:47:51Z INFO ] Connecting...
[2025-05-10T19:47:51Z DEBUG] Using UnixTight reset strategy with delay of 50ms
[2025-05-10T19:47:51Z DEBUG] Writing command: Sync
[2025-05-10T19:47:51Z DEBUG] Writing command: Sync
[2025-05-10T19:47:51Z DEBUG] Writing command: Sync
[2025-05-10T19:47:52Z DEBUG] Writing command: Sync
[2025-05-10T19:47:52Z DEBUG] Writing command: GetSecurityInfo
[2025-05-10T19:47:52Z DEBUG] Writing command: ReadReg { address: 40001000 }
[2025-05-10T19:47:52Z INFO ] Using flash stub
[2025-05-10T19:47:52Z DEBUG] Loading flash stub for chip: Esp32s2
[2025-05-10T19:47:52Z DEBUG] Writing command: ReadReg { address: 3ffffd14 }
...
```
7. You can continue using espflash again multiple times after first connection attempt with esptool

**What I've tried**
I've tried using different  `--before` flags.
- nothing, `default-reset`, `no-reset`, `usb-reset` - all works AFTER using esptool
- none of these options help WITHOUT using esptool

MerrimanInd, 2025-07-27:
I'm having a similar issue with an ESP32S2. I was able to flash on 4.0.0dev using `espflash`, upgrade to 4.0.1 and can't seem to make it work. But `esptool` still seems to connect fine. Can you tell me how you got the detailed logs? I don't see a documented CLI arg for it.

MerrimanInd, 2025-07-28:
This is my attempt to replicate. First, connecting using esptool.

```
xander@spareparts:~/Repos_Rust Embedded/espflash$ esptool --after no-reset chip_id
Warning: Deprecated: Command 'chip_id' is deprecated. Use 'chip-id' instead.
esptool v5.0.0
Found 33 serial ports...
...
Serial port /dev/ttyS20:
Connected to ESP32-S2 on /dev/ttyACM0:
Chip type:          ESP32-S2 (revision v0.0)
Features:           Wi-Fi, Single Core, 240MHz, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of eFuse V2
Crystal frequency:  40MHz
USB mode:           USB-OTG
MAC:                60:55:f9:eb:dd:0a

Stub flasher running.

Warning: ESP32-S2 has no chip ID. Reading MAC address instead.
MAC:                60:55:f9:eb:dd:0a

Staying in bootloader.
```

Then attempting a connection with espflash and seeing the chip.
```
xander@spareparts:~/Repos_Rust Embedded/espflash$ espflash board-info --no-stub
[2025-07-28T05:45:50Z INFO ] Serial port: '/dev/ttyACM0'
[2025-07-28T05:45:50Z INFO ] Connecting...
[2025-07-28T05:45:51Z WARN ] Could not detect flash size (FlashID=0x00, SizeID=0x00), defaulting to 4MB
Chip type:         esp32s2 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V2
MAC address:       60:55:f9:eb:dd:0a

Security Information:
=====================
Flags: 0x00000000 (0)
Key Purposes: [0, 0, 0, 0, 0, 0, 0]
Secure Boot: Disabled
Flash Encryption: Disabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0
```

MerrimanInd, 2025-08-07:
Yeah I tried every --before and --after flag, it doesn't seem to make a difference. No new insights from the debug logs either. But it does look like espflash successfully loads the stub but then fails to connect to it.

This template project is still on esp-hal-1.0.0-beta.0 so I'm going to try pulling it up to the latest RC release, add the `esp_bootloader_esp_idf::esp_app_desc!();` macro and see if that makes a difference.

SergioGasquez, 2025-12-11:
Hi! We just updated the stubs in `main` could you try using `cargo install espflash --git https://github.com/esp-rs/espflash` and checking if the issue persist?

SergioGasquez, 2026-08-28:
Ill be closing this issue since its inactive and it seems to be a stub issue, and they were updated. If the issue persist, feel free to reopen

closed 2026-08-28 by SergioGasquez