esp-rs/espflash #861: Add support for the ESP32-C5

jessebraham, 2025-05-19:


jessebraham, 2025-05-21:
Not sure how we should proceed with this, have things working locally. However, we do not have a HIL runner set up yet (this is fixable), and the bigger issue, we have no real way of generating test binaries for HIL testing for this target. Should we exclude this chip from any tests we're not able to currently run, just so that we can get support merged, or hold off on merging support until a later date?

cc @MabezDev @SergioGasquez

bugadani, 2025-05-21:
I know I'm not pinged here but I'd be fine with an untested initial implementation. We could maybe build a firmware with esp-idf, but I think all of us would like to avoid that if we can.

MabezDev, 2025-05-21:
Let's exclude the target from test for now, when we start adding support in esp-hal etc we can come back and add the test cases. For now I think just having some partial support for the C5 is valuable.

PR #863 (2025-05-22): Add support for the ESP32-C5
Closes #861
closed 2025-05-22 by SergioGasquez